fwupd/firmware-lenovo #82: Missing ThinkCentre M920s Firmware

joakim-tjernlund, 2020-11-06:


mrhpearson, 2020-11-06:
I'm tracking this one

joakim-tjernlund, 2020-12-17:
ping

joakim-tjernlund, 2021-03-18:
ping, we are getting more of these now and they have really old FW that needs updating, lots of new versions of this FW at Lenovo

mrhpearson, 2021-03-18:
ack.
The desktop team have not delivered any firmware to LVFS. It's a sore point and I've been fighting it but so far unsucessfully - they are supposed to be doing it but haven't. (Note, this is peculiar to the desktop team, and doesn't affect our Thinkpad and workstations Linux platforms)

I'll go and push again - if you're able to have an off forum discussion drop me a note (markpearson at lenovo com) so I can collect some more information that would help me escalate.

Mark

PhilippWendler, 2022-06-07:
Any update on this? We have some M70s for which only an [old firmware](https://fwupd.org/lvfs/devices/com.lenovo.M90t_M90s_M80t_M80s_M70t_M70s.BIOS.firmware) is on LVFS.

joakim-tjernlund, 2022-07-09:
@mrhpearson  , any news w.r.t M920s FW ?

joakim-tjernlund, 2023-05-22:
We have a few of these and see alot of:
```
i2c i2c-1: sendbytes: NAK bailout.
```
which I suspect is FW related.
It has been more than a year now and no FW yet.

mrhpearson, 2023-05-24:
Wow - this one is old...

Looks like I didn't raise a JIRA ticket for this, so I'm having trouble tracking down the old thread or any details.
Correct me if I'm wrong - but the M920s is 5+ years old? It means it's of the generation before the ThinkCenter team agreed to put FW on LVFS. 

I can see there is a version there - but it's four years old and never left embargo state so I suspect they had issues with it (but I'm guessing a lot)

I will send off a question to the FW team but my recommendation would be to update using the ISO image from the support site.

joakim-tjernlund, 2023-05-24:
*-firmware
          description: BIOS
          vendor: LENOVO
          physical id: 0
          version: M1UKT33A
          date: 03/14/2019
          size: 64KiB
          capacity: 12MiB

I am sure we have newer than that though. Just as sure that there will be no ISO upgrade unless
the machine is on fire.

 Jocke