fwupd/firmware-lenovo #83: Where do I find the checksums?

qoheniac, 2020-11-09:
```
$ fwupdmgr --version
client version:	1.5.1
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1
$ fwupdmgr --help | grep " block-firmware`
  block-firmware [CHECKSUM]         Blocks a specific firmware from being installed
```
`fwupdmgr get-updates` shows two updates and I want to block them, because my laptop got stuck during update and I was forced to remove the battery to get it back to work. It seems like the update failed in a very early stage and nothing was changed but `gnome-software` keeps reminding me of those updates and I definitely want to do this a second time. I thought maybe `fwupdmgr block-firmware` would help but I don't know where to find those checksums.

qoheniac, 2020-11-09:
`fwupdmgr block-firmware` without a checksum asks for what device I want to block firmware updates. It then blocks the latest firmware update for that device. As there are multiple firmware updates for the system firmware it still suggests the installation of the less recent updates. So I guess I still need to find out the checksum for those.

mrhpearson, 2020-11-09:
Thakns Mario.
I've flagged the issue to the firmware team and will get their input
(for my internal tracking: LO-898)

Thanks
Mark

mrhpearson, 2020-11-10:
Hi 
1.25 will be available very soon (they've finished testing and it's ready to be released - just going through the final steps of their process).
1.27 will follow shortly after. There were some internal reasons that require users to go via 1.25 first so that took precendence - and it will be a two step process. Update to 1.25 when it comes out and then 1.27 after when that becomes available (it still has to go through some internal verification).
Thanks
Mark

hughsie, 2020-11-10:
> Update to 1.25 when it comes out and then 1.27 after when that becomes available

We need to remember to set the `<requirement>` tags on 1.27 to be 1.25 else it'll be chaos :)

qoheniac, 2020-11-10:
Just for the sake of completeness, I was able to block the firmware update by adding the device GUIDs to `DisabledDevices` in `/etc/fwupd/daemon.conf`.

mrhpearson, 2020-11-11:
Looking at the above - I was putting the wrong comments for the wrong ticket...tempted to delete them but that might just be confusing. Apologies for the mess...(@hughsie - if it makes sense to delete my above two comments let me know)

The team looked a the T470s issue and haven't been able to reproduce the problem - going from 1.20 to 1.21 using fwupd 1.5.1 so right now we're not sure what the issue is. There have been a lot of succesful updates previously.

Mark

qoheniac, 2020-11-11:
I installed a fresh vanilla Fedora 33 system from an USB stick and started the firmware update via GNOME Software. I have a second T470s where I did not try it. But I'm also too afraid to do so as I do not want to damage the devices as they are both in active use. So I won't try to reproduce it, but I saw other reports on the Internet and I if I remember correctly all of them installed the firmware update from GNOME Software in a Fedora system.

qoheniac, 2020-11-11:
Also maybe not the update of the embedded controller but of the system firmware caused the problem. I clicked all available updates in GNOME Software, i.e. embedded controller, system firmware and Intel management engine and then rebooted. The system got stuck during the update and I had to take out the battery. After that the laptop booted into Fedora as if nothing happened and the Intel management engine was updated while the embedded controller and the system firmware were not.

mrhpearson, 2020-11-11:
Hi - do you have any links to other reports? I'll forward the details on to the firmware team
Mark

qoheniac, 2020-11-11:
For example [here](https://github.com/fwupd/firmware-lenovo-thinkpad/issues/43) and [here](https://www.reddit.com/r/thinkpad/comments/bwzsoe/thinkpad_t470s_firmware_update_hanging/)

qoheniac, 2020-11-11:
Here is the output of `fwupdmgr get-history`:

    20HGS01800
    │
    ├─Intel Management Engine:
    │ │   Device ID:          cbede50560aaa37366326cd54923421517427f39
    │ │   Previous version:   184.50.3425
    │ │   Update State:       success
    │ │   Last modified:      2020-11-04 23:53
    │ │   GUID:               e9124c4a-fdff-42e5-b2ad-f745db345953
    │ │   Device Flags:       • Internal device
    │ │                       • Updatable
    │ │                       • System requires external power source
    │ │                       • Supported on remote server
    │ │                       • Needs a reboot after installation
    │ │                       • Reported to remote server
    │ │                       • Device is usable for the duration of the update
    │ │ 
    │ └─ThinkPad T470s Corporate ME Update:
    │       New version:      184.77.3664
    │       Remote ID:        lvfs
    │       Summary:          Lenovo ThinkPad T470s Corporate ME Firmware
    │       License:          Proprietary
    │       Size:             7.5 MB
    │       Created:          2016-07-08
    │       Urgency:          High
    │       Vendor:           Lenovo Ltd.
    │       Description:      
    │       Version 11.8.77.3664 (LVFS: 184.77.3664) 2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR) Several security fixes and enhancements are on this release.
    │       
    │       Security issues fixed:
    │       
    │       • CVE-2020-8674
    │       • CVE-2020-0596
    │       • CVE-2020-0595
    │       • CVE-2020-0594
    │       • CVE-2020-0545
    │       • CVE-2020-0540
    │       • CVE-2020-0539
    │       • CVE-2020-0538
    │       • CVE-2020-0537
    │       • CVE-2020-0536
    │       • CVE-2020-0535
    │       • CVE-2020-0533
    │       • CVE-2020-0532
    │       • CVE-2020-0531
    │     
    ├─Embedded Controller:
    │ │   Device ID:          03a6930fc4c833443a7502dde379b98b961c8c57
    │ │   Previous version:   0.1.20
    │ │   Update State:       failed
    │ │   Update Error:       failed to run update on reboot
    │ │   Last modified:      2020-11-04 23:53
    │ │   GUID:               e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
    │ │   Device Flags:       • Internal device
    │ │                       • Updatable
    │ │                       • System requires external power source
    │ │                       • Supported on remote server
    │ │                       • Needs a reboot after installation
    │ │                       • Reported to remote server
    │ │                       • Device is usable for the duration of the update
    │ │ 
    │ └─  New version:      0.1.21
    │       Remote ID:        lvfs
    │       License:          Unknown
    │     
    ├─System Firmware:
    │ │   Device ID:          84149fb78009cf27cc1dd5520911f46f8792dbfe
    │ │   Previous version:   0.1.32
    │ │   Update State:       failed
    │ │   Update Error:       failed to run update on reboot
    │ │   Last modified:      2020-11-04 23:53
    │ │   GUID:               7a176688-0960-47ba-931b-7829849e8347
    │ │   Device Flags:       • Internal device
    │ │                       • Updatable
    │ │                       • System requires external power source
    │ │                       • Supported on remote server
    │ │                       • Needs a reboot after installation
    │ │                       • Reported to remote server
    │ │                       • Cryptographic hash verification is available
    │ │                       • Device is usable for the duration of the update
    │ │ 
    │ └─  New version:      0.1.38
    │       Remote ID:        lvfs
    │       License:          Unknown
    │     
    └─THNSF5256GPUK TOSHIBA:
      │   Device ID:          e11623b2caa18fee292058a5c09ca4e6152f7ecf
      │   Previous version:   51025KLA
      │   Update State:       success
      │   Last modified:      2020-11-04 23:53
      │   GUID:               a1aa7718-9375-5e24-9b93-5426989c82a0
      │   Device Flags:       • Internal device
      │                       • Updatable
      │                       • System requires external power source
      │                       • Supported on remote server
      │                       • Needs a reboot after installation
      │                       • Reported to remote server
      │                       • Device is usable for the duration of the update
      │ 
      └─  New version:      51055KLA
            Remote ID:        lvfs
            License:          Unknown