fwupd/firmware-lenovo #84: Wrong update for E495

yunusemrecan, 2020-11-10:
``` 
$ fwupdmgr --version
client version:	1.5.1
compile-time dependency versions
	 gusb:	0.3.5

daemon version:	1.5.1
```
``` 
$ fwupdmgr --help | grep " block-firmware"
  block-firmware [CHECKSUM]         Blocks a specific firmware from being installed
```
There is a update for my E495 yet it says it is for T14/P14s

```
20NE001RTX
│
└─System Firmware:
  │   Device ID:          3ba7ea853f9613adbcbb9326dbe5f0decd677a67
  │   Current version:    0.1.19
  │   Vendor:             LENOVO (DMI:LENOVO)
  │   GUIDs:              9bfd8b34-080b-40a1-47b3-ca1efa289e6d
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
  │                       c0b0e93a-4864-5803-8eb5-d31ee7fa4dba
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14/P14s Gen 1 AMD System Update:
        New version:      0.1.25
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T14 P14s AMD System Firmware
        License:          Proprietary
        Size:             17.9 MB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Fixed an issue where BIOS has incorrect GUID in ACPI table.In case user cannot update the UEFI
        
        BIOS  later version,must update v1.25 first.
        
        • Truncated TCO logo at POST screen on both HD andFHD Display.
        • Update PI patch code.
        • Fixed an issue that TP&Trackpoint/Keyboard have no function after close lid during
        
        Restart/Full shutdown/S4.
        
        • Fixed an issue that the POST hang up with Lindy cable.
```

mrhpearson, 2020-11-10:
That update is definitely for the T14 AMD - it shouldn't be flagged for the E495 (don't update to it!)
@hughsie - do I need to chase the firmware team for this or is there some fwupd quirk happening?
Mark

hughsie, 2020-11-10:
@mrhpearson this is the firmware: https://fwupd.org/lvfs/components/6883 -- the GUID is listed as 9bfd8b34-080b-40a1-47b3-ca1efa289e6d which seems to match the value found on the ESRT on the E495. Either the Lenovo team used the same "random" GUID between the two models or the wrong metadata was included.

I've moved the firmware back to embargo-lenovo_thinkpad -- can you ask internally what happened? If the GUIDs are the same (but the firmware is different!) then we can use a CHID restriction to make it match against DMI data.

@yunusemrecan thanks for reporting!

mrhpearson, 2020-11-10:
Ack - will follow up. Thanks from me too @yunusemrecan 
Mark

yunusemrecan, 2020-12-22:
Sorry to bother but same 1.25 bios for t14 reappered for my e495. Also I am on 1.19 while lastest bios is 1.20 for e495. Updating manually could solve problem?

kmauleon, 2020-12-22:
hi @yunusemrecan thank you very much for the information,,, already forwarded this to owner...

kmauleon, 2020-12-23:
hi @yunusemrecan please see below for owner's remarks... can you please check... thank you very much....
_As GUID problem, we add CHID to control publish version this morning and removed the original package, it should fix by new package now. we also confirmed the version won’t display on Ex95 project now. thank you._

closed 2021-12-31 by yunusemrecan