fwupd/firmware-lenovo #85: Lenovo T470s: Does not load Linux-Firmware-Updater

MakisH, 2020-11-11:
**Describe the bug**

After running `fwupdmgr update`, my laptop restarts but it proceeds to the normal GRUB2 screen, without performing the updates. Trying to manually select the "Linux Firmware Updater" in the boot menu does nothing: the boot menu reappears after a very short blank screen. I have a similar entry "ubuntu", which behaves the same.

In some aspects, this issue is similar to https://github.com/fwupd/fwupd/issues/1101 and https://github.com/fwupd/fwupd/issues/1538, but the information I gathered from there did not help. I am afraid I don't understand UEFI well enough at the moment, so I will try to comment with my observations on the related output and I would appreciate any hints.

**Steps to Reproduce**
1. Update the firmware either from Gnome Software or running `$ sudo  fwupmgr update`. Skipping `sudo` also works (I guess because the updates are signed).
2. fwupdmgr downloads the updates and schedules a reboot, setting the status as `needs-reboot`.

**Expected behavior**
- Rebooting the system should boot into the Linux-Firmware-Updater
- Manually selecting Linux-Firmware-Updater in the boot menu should also work

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.5
```
- I understand that this is recent enough. Nothing in the [release notes of the latest 1.4.6](https://github.com/fwupd/fwupd/releases/tag/1.4.6) seems to be related. 
- https://github.com/fwupd/fwupd/issues/1794 mentions an issue with buggy shim. I am using:
```
$ apt-cache policy shim
shim:
  Installed: 15+1552672080.a4a1fbe-0ubuntu2
  Candidate: 15+1552672080.a4a1fbe-0ubuntu2
  Version table:
 *** 15+1552672080.a4a1fbe-0ubuntu2 500
```
This is the same commit (`a4a1fbe`) mentioned by @julian-klode on https://github.com/rhboot/shim/issues/194, but I understand that this if fixed now in Ubuntu.

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): came preinstalled via APT.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

<details><summary>This is quite long, so I though I could save some screen space (click to reveal)</summary>

```shell
$ fwupdmgr get-devices --show-all-devices
20HGS0A600
│
├─Embedded Controller:
│     Device ID:           03a6930fc4c833443a7502dde379b98b961c8c57
│     Current version:     0.1.21
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─HD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               2886e312-afd8-5510-8993-12d568a85e00
│                          38f6c009-a25c-5b50-b3b5-fef4f9c6b846
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel Management Engine:
[... 172 lines cut ...]
│           ├── mmx64.efi
│           └── shimx64.efi
├── grub
 .... # removed
```
I understand that `fwupdx64.efi` is the `Linux-Firmware-Updater` and the `.cap` files are updates to be installed. The contents of my `/boot/efi/` seem to be identical to e.g. https://github.com/fwupd/fwupd/issues/1794. 

**Additional questions**
- Operating system and version:
   - Ubuntu 20.10 (kernel 5.8.0-25-generic)
   - Ubuntu 16.04 installed as well in a separate partition, included in the same GRUB menu. Until I install 20.10 I did not have a dedicated `/boot/efi` partition, so maybe I have somehow mixed UEFI and Legacy?
   - This system never had a Windows installation.
- Have you tried rebooting?
   - yes
- Is this a regression?
   - No. I have never managed to make it work before.
- Are you using an NVMe disk?
    - yes
- Is secure boot enabled?
    - yes & no: I get the same problem with both. Before I had it disabled, I enabled it just now.

A few additional points:
- The option "Boot order lock" is disabled.
- Firmware updates are allowed and I hava managed to update the Firmware with non-fwupd methods in the past. In that case (with Ubuntu 16.04), getting updates from fwupd was again not working (due to very old versions) and I did tinker a bit back then trying to upgrade.
- I already tried removing `Boot0001* Linux-Firmware-Updater	HD(3,MBR,0xff767ef,0x64bb800,0x100800)/File(\EFI\ubuntu\fwupdx64.efi)` with `sudo efibootmgr -B -b 0001` and re-running `fwupdmgr update`.

MakisH, 2020-11-15:
(trying to log my progress, any ideas/hints welcome)

It looks like I am always booting from:
```
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
```

I do see the options for "ubuntu" and for "Linux-Firmware-Updater" in my Boot Menu (F12), but I cannot execute either of them. After selecting any of these, I immediately get back to the boot menu, without any message.

What residue files could I have from a previous installation in "legacy" (non-uefi) mode?

MakisH, 2020-11-22:
A maybe important addition/update (following a discussion on [askubuntu](https://askubuntu.com/questions/1294353/understanding-my-boot-efi-tree)): I am using an MBR partition table, not GPT. Could this be an issue for fwupd?

hughsie, 2020-11-23:
> I am using an MBR partition table

As am I, and it seems to work okay.

s1shed, 2021-03-04:
I was able to solve it in Ubuntu 20.04 by installing shim-signed and grub-efi-amd64-signed. After doing so I was able to pick linux-firmware-updater from the the boot menu and my updates were finally applied.

For some reason I had grub-pc installed along with grub-efi, even though I am using efi. The fw updates _did_ work at some point last year but was apparently broken with some deb pkg update...

MakisH, 2021-03-05:
@s1shed good observation, thank you very much for replying here. I had already both packages installed. I also had `grub-pc` along with `grub-efi-amd64-signed`. However, removing `grub-pc` installed `grub-efi-amd64`, which confuses me. So, now I have both efi variants (simple and signed).

This still did not work for me: after installing `grub-efi-amd64`, I stopped seeing `ubuntu` in my boot menu options (which anyway was not working). The `Linux Firmware Updater` is still there, but still not loading. Strange, but it looks like I am closer now.

mlalkaka, 2021-03-06:
I think I've been experiencing this same issue, which is also probably the cause of #1101. The Linux-Firmware-Updater was last in the boot order in the BIOS, and I don't think I ever changed that. I'm guessing fwupd would automatically boot into Linux-Firmware-Updater when needed, even though it was last in the boot order. For now, I worked around this issue by reordering Linux-Firmware-Updater to be first in the boot order, then running `sudo fwupdmgr update`, and then restarting the laptop.

jrop, 2022-01-20:
@mlalkaka I just changed Linux-Firmware-Updater to be first in my EFI entries, and ran the update command, and it still failed to update the firmware.  I have a T14s (Gen 1).

hughsie, 2022-01-20:
> some Lenovo laptops only allow you to update one device say a time

We've added this restriction in fwupd for Lenovo since version 1.5.6

MakisH, 2022-01-20:
Since my main issue is still that I cannot even load the `Linux-Firmware-Updater`, is there any way I could debug this? For example, somehow reinstall it or get any kind of insight on why it is not loading but just immediately returning to the boot menu?

jrop, 2022-01-20:
@mlalkaka @hughsie Ah, I'm on fwupd 1.5.11, I'll try and make sure that it is running for one device at a time.  Not sure if it was multiple this morning or not, I'll have to jump back on this when I can reboot my laptop repeatedly (as it is now, I am in the middle of tasks that I don't want to terminate).

eyalzek, 2022-02-17:
Seeing the same behavior on X1 Nano running fwupd 1.7.4-2...

mbacchi, 2022-02-22:
> > some Lenovo laptops only allow you to update one device say a time
> 
> We've added this restriction in fwupd for Lenovo since version 1.5.6

Thanks for this information, as a Lenovo user for a number of years I didn't know this. How can we 'unselect' multiple devices if we've already selected more than one to be updated? I did a little digging on the fwupd wiki and don't see a way to do that.

hughsie, 2022-02-22:
> How can we 'unselect' multiple devices if we've already selected more than one to be updated

I don't think you need to; we only do the BootNext once and then we revert back to the normal bootloader. After that you can just save a new update and then it should only do one, not multiple.

mlalkaka, 2022-02-24:
> Thanks for this information, as a Lenovo user for a number of years I didn't know this. How can we 'unselect' multiple devices if we've already selected more than one to be updated? I did a little digging on the fwupd wiki and don't see a way to do that.

@mbacchi try my instructions in [this comment](https://github.com/fwupd/firmware-lenovo/issues/85#issuecomment-1017668999). That worked for me on a Lenovo T480s laptop.

MakisH, 2022-05-17:
Just an update: I have the same issue with Ubuntu 22.04. I have now deleted my 16.04 partition and I currently have:

```
$ efibootmgr -v
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0002,0001,0017,0010,0011,0012,0013,0018,0000,001A,001C,001D,001E,0019,001B
Boot0000* Linux-Firmware-Updater	HD(3,MBR,0xff767ef,0x800,0x1d5000)/File(\EFI\ubuntu\fwupdx64.efi)
Boot0001* ubuntu	HD(3,MBR,0xff767ef,0x800,0x1d5000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* ubuntu	HD(2,MBR,0xff767ef,0x800,0x1d5000)/File(\EFI\ubuntu\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017  USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018  USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A  ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C  PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0023* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

I assume the double "Ubuntu" record has something to do with either moving the partition, or with upgrading to 22.04. In any case, as always, neither of them starts (and same for the "Linux-Firmware-Updater").

MakisH, 2022-05-19:
:tada: **SUCCESS:** I [converted my partition table from MBR to GPT](https://serverfault.com/questions/963178/how-do-i-convert-my-linux-disk-from-mbr-to-gpt-with-uefi/963179#963179) and it now worked like a charm!

> > I am using an MBR partition table
> 
> As am I, and it seems to work okay.

@hughsie somehow, on my system, UEFI+MBR did not seem to ever work. But this in the end was not a fwupd issue: I installed Fedora 36 next to my Ubuntu 22.04, and I noticed that I could not boot that either. Comparing with another laptop, I could only nail it down to the difference between MBR and GPT.

Should I document this somewhere more prominently?

closed 2022-05-19 by MakisH

jrop, 2022-05-19:
My partition table is already GPT, and fwupd still does not work to update my system:

```
❯ sudo parted -l
Model: Linux device-mapper (crypt) (dm)
Disk /dev/mapper/cryptswap: 4294MB
Sector size (logical/physical): 512B/512B
Partition Table: loop
Disk Flags:

Number  Start  End     Size    File system     Flags
 1      0.00B  4294MB  4294MB  linux-swap(v1)

Model: Sabrent (nvme)
Disk /dev/nvme0n1: 1024GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags:

Number  Start   End     Size    File system     Name      Flags
 1      2097kB  524MB   522MB   fat32                     boot, esp
 2      524MB   4819MB  4295MB  fat32           recovery  msftdata
 3      4819MB  1020GB  1015GB  ext4
 4      1020GB  1024GB  4295MB  linux-swap(v1)            swap
```

hughsie, 2022-05-19:
> only nail it down to the difference between MBR and GPT.

Ohh interesting.

> Should I document this somewhere more prominently?

Yes please, there's a wiki article here: https://github.com/fwupd/fwupd/wiki/Common-Problems

MakisH, 2022-05-19:
> > Should I document this somewhere more prominently?
> 
> Yes please, there's a wiki article here: https://github.com/fwupd/fwupd/wiki/Common-Problems

Done.

> My partition table is already GPT, and fwupd still does not work to update my system:

But do you also have the issue of not being able to boot from any of the "named" entries in the boot menu (e.g., "Ubuntu", "Fedora", but only "nvme0n1" or "USB")?