fwupd/firmware-lenovo #87: FW for X1C7 is not installed correctly - still available after reboot

orclex, 2020-11-19:
Hi,
There are firmware updates available for my Lenovo Thinkpad X1 Carbon Gen7.
After updating and rebooting the firmware updates are available again.

root@m19040-lin:/usr/local/src# fwupdmgr get-updates
• Thunderbolt Controller has the latest available firmware version
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
20QDCTO1WW
│
├─Embedded Controller:
│ │   Device ID:           c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│ │   Current version:     0.1.17
│ │   Minimum Version:     0.1.17
│ │   Vendor:              DMI:LENOVO
│ │   Update State:        success
│ │   GUID:                f72e048b-65bd-4e71-9071-1ac7045223e5
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Device is usable for the duration of the update
│ │ 
│ ├─ThinkPad X1 Carbon 7th / X1 Yoga 4th (W-BIOS for Machine types: 20QF, 20QG, 20QD, 20QE) Embedded Controller Update:
│ │     New version:       0.1.20
│ │     Remote ID:         lvfs
│ │     Summary:           Lenovo ThinkPad X1 Carbon 7th / X1 Yoga 4th Embedded Controller Firmware
│ │     License:           Proprietary
│ │     Size:              767.2 kB
│ │     Created:           2016-07-08
│ │     Urgency:           High
│ │     Vendor:            Lenovo Ltd.
│ │     Flags:             is-upgrade
│ │     Description:       Lenovo ThinkPad X1 Carbon 7th / X1 Yoga 4th Embedded Controller Firmware Version 1.20
│ │     
│ │     The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│ │     
│ │     Problem fixes
│ │     
│ │      • Fixed an issue where ThinkVision T24m-10 monitor might not connected properly.
│ │   
│ └─ThinkPad X1 Carbon 7th / X1 Yoga 4th (W-BIOS for Machine types: 20QF, 20QG, 20QD, 20QE) Embedded Controller Update:
│       New version:       0.1.19
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad X1 Carbon 7th / X1 Yoga 4th Embedded Controller Firmware
│       License:           Proprietary
│       Size:              767.2 kB
│       Created:           2016-07-08
│       Urgency:           High
│       Vendor:            Lenovo Ltd.
│       Flags:             is-upgrade
│       Description:       Lenovo ThinkPad X1 Carbon 7th / X1 Yoga 4th Embedded Controller Firmware Version 1.19
│       
│       The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│       
│       Problem fixes
│       
│        • Fixed an issue where Accelerometer initialization might fail just after ECP update.
│     
├─Intel Management Engine:
│ │   Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│ │   Current version:     192.47.1524
│ │   Minimum Version:     192.47.1524
│ │   Vendor:              DMI:LENOVO
│ │   Update State:        success
│ │   GUID:                c3e4be53-e714-4ea1-bb9c-7fe13a98b556
[... 214 lines cut ...]
20QDCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…********************************]
Scheduling…              [***************************************]
Successfully installed firmware
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
Upgrade available for System Firmware from 0.1.33 to 0.1.41
20QDCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[***************************************]
Scheduling…              [***************************************]
Successfully installed firmware
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates

An update requires a reboot to complete. Restart now? [y|N]: 

Is that a know issue and can I give you any more information to fix this?
Thanks!

orclex, 2020-11-19:
On reboot I see for a few seconds a blue screen with "Boot Option Restored" in title bar. If I press a key to break countdown I can choose between "Reset system", "Continue boot" and "Always continue boot". I've tried both first choices. For both firmware updates are available again on next boot and so the upgrade seems never be successful.

The system is (K)ubuntu 20.10 with fwupd 1.4.5-1.

kmauleon, 2020-11-19:
hi @orclex are you referring to MEFW update?  Can you please check MEFW version in BIOS Setup Menu and confirmed that it was actually updated?  This is a known issue for MEFW that ESRT table was not updated, thus, fwupd still recognize the old version. We plan to release new MEFW by December.  Please wait for that and let's see if behavior will change.  Thank you very much...

orclex, 2020-11-20:
It looks like it is still the old version. In UEFI setup there is no leading zero for the version number... 
I can see these versions:
```
UEFI BIOS Version: N2HET50W (1.33)
UEFI BIOS Date:     2020-05-13
Emedded Controller Version:  N2HHT34W (1.17)
ME Firmware Version: 12.0.49.1534
Machine Type Model:  20QDCTO1WW
```

kmauleon, 2020-11-23:
hi @orclex LVFS site available versions are [12.0.47.1524](https://fwupd.org/downloads/32100a49cc474b16c9534a1a2963128d6e84bf5b-Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-CorporateMEFirmware-12.0.47.1524.cab) > [12.0.64.1551](https://fwupd.org/downloads/1bf0689e23de19521c00e56794adc04bcfe5da26311afefce5f8dcf70eae8a0a-Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-CorporateMEFirmware-12.0.64.1551.cab) > [12.0.68.1606](https://fwupd.org/downloads/78cd8451526e66ea2dabc37a2ac5ba72ea475d8be54118823baed8ef931a7619-Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-CorporateMEFirmware-12.0.68.1606.cab)
may I know how were you able to update to 12.0.49.1534?
can you please show results of below command... thank you very much
**fwupdmgr get-devices**

orclex, 2020-11-24:
```
# fwupdmgr get-devices
20QDCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           bce920702debbc6f20c5c4a5dd83cb588b0e333d
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               74cdddba-f7b6-5a31-a3dc-64186b5a5491 ← THUNDERBOLT\VEN_0109&DEV_1801&REV_00
│                          18bb00e3-5d7a-55d3-831e-0e000c184b0e ← THUNDERBOLT\VEN_0109&DEV_1801
│                          c7920601-0cda-507f-851e-92229eb1d370 ← TBT-01091801-native
│                          96c6512b-0138-51da-b02f-f44f2c4c7744 ← TBT-01091801-native-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:     0.1.17
│     Minimum Version:     0.1.17
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     Last modified:       2020-11-20 16:35
│     GUIDs:               f72e048b-65bd-4e71-9071-1ac7045223e5
│                          9cd54062-d764-51a7-a780-51a3695beae2 ← UEFI\RES_{F72E048B-65BD-4E71-9071-1AC7045223E5}
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:     192.47.1524
│     Minimum Version:     192.47.1524
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     Last modified:       2020-11-20 16:35
│     GUIDs:               c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│                          35cbf8ae-bee1-52fa-bf0d-11f1fc4c12f5 ← UEFI\RES_{C3E4BE53-E714-4EA1-BB9C-7FE13A98B556}
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─SAMSUNG MZVLB1T0HBLR-000L7:
│     Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:             NVM Express Solid State Drive
│     Current version:     4M2QEXF7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S4EMNF0MC40413
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                          b77af583-f45c-537c-8584-fc5d1c80fa20 ← SAMSUNG MZVLB1T0HBLR-000L7
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f
│     Current version:     0.1.33
│     Minimum Version:     0.0.1
[... 36 lines cut ...]
├─UEFI Device Firmware:
│     Device ID:           f61a138e8240c473a88d703babf7be5bc0e098e6
│     Current version:     4784136
│     Minimum Version:     4784136
│     Vendor:              DMI:LENOVO
│     GUID:                db246b7d-a882-4e0e-984f-23ac1b0270db
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           97d441c3228fe8615f883b166a8809389803f9e2
      Current version:     26406
      Minimum Version:     1
      Vendor:              DMI:LENOVO
      GUID:                68d3b08f-6ed2-43ca-86f0-f1ad6e41b390
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

```

kmauleon, 2020-11-25:
hi @orclex can you please try to download this cab file and install as below.  Thank you very much.  >> 
[github87.zip](https://github.com/fwupd/firmware-lenovo-thinkpad/files/5593550/github87.zip)
(note: same as LVFS uploaded version but without MEFW version requirement)
1. extract zip file
2. navigate to the location of the cab file
3. install cab file as below
    fwupdmgr --allow-older --allow-reinstall install <cab file name>
4. restart the system
5. wait for updates to be completed
6. prior to system boot check BIOS Setup Menu MEFW version, should be 12.0.68.1606

orclex, 2020-11-25:
Thanks, I tried this but unfortunately the firmware is not installed.
On reboot I got the messages

```
Found update fwupd-ux-capsule
Found update fwupd-c3e4be53-e714-bb9c-7fe13a98b556-0
Adding new capsule
Adding new capsule
WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
start_image() returned Unsupported
```

I waited about 20min but nothing happend and I need to power off/power on to restart. Firmware is not updated and update is still available.
Any ideas?

hughsie, 2020-11-25:
I think you need a new system firmware with a new enough UpdateCapsule implementation. Updating your BIOS using windows or a LiveCD should allow this to apply.

orclex, 2020-11-25:
Unfortunately I don't have Windows. Will a new UpdateCapsule implementation delivered with the next LVFS update?

hughsie, 2020-11-25:
> Will a new UpdateCapsule implementation delivered with the next LVFS update

Unfortunately the thing that needs fixing is the update routine that's on your current system firmware, so the update never gets run and thus we can't actually load any code. I believe Lenovo offers a LiveCD download .iso option as an alternative for updating most system firmwares affected by this.

orclex, 2020-11-25:
>  I believe Lenovo offers a LiveCD download .iso option as an alternative for updating most system firmwares affected by this.

Good idea. I could install 2 of 3 updates but one is still missing.

I downloaded BIOS Update (Bootable CD)1.41 / 1.30
(https://download.lenovo.com/pccbbs/mobiles/n2hur34w.txt), booted it and I've selected _2. Update system program_.
It took a while with a blank screen until it flashed the bios and finally showed me that I have to disable **Secure RollBack Revention**. So I disabled it in BIOS and booted again the bootable ISO. Now pressing _2. Update system program_ it says 
`An update is not necessary at this time. The process has been canceled.`

I booted linux and checked with
```
fwupdmgr get-updates
fwupdmgr update
```

2 of 3 updates are installed but the one for ME is still left:
```

# fwupdmgr update
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
Upgrade available for Intel Management Engine from 192.47.1524 to 192.68.1606
20QDCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…********************************]
Scheduling…              [***************************************]
Successfully installed firmware
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
• System Firmware has the latest available firmware version
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates

An update requires a reboot to complete. Restart now? [y|N]:
```

After reboot I get the same error message as above:

```
Found update fwupd-ux-capsule
Found update fwupd-c3e4be53-e714-bb9c-7fe13a98b556-0
Adding new capsule
Adding new capsule
WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
start_image() returned Unsupported
```
 
The update is not installed. Again I tried
`# fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.68.1606-github87.cab`

But unfortunately on restart I still get the same unsupported error.

Any ideas?

kmauleon, 2020-11-25:
hi @orclex 
this is weird though having 2 GUIDs for MEFW
**│     GUIDs:               c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│                          35cbf8ae-bee1-52fa-bf0d-11f1fc4c12f5 ← UEFI\RES_{C3E4BE53-E714-4EA1-BB9C-7FE13A98B556}**

may I know how were you able to update to 12.0.49.1534?  LVFS does not have this version uploaded...
please advise.  Thank you very much.

kmauleon, 2020-11-25:
@orclex maybe you can try the following options as well if it'll work for you...
1. fw downgrade
     **fwupdmgr downgrade (choose which firmware to downgrade)
     fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.68.1606-github87.cab**
2. [uninstalling and reinstalling fwupdmgr](https://github.com/fwupd/firmware-lenovo-thinkpad/issues/70#issuecomment-729565648)
3. update BIOS/EC first via Gnome Software then after successful update, execute MEFW update via terminal
     **fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.68.1606-github87.cab**

hughsie, 2020-11-26:
>  this is weird though having 2 GUIDs for MEFW

Nah, newer fwupd versions use the "raw" GUID from the ESRT entry and *also* use the Microsoft `UEFI\RES_{}` format so we can use the same values as Windows Update -- which confuses some vendor _less_.

orclex, 2020-11-27:
Unfortunately there is no downgrade available:

```
# fwupdmgr downgrade 
Choose a device:
0.      Cancel
1.      bce920702debbc6f20c5c4a5dd83cb588b0e333d (Thunderbolt Controller)
2.      c085a3cbf4dc1d60d110b7616816e2ffd279b64e (Embedded Controller)
3.      0ec6a2d51c2c55a649247d119ffb8e229168d6c9 (Intel Management Engine)
4.      f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f (System Firmware)
5.      2bd025a21741529fac39d6f088b9debfee7d4c9a (Touchpad)
3
No downgrades for Intel Management Engine: current version is 192.47.1524: 192.68.1606=newer, 192.64.1551=newer, 192.47.1524=same
```
I wonder if I can download a previous update from lenovo?

https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-7th-gen-type-20qd-20qe/20qd/20qdcto1ww/pf1rvln0/downloads/ds540232-bios-update-utility-bootable-cd-for-linux-windows-10-64-bit-thinkpad-x1-carbon-7th-gen-x1-yoga-4th-gen

I downloaded
`https://download.lenovo.com/pccbbs/mobiles/n2hul30w.zip`
The zip file contains
```
N2HET55P.cab
N2HET55W.cab
N2HHT38W.cab
n2hul30w.zip
N2QET31P.cab
N2QET31W.cab
N2QHT19W.cab
```
but I'm not sure if one is working for me and which one I have to try. So until now I didn't try a downgrade to not get things worse.

I don't know exactly where I got 12.0.49.1534. I always use _fwupdmgr_ to update firmware. A half year ago I also had Windows installed and used the Windows Lenovo updater. I had critical problems with empty battery (although connected to power supply) and system always shutting down. Some months ago I was advised in a support ticket to update latest test update:
```
fwupdmgr enable-remote lvfs-testing
fwupdmgr get-devices
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr update
fwupdmgr disable-remote lvfs-testing
```
I updated only one time from lvfs-testing. Maybe the version came from lvs-testing or Windows updater? Since that update I run stable updates again....

kmauleon, 2020-12-01:
hi @orclex BIOS latest should be as below image... 
[image]

Please update to the following version
1.41      (N2HUL34W)  1.41 (**N2HET58W**)      1.21 (**N2HHT38W**)   01    2020/11/12

**fwupdmgr install** <cab file>

kmauleon, 2020-12-01:
update BIOS/EC as below command... thank you
     **fwupdmgr install <cab filename>**

kmauleon, 2020-12-01:
update BIOS/EC as below... thank you
     **fwupdmgr install** xxx.cab 
xxx >> cab filename

orclex, 2020-12-01:
Unfortunately it says it is already installed:

I downloaded **n2hul34w.zip** for linux, unzipped it and tried to install:

```
# fwupdmgr install N2HET58W.cab 
Decompressing…           [***************************************]
Specified firmware is already installed '0.1.41'
# fwupdmgr install N2HHT38W.cab
Decompressing…           [***************************************]
Specified firmware is already installed '0.1.21'
```

fwupdmgr still shows an available update which is also available again after upgrading and reboot:
```

# fwupdmgr get-updates && fwupdmgr update
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
• System Firmware has the latest available firmware version
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
20QDCTO1WW
│
└─Intel Management Engine:
  │   Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
  │   Current version:     192.47.1524
  │   Minimum Version:     192.47.1524
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   Last modified:       2020-11-27 11:41
  │   GUIDs:               c3e4be53-e714-4ea1-bb9c-7fe13a98b556
  │                        35cbf8ae-bee1-52fa-bf0d-11f1fc4c12f5 ← UEFI\RES_{C3E4BE53-E714-4EA1-BB9C-7FE13A98B556}
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  ├─ThinkPad X1 Yoga 4th/ThinkPad X1 Carbon 7th Consumer ME Update:
  │     New version:       192.68.1606
  │     Remote ID:         lvfs
  │     Summary:           Lenovo ThinkPad X1 Yoga 4th/X1 Carbon 7th Consumer ME Firmware
  │     License:           Proprietary
  │     Size:              3.0 MB
  │     Created:           2016-07-08
  │     Urgency:           High
  │     Details:           https://pcsupport.lenovo.com/de/en/search?query=N2HRN27WA
  │     Vendor:            Lenovo Ltd.
  │     Flags:             is-upgrade
  │     Description:       • 0 Intel Platform Update 2020.1 Hot Fix Release
  │     
  │     Version 12.0.68.1606 (LVFS: 192.68.1606)
  │     
  │     Problem Fixes
  │     
  │      • Mitigated security vulnerabilities.
  │     
  │     Security issues fixed:
  │     
  │      • CVE-2020-8758
  │      • INTEL-TA-00404
  │   
  └─ThinkPad X1 Yoga 4th/ThinkPad X1 Carbon 7th Consumer ME Update:
        New version:       192.64.1551
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Yoga 4th/X1 Carbon 7th Consumer ME Firmware
        License:           Proprietary
        Size:              3.0 MB
        Created:           2016-07-08
        Urgency:           High
        Details:           https://pcsupport.lenovo.com/de/en/search?query=N2HRN24WA
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       • 0 Intel Platform Update 2020.1 (Maintenance Release)
        
        Version 12.0.64.1551 (LVFS: 192.64.1551)
        
        Problem Fixes
        
         • BIOS post will be over 30 sec after G3 reset.After flashing image by fpt.efi -f BIOS.rom -me and fpt.efi -greset
         • Intel CSME PSIRT-TA-2019-10-001(IPU 2020.1)
         • Mitigated security vulnerabilities.
        
        Security issues fixed:
        
         • CVE-2020-0542
         • CVE-2020-0538
         • CVE-2020-0534
         • CVE-2020-0532
         • CVE-2020-0541
         • CVE-2020-0531
         • CVE-2020-0535
         • CVE-2020-0540
         • CVE-2020-0536
         • CVE-2020-0537
         • CVE-2020-0539
      
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
Upgrade available for Intel Management Engine from 192.47.1524 to 192.68.1606
20QDCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Unknown                  [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…                                ]
Scheduling…              [***************************************]
Successfully installed firmware
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
• System Firmware has the latest available firmware version
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates

An update requires a reboot to complete. Restart now? [y|N]: 
```

kmauleon, 2020-12-01:
hi @orclex can i have a screenshot of your BIOS Setup Menu again.... thank you very much

orclex, 2020-12-02:
[image]

kmauleon, 2020-12-02:
hi @orclex can you please try to manually install this cab file >> 
[Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.64.1551.zip](https://github.com/fwupd/firmware-lenovo-thinkpad/files/5632643/Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.64.1551.zip)
**fwupdmgr --allow-older --allow-reinstall install** xxx.cab

orclex, 2020-12-03:
Unfortunately this lets the system freeze as in [thread above](https://github.com/fwupd/firmware-lenovo-thinkpad/issues/87#issuecomment-733579647). I had to power off and on again.
[image]

kmauleon, 2020-12-03:
hi @orclex sorry but I did not encounter this error... asking help from BIOS team... I will update you once I got some information... Thank you very much

orclex, 2020-12-04:
Thanks to you for asking and helping!

kmauleon, 2020-12-16:
hi @orclex sorry but unfortunately BIOS team did not encounter this error as well.... is it possible for you to try the following commands again...
**fwupdmgr refresh
fwupdmgr update**

orclex, 2020-12-16:
This is bad news! 
Unfortunately a refresh && update does not change anything. On reboot it still stops with 
`WARNING: Could not apply capsules: Unsupported`
So what now?

kmauleon, 2020-12-16:
hi @orclex... somehow something broke with your last update.... but since your last MEFW update was with Windows or from different method (12.0.49.1534 was not release to LVFS), i'm afraid you might have to revert back to windows and upgrade from there.  i'm really sorry but i don't have any other options for now.  @hughsie please help if any... thank you.
or you may want to try with new version coming this december/early january to LVFS?  currently still under testing but i will let you know once uploaded... thank you very much.

hughsie, 2020-12-17:
> Could not apply capsules: Unsupported

This means the old system firmware update capsule implementation is broken. If you install a new system firmware either using Windows or a .iso image from Lenovo then you should be able to schedule future updates with fwupd.

orclex, 2020-12-17:
@hughsie  I already tried a .iso image for updating but unfortunately with no success (see above). I can wait until next firmware is released and will try it. Does it matter wheather I use LVFS or an iso image boot for this update?

kmauleon, 2020-12-17:
with this we may need to wait for new system firmware.  as i checked BIOS 1.41 can not be downgraded anymore. as @hughsie suggestion, for new system firmware, still need to update via Windows or ISO image

kmauleon, 2020-12-23:
hi @orclex latest 12.0.70.1652 is available already... can you please try with this?  thank you very much...
https://fwupd.org/downloads/e78d2c1d0372734379e135b5fe26bdf479642529ea5136036cffd1d5f355429e-Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.70.1652.cab

orclex, 2020-12-23:
Unfortunately this is not working:

# fwupdmgr install /usr/local/src/bios/Lenovo-ThinkPad-X1Yoga4thX1Carbon7th-ConsumerMEFirmware-12.0.70.1652.cab
Decompressing…           [***************************************]
Not compatible with firmware version 192.47.1524, requires >= 192.68.1606

kmauleon, 2020-12-23:
hi @orclex can you please try the following commands again... thank you very much...
**fwupdmgr refresh
fwupdmgr update**

orclex, 2020-12-23:
That looks good!

```
# fwupdmgr refresh
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 6 local devices supported
```
```
# fwupdmgr update
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• Intel Management Engine has no available firmware updates
• Prometheus has the latest available firmware version
• Prometheus IOTA Config has the latest available firmware version
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
• System Firmware has the latest available firmware version
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
```

Now my installed version is

```
# fwupdmgr get-devices
...
├─Intel Management Engine:
│     Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:     192.47.1524
│     Minimum Version:     192.47.1524
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
```

What has changed so that it was working now?

orclex, 2020-12-23:
However, thank you very much for your help! Happy Christmas!

kmauleon, 2020-12-24:
hi @orclex I removed the MEFW required version temporarily to be able to install the latest ^_^
however it seems that fwupd still detected the lower version.... to be sure can you please show screen shot of BIOS Setup Menu?
thank you very much... Merry Christmas!!  ^_^

orclex, 2021-01-05:
Thanks! Here is the screen shot of BIOS Setup Menu. Is the correct version installed?

[image]

kmauleon, 2021-01-06:
hi @orclex bad news... I don't think your MEFW was updated T_T
last try 'coz i'm running out of options here....

1. disable UEFI Secure Boot in BIOS
2. uninstall / reinstall fwupdmgr
3. execute fwupdmgr --force refresh
4. execute fwupdmgr update
5. if ok... capsule found will appear on the upper left of the screen after reboot
6. update will be processed (with Lenovo Logo)
7. please press any key when you see "Press any key" on the upper left most of the screen

hope this works... thank you very much...

orclex, 2021-01-06:
Unfortunately this does not work.
I disabled Secure Boot, purged fwupd and reinstalled it.
`fwupdmgr --force refresh; fwupdmgr update `
tried to update Firmware..on reboot it tried 2x "adding new capsule" but failed with an error message which I could not read because it was too quick. But I think it's like above "unsupported"?

Now I'm back with getting firmware updates available I cannot install:

```
# fwupdmgr get-updates
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
• System Firmware has the latest available firmware version
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
20QDCTO1WW
│
└─Intel Management Engine:
  │   Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
  │   Current version:     192.47.1524
  │   Minimum Version:     192.47.1524
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                c3e4be53-e714-4ea1-bb9c-7fe13a98b556
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  ├─ThinkPad X1 Yoga 4th/ThinkPad X1 Carbon 7th Consumer ME Update:
  │     New version:       192.70.1652
  │     Remote ID:         lvfs
  │     Summary:           Lenovo ThinkPad X1 Yoga 4th/X1 Carbon 7th Consumer ME Firmware
  │     License:           Proprietary
  │     Size:              3.0 MB
  │     Created:           2016-07-08
  │     Urgency:           High
  │     Details:           https://pcsupport.lenovo.com/de/en/search?query=N2HRN28W
  │     Vendor:            Lenovo Ltd.
  │     Flags:             is-upgrade
  │     Description:       • 0 Intel Platform Update 2020.2 Release
  │     
  │     Version 12.0.70.1652 (LVFS: 192.70.1652)
  │     
  │     Problem Fixes
  │     
  │      • Issue #22010596778: BIOS returns wrong value following an Intel CSME update to 12.0.64.1551
  │      • Mitigated the following security vulnerabilities under issues:
  │     
  │     Security issues fixed:
  │     
  │      • CVE-2020-8749
  │      • CVE-2020-8746
  │      • CVE-2020-12356
  │      • CVE-2020-8755
  │      • CVE-2020-8747
  │      • CVE-2020-8754
  │      • CVE-2020-8751
  │      • CVE-2020-8760
  │      • CVE-2020-8756
  │      • CVE-2020-8757
  │      • CVE-2020-12303
  │      • CVE-2020-8705
  │      • CVE-2020-8744
  │      • CVE-2020-8745
  │      • CVE-2020-1297
[... 58 lines cut ...]
         • CVE-2020-0537
         • CVE-2020-0539

# fwupdmgr update 
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
Upgrade available for Intel Management Engine from 192.47.1524 to 192.70.1652
20QDCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Unknown                  [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…                                ]
Scheduling…              [***************************************]
Successfully installed firmware
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
• System Firmware has the latest available firmware version
• Touchpad has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates

An update requires a reboot to complete. Restart now? [y|N]:

```
Any ideas? Thank you very much!

kmauleon, 2021-01-07:
hi @orclex let me get back to you... i will try if i can downgrade to your current ME version and update to latest....

kmauleon, 2021-01-07:
hi @orclex I was able to update MEFW successfully from **12.0.49.1534 to 12.0.70.1652 with Ubuntu 20.04**.  Please see attached images for your reference.  I'm not sure what could be the problem in your case.  Is there a possibility for you to reinstall windows and update from there (https://github.com/fwupd/fwupd/issues/1190#issuecomment-527468880)?  thank you very much...
[image]
[image]

kmauleon, 2021-01-08:
hi @orclex would it be possible to ask for a video clip? hopefully we could catch the error on video and will try to ask BIOS dev for help.... thank you very much....
please take a video following below procedure...
just disable Secure Boot and execute the following commands again
**fwupdmgr --force refresh
fwupdmgr update**

orclex, 2021-01-09:
Hi,
I did a
```
fwupdmgr --force refresh
fwupdmgr update
```
and recorded a short video as attached. You can see 

```
Found update fwupd-ux-capsule
Found update fwupd-c3e4be53-e714-4ea1-bb9c-7fe13a98b556-0
Adding new capsule
Adding new capsule
WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
```

Any ideas?
Maybe I can install Windows on a USB-hdd and boot from it and I'll also try updating from Windows.

orclex, 2021-01-09:
short video from not working firmware update:
https://owncloud.gwdg.de/index.php/s/YzHYxGBY22CG3Ed/download

kmauleon, 2021-01-11:
hi @orclex while waiting for BIOS dev reply, can you please try updating in windows?  thank you very much...

orclex, 2021-01-11:
Yesterday I tried installing Windows 10 20.04 to an external device but unfortunately the installer quits because of missing drivers for any found harddrive :-(

kmauleon, 2021-01-11:
ohhh... this is unfortunate... hopefully we can have support from BIOS dev....

kmauleon, 2021-02-04:
update from BIOS Dev... thank you very much....
**no clear progress yet regarding BIOS ESRT problem investigation**

orclex, 2021-04-27:
After upgrading to Kubuntu 21.04 and doing a `fwupdmgr get-updates && fwupdmgr update` the firmware could be upgraded:
```
# fwupdmgr get-updates && fwupdmgr update
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • Embedded Controller
 • Intel Management Engine
 • SAMSUNG MZVLB1T0HBLR-000L7
 • System Firmware
No updates available for remaining devices
```

```
# fwupdmgr get-devices
20QDCTO1WW
│
├─Thunderbolt host controller:
│     Device ID:          bce920702debbc6f20c5c4a5dd83cb588b0e333d
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    44.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              74cdddba-f7b6-5a31-a3dc-64186b5a5491 ← THUNDERBOLT\VEN_0109&DEV_1801&REV_00
│                         18bb00e3-5d7a-55d3-831e-0e000c184b0e ← THUNDERBOLT\VEN_0109&DEV_1801
│                         c7920601-0cda-507f-851e-92229eb1d370 ← TBT-01091801-native
│                         96c6512b-0138-51da-b02f-f44f2c4c7744 ← TBT-01091801-native-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     GUIDs:              f72e048b-65bd-4e71-9071-1ac7045223e5
│                         9cd54062-d764-51a7-a780-51a3695beae2 ← UEFI\RES_{F72E048B-65BD-4E71-9071-1AC7045223E5}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:    192.70.1652
│     Minimum Version:    192.70.1652
│     Vendor:             DMI:LENOVO
│     GUIDs:              c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│                         35cbf8ae-bee1-52fa-bf0d-11f1fc4c12f5 ← UEFI\RES_{C3E4BE53-E714-4EA1-BB9C-7FE13A98B556}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─SAMSUNG MZVLB1T0HBLR-000L7:
│     Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:            NVM Express Solid State Drive
│     Current version:    5M2QEXF7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:      S4EMNF0MC40413
[... 60 lines cut ...]
│     Vendor:             DMI:LENOVO
│     GUIDs:              db246b7d-a882-4e0e-984f-23ac1b0270db
│                         0253277e-3c60-502c-8423-74b9e15950b5 ← UEFI\RES_{DB246B7D-A882-4E0E-984F-23AC1B0270DB}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          97d441c3228fe8615f883b166a8809389803f9e2
      Current version:    26406
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      GUIDs:              68d3b08f-6ed2-43ca-86f0-f1ad6e41b390
                          64c2832d-137c-5b2f-af14-028dc25b659a ← UEFI\RES_{68D3B08F-6ED2-43CA-86F0-F1AD6E41B390}
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

Thank you all!
Best regards!

kmauleon, 2021-04-27:
> After upgrading to Kubuntu 21.04 and doing a `fwupdmgr get-updates && fwupdmgr update` the firmware could be upgraded:
>
> ```
> # fwupdmgr get-updates && fwupdmgr update
> Devices with no available [...]

Great!  Thank you very much for your update @orclex ... closing the issue...

closed 2021-04-27 by kmauleon