fwupd/firmware-lenovo #89: T460s cannot update intel ME

dkwo, 2020-11-26:
On my T460s with ME permanently disabled in BIOS, fwupd still suggests the update, and if I accept it and reboot, the system BIOS sees it and applies it, but when I run fwupd again after reboot, it says it was unable to apply ME update.

Shouldn't fwupd detect that the ME is disabled and just skip the update? or am I supposed to apply it anyway? in this case, why is it not working?

```
$ fwupdmgr --version
client version: 1.5.2
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.2
```
```
$ fwupdmgr refresh
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 3 local devices supported
```
```
$ fwupdmgr get-updates
Devices with no available firmware updates:
 • Lenovo ThinkPad Dock
 • Lenovo ThinkPad Dock
 • Intuos S
 • Intuos S [Legacy Bluetooth Module]
 • SAMSUNG MZVKV512HAJH-000L1
 • VMM2322
Devices with the latest available firmware version:
 • Embedded Controller
 • System Firmware
20F9CTO1WW
│
└─Intel Management Engine:
  │   Device ID:          25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f
  │   Current version:    184.70.3626
  │   Minimum Version:    184.70.3626
  │   Vendor:             DMI:LENOVO
  │   GUIDs:              63451986-0c42-42db-9a18-342a8455ac68
  │                       0e559cf4-f432-5cd4-9611-bd774afb417a
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │
  ├─ThinkPad T460s Corporate ME Update:
  │     New version:      184.79.3722
  │     Remote ID:        lvfs
  │     Summary:          Lenovo ThinkPad T460s Corporate ME Firmware
  │     License:          Proprietary
  │     Size:             7.5 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N1CRG64W
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Description:
  │     • 8 Intel Platform Update 2020.1 Hot Fix Release
  │
  │     Version 11.8.79.3722 (LVFS: 184.79.3722)
  │
  │     Problem Fixes
  │
  │     • Mitigated security vulnerabilities.
  │     • IntelR ME SW 2020.14.0.1604 fails to install IntelR DAL and IntelR SOL with error gno driver was found
  │
  │     Security issues fixed:
  │
  │     • CVE-2020-8758
  │
  └─ThinkPad T460s Corporate ME Update:
        New version:      184.77.3664
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T460s Corporate ME Firmware
        License:          Proprietary
        Size:             7.5 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N1CRG63W
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:
        • 8 Intel Platform Update 2020.1

        Version 11.8.77.3664 (LVFS: 184.77.3664)

        Problem Fixes

        • Intel CSME PSIRT-TA-2019-10-001 (IPU 2020.1):
        • Mitigated security vulnerabilities.

        Security issues fixed:

        • CVE-2020-0531
        • CVE-2020-0532
        • CVE-2020-0533
        • CVE-2020-0534
        • CVE-2020-0535
        • CVE-2020-0536
        • CVE-2020-0537
        • CVE-2020-0538
        • CVE-2020-0539
        • CVE-2020-0540
        • CVE-2020-0541
        • CVE-2020-0542
        • CVE-2020-0545
```
```
$ sudo fwupdmgr upgrade
Devices with no available firmware updates:
 • Lenovo ThinkPad Dock
 • Lenovo ThinkPad Dock
 • Intuos S
 • Intuos S [Legacy Bluetooth Module]
 • SAMSUNG MZVKV512HAJH-000L1
 • VMM2322
Devices with the latest available firmware version:
 • Embedded Controller
Upgrade available for Intel Management Engine from 184.70.3626 to 184.79.3722
20F9CTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update
? [Y|n]:
Downloading…             [***************************************]
Successfully installed firmware
 • System Firmware
```
REBOOT
```
$ fwupdmgr get-updates
Devices with no available firmware updates:
 • Lenovo ThinkPad Dock
 • Lenovo ThinkPad Dock
 • Intuos S
 • Intuos S [Legacy Bluetooth Module]
 • SAMSUNG MZVKV512HAJH-000L1
 • VMM2322
Devices with the latest available firmware version:
 • Embedded Controller
 • System Firmware
________________________________________________

Devices that were not updated correctly:

 • Intel Management Engine (184.70.3626 → 184.79.3722)
```

Thanks.

kmauleon, 2020-11-26:
hi @dkwo can you please check MEFW version in BIOS Setup Menu?  Thank you very much..

dkwo, 2020-11-27:
Sure, it says: 11.8.79.3722

Thanks.

kmauleon, 2020-12-01:
hi @dkwo your MEFW is updated already to the latest... though fwupd is not seeing the correct version.  this is still under investigation by BIOS team, issue is similar with #65. let's keep this open for tracking... thank you very much.

dkwo, 2020-12-01:
Sure, thank you.

kmauleon, 2020-12-23:
hi @dkwo can you please try with latest 11.8.80.3746 thank you!
https://fwupd.org/downloads/5668a0bc71f266ea6d0ee9283995c4503dd8e4c5fd0922840b422575b2b2f83b-Lenovo-ThinkPad-T460s-CorporateMEFirmware-11.8.80.3746.cab

dkwo, 2020-12-23:
Hi, I get the same behavior.

sudo fwupdmgr update
Devices with the latest available firmware version:
  • Embedded Controller
Upgrade available for Intel Management Engine from 184.70.3626 to 
184.80.3746
20F9CTO1WW must remain plugged into a power source for the duration of 
the update to avoid damage. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Successfully installed firmware
  • System Firmware

Perform the update.
BIOS now reports: 11.8.80.3746

fwupdmgr get-updates
Devices that were not updated correctly:

  • Intel Management Engine (184.70.3626 → 184.80.3746)

Uploading firmware reports helps hardware vendors to quickly identify 
failing and successful updates on real devices.

Thanks.

kmauleon, 2020-12-23:
hi @dkwo I will forward this result to BIOS team for checking.  Thank you very much for reporting.

sjzaib, 2021-01-30:
I am seeing same issue with my T580, and @kmauleon said when i checked ME version in bios its indeed successfully updated 184.80.3746 but fwupdmgr is still showing the old version (184.70.3626) and therefore my ubuntu 18.04 is prompting me to upgrade firmware on every reboot.

rfrackowiak, 2021-01-30:
I've got the same issue with my T480s. ME in BIOS states 184.80.3746, fwupdmgr shows:
"Upgrade available for Intel Management Engine from 184.77.3664 to 184.80.3746"

alfredorico, 2021-01-30:
The same for Thinkpad X280.. With every reboot, wizard for installing ME update to 184.80.3746 is showing, besides of reinstalling over and over again.

kmauleon, 2021-02-01:
very sorry for the slow response but we have not received any feedback yet regarding the investigation... will update once i received any information.... thank you very much....

kmauleon, 2021-02-04:
update from BIOS Dev... thank you very much....
**no clear progress yet regarding BIOS ESRT problem investigation**

kcrmson, 2021-02-20:
ThinkPad P50 here, my situation was similar to everyone else's until today, now it's a bit different.
Normally the same *.3746 IME update would show, it looked like it would install but would not (verified in the BIOS) and still always show as being an available update.

I decided to try and do the firmware upgrade in Windows, which I no longer had installed. Installing that was an adventure in finding out that I had to make sure none of my Linux partitions (on my two internal NVMe SSDs) weren't present, so I removed them and successfully installed W10. I ran all the updates, ran all the Lenovo Commercial Vantage updates to eek out the IME update. It installed it successfully as verified in the BIOS (version *.3838 now, newer than what's available in LVFS).

But now the same *.3746 update still shows as being available via fwupdmgr and trying to update that (I have rollbacks enabled, nothing is restricted in the BIOS) and it will download, reboot but immediately boot to systemd-boot and boot normally. fwupdmgr still shows the *.3746 update being available, I'm guessing because *.3838 isn't in LVFS yet.

If any other information could be useful, please let me know. Hoping my situation helps out the BIOS devs.

kmauleon, 2021-02-23:
hi @kcrmson your MEFW is already the latest.  though .3838 was not uploaded to LVFS.  "fwupdmgr reporting older version" is a known problem and still under investigation by BIOS dev (BIOS ESRT problem). thank you very much.

kmauleon, 2021-05-26:
hi latest 11.8.83.3874 (184.83.3874) has been put to stable... thank you very much....

kmauleon, 2021-05-26:
hi anyone ok to provide logs for BIOS ESRT problem?  This will need to run some tool via EFI Shell... please ping me... thank you very much...

dkwo, 2021-05-26:
I updated ME, but I still get the same behaviour in fwupd (it does not realise update has taken place).
I'm on 1.5.8 though, will try to update later.

On the other hand, I also updated firmware 50->51, and this erased my efi partition, so I had to reinstall grub from chroot. I believe this is the 2nd time happening: do you know who's mother I have to blame?

kmauleon, 2021-05-27:
hi @dkwo can you please contact me at kmauleon@lenovo.com  ... i would like to send you a tool if you would permit... as for the BIOS firmware, i will try to ask our BIOS dev regarding the issue... hoping to hear from you... thank you very much...

kmauleon, 2021-05-27:
hi @dkwo BIOS Dev reply regarding efi partition issue... thank you very much...
--------------------------------------------
Please try on following condition. We met this symptom before on **old version of GRUB** and performed BIOS Load Default, but it disappears by **updating GRUB/GRUB2**.

1) Install GRUB2 (2.04 or newer version). Or,
2) Do not perform BIOS Load Default.

dkwo, 2021-05-27:
I have grub 2.04 already. could it just be https://github.com/fwupd/fwupd/issues/3186 ?

dkwo, 2021-05-27:
I'll write you now.

kmauleon, 2021-05-28:
> I have grub 2.04 already. could it just be [fwupd/fwupd#3186](https://github.com/fwupd/fwupd/issues/3186) ?

this could be it... but not an expert to confirm ^_^

kmauleon, 2021-09-03:
hi @dkwo for ESRT issue, BIOS dev can not investigate further without recreating the issue on our side... not sure how to proceed with this.  @mrhpearson @hughsie any suggestion?

dkwo, 2021-09-03:
@kmauleon Sure, can't you just grab a thinkpad machine from lenovo and recreate the issue?

kmauleon, 2021-09-03:
hi @dkwo I've been testing 9 different systems with 4 different Linux OS and still can not recreate the issue.  I was able to ask some assistance with a few customers to help me recreate the issue using their scenario and procedures but still no luck in recreating the issue T_T

dkwo, 2021-09-03:
Hi @kmauleon , I see. Is any of your machines a T460s? would you be able to try with void linux os, the x86_64-musl variant?

kmauleon, 2021-09-03:
Yes, I already tried with Ubuntu, Debian, RHEL, Suse, Fedora, Arch but no luck

dkwo, 2021-09-03:
Did you test on T460s or not?

kmauleon, 2021-09-03:
Yes I did test T460s

dkwo, 2021-09-03:
Could you try with void linux (musl) there? https://voidlinux.org

mrhpearson, 2021-09-03:
Can I bounce the question back the other way - can you put a Live Fedora on your system and try that please?
Thanks!

dkwo, 2021-09-06:
On 9/3/21 4:53 PM, Mark Pearson wrote:
> Can I bounce the question back the other way - can you put a Live Fedora on your system and try that please?
> Thanks!
> 
> 

I do get the same exact result with Fedora live.

kmauleon, 2021-09-08:
hi @dkwo sorry i am not familiar with void linux but i tried to install it following some video from youtube and i was able to install void xfce but unfortunately i can't make fwupd to work... received UEFI ESP warning

dkwo, 2021-09-08:
@kmauleon I'm assuming you installed on bare metal, not some virtual machine etc.
I do `mount -t efivarfs efivarfs /sys/firmware/efi/efivars` before runnig fwupd, can you do it?

kmauleon, 2021-09-10:
@dkwo sorry for the delay as we have no internet yesterday... anyway, it says already mounted on /sys/firmware/efi/efivars and with the same warning error UEFI ESP partition not detected or configured

dkwo, 2021-09-10:
Can you follow this instead?
https://docs.voidlinux.org/installation/live-images/guide.html

kmauleon, 2021-09-10:
@dkwo i think the instruction is the same with the one i followed in youtube.  by the way, this is the iso i tried >> void-live-x86_64-musl-20210218-xfce.iso

dkwo, 2021-09-10:
https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found
looked here?

kmauleon, 2021-09-15:
@dkwo still can't resolve the issue with void linux... by the way, others resolve their issue by doing this >> https://github.com/fwupd/firmware-lenovo-thinkpad/issues/70#issuecomment-919238383
please try.  thank you very much

dkwo, 2021-09-15:
Doesn't work for me, I still get
```
# fwupdmgr refresh
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 3 local devices supported
# fwupdmgr update
Devices with no available firmware updates:
 • Lenovo ThinkPad Dock
 • Lenovo ThinkPad Dock
 • SAMSUNG MZVKV512HAJH-000L1
 • VMM2322
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.70.3626 to 184.86.3909?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 8 Intel Platform Update 2021.1 Release                                     ║
║ Version 11.8.86.3909 (LVFS: 184.86.3909)                                     ║
║ Problem Fixes                                                                ║
║ • Mitigated the following security vulnerabilities under issues. Please      ║
║ see fixed issues for details.                                                ║
║ 20F9CTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: n
Operation was cancelled
```

kmauleon, 2021-09-15:
not sure if it will make a difference but please try with "sudo"
**sudo** fwupdmgr refresh **--force**
**sudo** fwupdmgr update

dkwo, 2021-09-15:
I do get the same result as above.

kmauleon, 2021-09-15:
do you also have windows installed in your system? can you please try to update thru windows?

dkwo, 2021-09-15:
Nope.

kmauleon, 2021-09-17:
hi @dkwo are you familiar with this error...
[image]

dkwo, 2021-09-17:
It seems you did not partition correctly.
Can you show output of `cfdisk` on your diks?
This is what you expect
```
 Disk: /dev/nvme0n1
                                        Size: 476.94 GiB, 512110190592 bytes, 1000215216 sectors
                                      Label: gpt, identifier: 7A25233E-506D-4E75-B8BD-6C7847C0DF1E

    Device                                 Start                   End              Sectors              Size Type
>>  /dev/nvme0n1p1                          2048               1050623              1048576              512M EFI System
    /dev/nvme0n1p2                       1050624            1000215182            999164559            476.4G Linux filesystem
```
in particular
```
 Partition name: EFI System                                                                                                         │
 │ Partition UUID: 4A04D763-6E6C-4F50-A11B-C5D1092CCEE5                                                                               │
 │ Partition type: EFI System (C12A7328-F81F-11D2-BA4B-00A0C93EC93B)                                                                  │
 │Filesystem UUID: 28DC-1361                                                                                                          │
 │     Filesystem: vfat                                                                                                               │
 │     Mountpoint: /boot (mounted)
```
If you don't have anything like that, then please follow this https://docs.voidlinux.org/installation/live-images/partitions.html (not some youtube video).

kmauleon, 2021-09-20:
hi @dkwo i successfully updated mefw for t460s with void linux.  please see below images.  thank you very much.
[image]
[image]
[image]

dkwo, 2021-09-20:
@kmauleon Does it persist even after reboot and `fwupdmgr get-updates`?

kmauleon, 2021-09-21:
did not encounter problem even after reboot and fwupdmgr get-updates...
[image]

dkwo, 2021-09-21:
Ok, then it is not likely a void linux or other distro issue.

mrhpearson, 2021-09-21:
Apologies if this has been asked before - just want to confirm ESRT table is incorrect. Can you do:
cat /sys/firmware/efi/esrt/entries/entry?/fw_version 
Thanks

dkwo, 2021-09-22:
@mrhpearson Does this help?
```
$ doas cat /sys/firmware/efi/esrt/entries/entry0/fw_version
65587
$ doas cat /sys/firmware/efi/esrt/entries/entry1/fw_version
3091598890
$ doas cat /sys/firmware/efi/esrt/entries/entry2/fw_version
65550
```

mrhpearson, 2021-09-22:
Yep - thanks. It confirms the ESRT table is wrong (as expected) - 3091598890 -> 184.70.3626

@kmauleon - let's discuss internally but the BIOS team **have** to understand that they're doing something wrong here. I know they can't reproduce it but we have enough of these cases that somewhere their implementation is flawed.

CHBMB, 2021-09-26:
Just chiming in to confirm this is affecting my Lenovo T470s running Pop_OS! 21.04 

```
chbmb@thinkpad:~$ fwupdmgr --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.11
```
```
chbmb@thinkpad:~$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 5 local devices supported
```
```
chbmb@thinkpad:~$ fwupdmgr get-devices --show-all-devices
20HGS2HT00
│
├─ThinkPad Thunderbolt 3 Dock:
│ │   Device ID:          a7eeadb04e19aebdba4ec0e1996049194cd17aee
│ │   Current version:    62.00
│ │   Vendor:             Lenovo (THUNDERBOLT:0x0108, TBT:0x0108)
│ │   GUIDs:              45c1814e-eb83-5b03-8997-ef8d4d8ade29
│ │                       03b97e38-f4b8-5a19-917c-28f95ca29768
│ │                       75039175-c354-5df9-a798-5da41461c007
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │ 
│ ├─USB3.1 Hub:
│ │ │   Device ID:        7e452eb6c025449c23b67d34f76de2b1fdf4c815
│ │ │   Summary:          USB 3.x Hub
│ │ │   Current version:  51.63
│ │ │   Vendor:           VIA Labs, Inc. (USB:0x17EF)
│ │ │   Install Duration: 15 seconds
│ │ │   GUIDs:            8ee94f0e-9b44-596a-bdd9-6f90401664cc
│ │ │                     82aae1d4-67be-54c0-8256-ed2a7108650b
│ │ │                     8aab7e9b-2ae8-559b-9be0-49bf5a647a2a
│ │ │                     a9866465-e1ec-5ce0-855d-162d817aeca2
│ │ │                     06fb2b19-690e-5296-8d0a-6e32a6fccd84
│ │ │   Device Flags:     • Updatable
│ │ │                     • Cryptographic hash verification is available
│ │ │                     • Device stages updates
│ │ │                     • Device can recover flash failures
│ │ │ 
│ │ ├─USB3.1 Hub:
│ │ │ │   Device ID:      e487b5ff307494d6ff44d1b6dc1312bb69b0fe90
│ │ │ │   Summary:        USB 3.x Hub
│ │ │ │   Current version:51.33
│ │ │ │   Vendor:         VIA Labs, Inc. (USB:0x17EF)
│ │ │ │   Install Duration:15 seconds
│ │ │ │   GUIDs:          8ee94f0e-9b44-596a-bdd9-6f90401664cc
│ │ │ │                   3e9ec9de-6c88-520b-b768-d3839446d508
│ │ │ │                   8aab7e9b-2ae8-559b-9be0-49bf5a647a2a
│ │ │ │                   672d37eb-2a8c-54fd-88f2-0faecea5f5e6
│ │ │ │                   06fb2b19-690e-5296-8d0a-6e32a6fccd84
│ │ │ │   Device Flags:   • Updatable
│ │ │ │                   • Cryptographic hash verification is available
│ │ │ │                   • Device stages updates
│ │ │ │                   • Device can recover flash failures
│ │ │ │ 
│ │ │ └─ThinkPad Thunderbolt 3 Dock USB Audio:
│ │ │       Device ID:    8fa860edf16baebf86cc3aaa5ce677003d5b9118
│ │ │       Summary:      CX20985 USB audio device
│ │ │       Current version:04-0E-92
│ │ │       Bootloader Version:02.00.00.00
│ │ │       Vendor:       Lenovo (USB:0x17EF)
│ │ │       Install Duration:3 seconds
[... 276 lines cut ...]
  │     Version 11.8.82.3838 (LVFS: 184.82.3838)
  │   
  └─ThinkPad T470s Consumer ME Update:
        New version:      184.80.3746
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T470s Consumer ME Firmware
        Licence:          Proprietary
        Size:             2.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N1WRN26W
        Vendor:           Lenovo
        Flags:            is-upgrade
        Description:      
        Version 11.8.80.3746 (LVFS: 184.80.3746)
```
```
chbmb@thinkpad:~$ sudo cat /sys/firmware/efi/esrt/entries/entry?/fw_version
[sudo] password for chbmb: 
65578
3092057680
65558
4653068
```
Entering the BIOS at boot does indeed confirm that I actually do have version `184.80.3746` of the Intel Management Engine installed.

pieska, 2021-11-10:
just want to confirm the same issue with my T460s, BIOS says 184.86.3909, fwupmgr 184.80.3746.

# fwupdmgr --version
client version:	1.7.1
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.1

# cat /sys/firmware/efi/esrt/entries/entry?/fw_version
65589
3092254370
65550

akaWolf, 2021-12-20:
hello.
I decide not to open a new issue.
but at my X1 Carbon Gen 6 I have the same issue:
incorrect fw_version showing in userspace.
current installed version: 184.86.3909
showing in userspace: 184.70.3626
feel free to ask me for additional debug info.
cc @mrhpearson

deimosfr, 2022-01-27:
Hi!

Same for me with X1 Extreme Gen2

Thanks

dkwo, 2022-03-21:
@kmauleon Any progress on this? I can see at least 3 issues related to firmware and efi variables on the t460s: this one, https://github.com/Foxboron/sbctl/issues/102 and https://github.com/fwupd/firmware-lenovo/issues/209
Perhaps, they could benefit being addressed simutaneously.

mrhpearson, 2022-03-21:
We have a bunch of open tickets against this issue :(
The FW team think they have a fix, that they will have in the next BIOS release, but we're not 100% sure. We can't reproduce the issue internally and the ME FW is not allowed to be downgraded for security reasons so at the moment there is some guesswork involved.
I've asked the FW team to co-ordinate so we can get a new BIOS released _before_ we release a new ME FW - hopefully that happens and then we should know if the fix worked.
Mark

kmauleon, 2022-03-22:
Yes @dkwo BIOS dev is already looking at this... unfortunately as Mark mentioned we can't recreate this... let's just hope their hunch will work.... thank you very much....

[18 more replies]

closed 2022-11-10 by dkwo