fwupd/firmware-lenovo #90: Intel ME firmware update on X390 spews debug messages

mgedmin, 2020-12-01:
Here's what it looks like:
[image]

mgedmin, 2020-12-01:
fwupdmgr described the update as `ThinkPad X390/T490s Corporate ME Update` for the `Intel Management Engine` on `20Q0CTO1WW`.  The new version of the firmware was `192.71.1681` although the description also showed a version number as `Version 12.0.71.1681(LVFS:192.71.1681)`.

kmauleon, 2020-12-02:
hi @mgedmin ... i will forward your concern to owner... will get back once i have update from them... thank you very much.

kmauleon, 2020-12-07:
hi @mgedmin ... can you please try newly uploaded cab file >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRM.firmware
thank you very much...

mgedmin, 2020-12-07:
> can you please try newly uploaded cab file

I have no idea how to do that.

`fwupdmgr get-upgrades` finds no available firmware updates.

kmauleon, 2020-12-07:
@mgedmin can you please download the cab file from this link >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRM.firmware
then run the following command from terminal
**fwupdmgr --allow-older --allow-reinstall install** _cab filename_

mgedmin, 2020-12-09:
```
$ sudo fwupdmgr --allow-older --allow-reinstall install 6e5b1f928ee89e1ad54bce82a102c08d7080a0fb072e70f1b1bc631de42bfa62-Lenovo-ThinkPad-X390T490s-CorporateMEFirmware-12.0.71.1681.cab
No supported devices found
```

(Unrelated, but probably something for @hughsie to fix -- Firefox offered me to open the .cab file in w3m, because it's being served with `Content-Type: text/html; charset=utf-8`.  There's also no progress indication for the download, because the web server doesn't report the file size.)

kmauleon, 2020-12-09:
hi @mgedmin for "No supported devices found" please show results of the following command...
**fwupdmgr get-devices**

mgedmin, 2020-12-09:
The IME is

```
├─Intel Management Engine:
│     Device ID:           07edea1cf12d20814225b44de22b33758db1a1c8
│     Current version:     192.71.1681
│     Minimum Version:     192.71.1681
│     Vendor:              DMI:LENOVO
│     GUID:                ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
```

and here's the full list:

<details>

```console
mg@blynas! ~ $  fwupdmgr get-devices
20Q0CTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           8685856be84028011ec101dbce36216f842bd287
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               ea3c7ac8-d937-568a-aab5-e053bd6af9ce
│                          3d3f953a-3408-56fa-81f3-042d73c59c16
│                          e773c51e-a20c-5b29-9f09-6bb0e0ef7560
│                          f1793149-6c12-5a7f-95db-9fefd116748d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           09ce7908b2d90a3b53473c3f328410590a527f29
│     Current version:     0.1.15
│     Minimum Version:     0.1.15
│     Vendor:              DMI:LENOVO
│     GUID:                be9a5f93-930b-40bf-8864-815268ea4ea3
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           07edea1cf12d20814225b44de22b33758db1a1c8
│     Current version:     192.71.1681
│     Minimum Version:     192.71.1681
│     Vendor:              DMI:LENOVO
│     GUID:                ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─KXG6AZNV512G TOSHIBA:
│     Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:             NVM Express Solid State Drive
│     Current version:     5107AGLA
[... 45 lines cut ...]
├─UEFI Device Firmware:
│     Device ID:           2ea1b362db45581c8be7dc1094a973e2a0b45fec
│     Current version:     16842753
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                815e2465-da45-4b49-93c3-1517e649e1e6
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           b028a468cd8bdc225e88bdb99988cd7646b2687f
      Current version:     4784192
      Minimum Version:     4784192
      Vendor:              DMI:LENOVO
      GUID:                ee246b7d-a882-4e0e-984f-23ac1b0270db
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```
</details>

kmauleon, 2020-12-10:
hi @mgedmin , your system shows that MEFW was already updated to the latest 192.71.1681... but let me confirm the error with the owner... thank you very much...

kmauleon, 2020-12-10:
can you please try this instead...thank you
**fwupdmgr refresh
fwupdmgr update**

kmauleon, 2020-12-10:
hi @mgedmin .. your system MEFW GUID indicates a Non-VPro system, thus you need a consumer version for this as per owner.
can you please download the cab file from this link >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRN.firmware
then run the following command from terminal
**fwupdmgr --allow-older --allow-reinstall install** _cab filename_

mgedmin, 2020-12-10:
> hi @mgedmin , your system shows that MEFW was already updated to the latest 192.71.1681...

Yes, the update process was successful.  This issue is cosmetic: all the DEBUG messages that should not have shown up during the update.

fwupdmgr update finds a firmware update for the NVME SSD, which I think is unrelated.

Consumer version sounds right.

fwupdmgr --allow-reinstall install 83f9b282b96ca6e6de26d9509b0ad39dcfa89e677a7f01cd3a76e1f2fff2e5b1-Lenovo-ThinkPad-X390T490s-ConsumeMEFirmware-12.0.71.1681.cab

did its thing, so now I'll reboot and see how it goes

mgedmin, 2020-12-10:
The upgrade went smoothly, no debug output:

[image]

Nicely translated description.  It's a bit strange that the text spinner (`|` `/` `-` `\`) overlaps the LENOVO logo, but every previous firmware update looked like that too.  (Well, some had a progress bar below the text.)

I would say the bug is fixed.

kmauleon, 2020-12-11:
hi @mgedmin thank you very much for the good news... with this, i will close this issue now.  please reopen if you don't agree.... thank you very much ^_^

closed 2020-12-11 by kmauleon