fwupd/firmware-lenovo #91: Lenovo dock Synaptics VMM2322 Display Port HUB firmware is missing for Thinkpad Pro Dock

mscherer, 2020-12-01:
I have a Thinkpad Pro Dock but despite using the same protocol as the regular Thinkpad Workstation Dock (cf https://download.lenovo.com/pccbbs/mobiles/fwdphb07.txt ), it doesn't seems to be supported by fwupd on Fedora 33, likely because there is a quirk for the latter, and not the former, cf https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/synaptics-mst.quirk#L42 

I wrote a quick python script to read the vendor ID, and I can confirm that indeed, /dev/drm_dp_aux2 return the right same value as  https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/fu-synaptics-mst-device.c#L1022 (so synaptics vendor id).

However, I am unsure on what is needed to add the quirk, and how to easily get the board id.

hughsie, 2020-12-02:
You don't need a quirk; `ParentGuid` is just setting up the logical mapping so that it appears as a composite "tree" rather than lots of individual discrete devices.

It should enumerate fine if you do something like `sudo fwupdtool --plugins synaptics_mst get-devices --verbose` -- although without firmware uploaded by Lenovo (or rather, it's ODM) it's not going to be easy to flash firmware. If you have the raw file (not the .cab) in your hands, you can use `fwupdtool install-blob` although _caveat emptor_.

mscherer, 2020-12-02:
So, using --verbose flag, I see that error message:
```
09:46:40:0174 FuPlugin             udev_device_added(synaptics_mst)
09:46:40:0174 FuDevice             using 585c346c009da3cc19a400fb59a4ed8d0cbd0151 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux2
09:46:40:0197 FuPluginSynapticsMST no device found on drm_dp_aux2: invalid chip ID
```

from https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/fu-synaptics-mst-device.c#L1063-L1071 

And I confirm, it return 0x0,0x0 on that register, so I guess that's a different chipset, or a different way to get the ID.

But so, Thinkpad Workstation Dock is supported by the plugin, or I misunderstood the source code  ?

As for the firmware, it seems to be in the exe file. I guess my next step is to find a way to get it out, which is not trivial

hughsie, 2020-12-02:
> or a different way to get the ID

Possibly; I can ask Synaptics the next time I sync with that team.

> Thinkpad Workstation Dock is supported by the plugin

I can't really say; if there's public firmware on the LVFS then it's supported otherwise my lips have to stay sealed unfortunately.