fwupd/firmware-lenovo #94: Missing Thunderbolt and Dock Firmware for T480s attached to 40AC

jwesterdale-rh, 2020-12-14:
I have a T480s attached to a 40AC dock via Lenovo USB-C. Works mostly fine with 2x monitors currently over DisplayPort. Too often both go black at once, and then pop back on 5 Sec later.

Figure its time to flash the Thunderbolt host controller and Dock firmware.

I get this:  (TB host controller and dock fw are unavailable).

# fwupdmgr update
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
Devices with no available firmware updates:
 **• ThinkPad Thunderbolt 3 Dock**
 **• Thunderbolt host controller**
 • Integrated Webcam™
 • Samsung SSD 970 EVO Plus 500GB
 • VMM3320
Devices with the latest available firmware version:
 • Embedded Controller
 • System Firmware
 • UEFI dbx

For missing firmware please provide the following items:
Thunderbolt Host Controller: 

Version: 17.4.80.550
Date: 30 Nov 2020
Link to TB host controller (on T480s)
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480s-type-20l7-20l8/20l8/20l8s2n800/pc0xgy9w/downloads/ds502153-intel-thunderbolt-software-version-17x-for-windows-10-1709-and-later-thinkpad-t480-t480s-x1-carbon-gen-6
 
Dock: 
Version: V10023
Date: 15 Nov 2020
Link to TB3 dock FW (for dock):
https://pcsupport.lenovo.com/us/en/products/accessory/docks/thinkpad-thunderbolt-3-dock/40ac/40ac0135us/zbk0jyak/downloads/driver-list/component?name=Product%20Firmware

Link to the device page on LVFS:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22TF.firmware

Ref:  OS: Fedora 33

fwupdmgr --version
client version:	1.5.2
compile-time dependency versions
	gusb:	0.3.5
daemon version:	1.5.2

fwupdmgr get-topology --show-all
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
20L8S2N800
│
├─ThinkPad Thunderbolt 3 Dock:
│     Device ID:          7e8cb988e5ccbec864aa17f7755c8236db2be643
│     Current version:    33.00
│     Vendor:             Lenovo (TBT:0x0108)
│     GUIDs:              fab08d10-6524-5cb6-ab20-6d365c812f9e ← THUNDERBOLT\VEN_0108&DEV_1630&REV_00
│                         0f511908-5c9f-511e-9099-06823ec33614 ← THUNDERBOLT\VEN_0108&DEV_1630
│                         df675f10-53ac-59ca-bddd-0a86ee492920 ← TBT-01081630
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
│
├─Thunderbolt host controller:
│     Device ID:          4d2b831f1d8e7157c03d279f5924679f263c682c
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    12.02
│     Vendor:             Lenovo (TBT:0x0109)
│     GUIDs:              3849839e-d8b2-5941-9e69-5acc397d3ed3 ← THUNDERBOLT\VEN_0109&DEV_1708&REV_00
│                         7e3190a2-d5a5-5a9e-b582-494b47f7288c ← THUNDERBOLT\VEN_0109&DEV_1708
│                         3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc ← TBT-01091708
│                         6783c3fd-8932-5306-bb7f-4dfec968e44b ← TBT-01091708-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│

mrhpearson, 2021-10-15:
TBT3 WS G2 dock firmware is on LVFS now - I think this can be closed. Let me know if you're still seeing problems

closed 2021-10-15 by mrhpearson

eBug, 2022-10-19:
> Cant see the dock over TB3 cable ... have tried a few cables last was a 0.5M 100W40Gbps rated cable with a 3 and a lightning bolt unit from cabletime. ***@***.*** ~]$ fwupdmgr get-topology --show-all WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers! 20L8S2N800 │ ├─Thunderbolt host controller: │ Device ID: 4d2b831f1d8e7157c03d279f5924679f263c682c │ Summary: Unmatched performance for high-speed I/O │ Current version: 20.00 │ Vendor: Lenovo (THUNDERBOLT:0x0109|TBT:0x0109) │ GUIDs: 3849839e-d8b2-5941-9e69-5acc397d3ed3 │ 7e3190a2-d5a5-5a9e-b582-494b47f7288c │ 3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc │ 6783c3fd-8932-5306-bb7f-4dfec968e44b │ Device Flags: • Internal device │ • Updatable │ • System requires external power source │ • Device stages updates │ ├─Embedded Controller: │ Device ID: 323bda90831520867f8c02544efb9711dfca9fb6 │ Current version: 0.1.22 │ Minimum Version: 0.1.22 │ Vendor: DMI:LENOVO │ GUIDs: 4d254d6e-cd67-477b-97d5-bc3048af45c4 │ cc9cf2fe-ba41-532b-bc58-17ca9cc3c63d │ Device Flags: • Internal device │ • Updatable │ • System requires external power source │ • Supported on remote server │ • Needs a reboot after installation │ • Device is usable for the duration of the update │ I do see this on any fwupmgr command: (fwupdmgr:7176): Fwupd-DEBUG: 14:36:06.522: Emitting ::status-changed() [idle] WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers! Not sure which device provides that. There are a few other oddities going on. (after reflashing BIOS and Embedded Controller to current): At boot: (A7) Me FW Downgrade - Request MeSpiLock Failed... In BIOS, Management Engine ( T480s ) has a blank Value and doesn't respond to Control-P at boot? Video (40AC dock hosting NIC and 2 x Display port monitors) will periodically restart. Is there any way to debug a dock? Where should I start? thanks John Westerdale Sr Container Consultant * RHCSA-8.2* Red Hat <https://www.redhat.com/> NYC Office/WFH ***@***.*** M: 201-376-9993 IM: jwesterd He / Him / His Envision the future, tolerate the present <https://www.redhat.com/> I respect your Life-Work balance. No need to respond immediately if you receive this outside your normal working hours.
> […](#)

Please be aware that you have to use the correct usb-c port on the computer: there are 2 connectors on the left side, only one is thunderbolt-enabled.

jwesterd-redhat, 2022-10-19:
the second USB-C port from the back is the one for sure .. Over that, the
Display and Network and sound and USB and .. all sorta stuff goes over that
link.

What tools do we have to query/interact at that level?

lsusb -t shows the dock:
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 1: Dev 2, If 1, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 1: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid,
12M
    |__ Port 1: Dev 2, If 0, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 5, If 2, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 3: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 3: Dev 5, If 3, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 3: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M

What other information or interaction is available?

Dock NIC:
enp11s0u1 ->
../../devices/pci0000:00/0000:00:1c.4/0000:04:00.0/0000:05:01.0/0000:07:00.0/0000:08:02.0/0000:0b:00.0/usb6/6-1/6-1:1.0/net/enp11s0u1

A few things there.

Ex: Is there any way to reset the NIC on the docking station, as the
display stays up perfectly, but the nic drops without warning or error log
that I can see.
Tried unloading and reloading the kernel module, nope.  maybe there is a
way to poke the doc to reset the NIC Or the USB HUb withing doc or the
external TB3 interface on laptop?
Thx.

eBug, 2022-10-19:
I was referring to the port of the laptop. But it also happens that there's only one TB usb port on the 40AC.

girante, 2023-07-28:
> TBT3 WS G2 dock firmware is on LVFS now - I think this can be closed. Let me know if you're still seeing problems

G2 is on LVFS, but how about G1?

I am having problems with monitors not waking up after sleep/hibernation and it is driving me nuts. It happens most of the times when the display power saving mode is activated, so almost every time I return to the computer I have to deal with this issue.

Having to disconnect and reconnect the dock until it all works is not something I would expect from a Lenovo product.

I see this described for the Lenovo Thunderbolt 3 Dock G2: https://download.lenovo.com/pccbbs/mobiles/tbt3wsdkfw3182_2.txt

But not for the G1: https://download.lenovo.com/pccbbs/mobiles/ar_tbtdockfw10.txt

But even so, I would like to try upgrading to see if this somehow improves...

Would it be possible to either add G1 to LVFS, or maybe the .bin is available somewhere for upgrading using "fwupdmgr install FILE.img"?

girante, 2023-07-28:
I found another issue here that is still open, and is perhaps the best place to discuss about the Lenovo Thunderbolt 3 Dock (G1):

https://github.com/fwupd/firmware-lenovo/issues/5