fwupd/firmware-lenovo #95: ThinkPad X1 Carbon Gen 8 - TPM PCR0 differs from reconstruction

jmbaur, 2020-12-14:
**Describe the bug**
TPM PCR0 differs from reconstruction, uploading based on instructions from https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction. I apologize for the long output!

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Came preinstalled (Ubuntu)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20U9001PUS
│
├─Thunderbolt Controller:
│     Device ID:           6af3a4488cc4fcf49df8832eeb6b4e6d8256bc12
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               4117c99d-0660-5db6-bbfe-65ed036debe5
│                          bb3a7757-29b3-5950-9dba-71cc18c6fb86
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Prometheus:
│ │   Device ID:           d432baa2162a32c1554ef24bd8281953b9d07c11
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3158804
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:        • Updatable
│ │                        • Supported on remote server
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         8dcffb5d059857368caa2b7b22371ebd831b0c0c
│       Current version:   0022
│       Minimum Version:   0022
│       Vendor:            Synaptics (USB:0x06CB)
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4
│       Device Flags:      • Updatable
│                          • Supported on remote server
│     
├─System Firmware:
│     Device ID:           6f1acb499f6e604cfd7563f33d04f26ecdd07138
│     Current version:     0.1.14
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update Error:        TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUID:                31245a58-b272-4776-9398-3df5e8b532db
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
[... 909 lines cut ...]
  2:                     312c7ef199b63d37031e62e4898c487de4cfe165b5631ca74a204e06135b97a8
  3:                     b2a83b0ebf2f8374299a5b2bdfc31ea955ad7236
  3:                     3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
  4:                     8d974945665574ac39ecc3664cf8927e05323ed8
  4:                     13342153f9ab140c857001246889211f023c53ae200acac44fa09018c6ab90dd
  5:                     e149b546e5e0984d981fa510315e8ff098b7bfe1
  5:                     276c054ab909069c11cf8354155777b921a90061cf7d3a8cb114affbb24a45fa
  6:                     b2a83b0ebf2f8374299a5b2bdfc31ea955ad7236
  6:                     3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
  7:                     c0312baf908a23ca5faa3bf6d756f2fcbd07041c
  7:                     5641ca05b7b9aede011bfe30b1ea6e4ae183f9fc7b382a289bebf68b53452daf
  8:                     b338c6c5d11e3bdb9e67af491038d59f4f78cc1c
  8:                     a985b2bbdf2972fa6291a7ca231e412dbd9136027301c033eb7a4258f8531022
  9:                     87c016587d44b8c169bec09c1fdfce569f897927
  9:                     dc03dccdca71bc7e58526baa32a4c5f93bef3e9b35f15c7b126bf010201057fb
```

**Additional questions**
- Operating system and version:
Ubuntu 20.04

- Have you tried rebooting?
N/A
- Is this a regression?
N/A

superm1, 2020-12-14:
I'm fairly confident that this is fixed in newer fwupd versions outlined in that bug and if there are problems that remain they are firmware based, so moving repos.

jmbaur, 2020-12-16:
Yep! The warning does not show anymore using an updated version via the snap store.
Using version:
```shell
$ /snap/bin/fwupdmgr --version
client version:	1.5.4
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.4
```

closed 2020-12-16 by jmbaur