fwupd/firmware-lenovo #96: P72: ME version wrongly detected

dgutson, 2020-12-15:
Ubuntu reports
[image]

However, the intel tool reports that I already have 192.68.1606:
```
daniel@daniel-ThinkPad-P72:~/src/me$ ./intel_csme_version_detection_tool
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2020, Intel Corporation, All rights reserved.

Application Version: 4.0.1.0
Scan date: 2020-12-14 14:16:36 GMT

*** Host Computer Information ***
Name: daniel-ThinkPad-P72
Manufacturer: LENOVO
Model: 20MBCTO1WW
Processor Name: Intel(R) Xeon(R) E-2176M  CPU @ 2.70GHz
OS Version: Ubuntu 20.04.1 LTS (5.4.0-58-generic)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.68.1606
```

ijiang1, 2021-01-20:
hi @dgutson 
could you help to provide fwupdmgr get-devices info? and there is a newer version, you may have a try.

kmauleon, 2021-02-04:
update from BIOS Dev... thank you very much....
**no clear progress yet regarding BIOS ESRT problem investigation**

dgutson, 2021-02-12:
hi @ijiang1, I run that command and uploaded the report.
This was relevant:
```
"UpdateState" : 2,
                               "UpdateError" : "TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction",
                               "Guid" : [
                                 "1e1fe415-74e8-49e1-9508-106b3d13d50d"
                               ],
                               "Plugin" : "uefi",
                               "VersionOld" : "0.1.36",
                               "VersionNew" : "0.1.37",
                               "Flags" : 574619947,
                               "Created" : 1598894156,
                               "Modified" : 1598894409,
```

dgutson, 2021-02-12:
@ijiang1 `"MachineId" : "556bc466e9dea674e88478c482b763b5c97f4dde78e2694cdd4abe6afa88b6d2"`

kmauleon, 2022-11-16:
Closing this as BIOS ESRT fix has been rolled out already.  Please reopen if you still see the problem.  Thank you very much.