fwupd/firmware-lenovo #102: Missing ME Firmware version after update on P50

robbyki, 2021-02-18:
After a recent update on fedora 33 completely missing an ME Firmware version all together from bios settings and have the following flash on screen before completing startup:

```
(A7) Me FW Downgrade - Request MeSpiLock Failed
```
`fwupdmgr get-history` also shows the following:

```
Unknown                  [***************************************]
20EQS3B400
│
├─Intel Management Engine:
│ │   Device ID:          1f9ed21ce218ebcd0e1c803b7da08dd4e33d01a3
│ │   Previous version:   184.50.3448
│ │   Update State:       needs-reboot
│ │   Last modified:      2021-02-17 21:13
│ │   GUID:               671d19d0-d43c-4852-98d9-1ce16f9967e4
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      184.55.3510
│       Remote ID:        lvfs
│       License:          Unknown
│     
```

kmauleon, 2021-02-23:
hi @robbyki you have very old MEFW version... can you please show screen shot of BIOS setting and results of following command
**fwupdmgr get-devices**
thank you very much

kmauleon, 2021-03-02:
hi @robbyki without ME version in BIOS setting would mean that ME is corrupted.  may we know what was the prior setup before you encountered the error?  Were you updating BIOS? EC? ME?

kmauleon, 2021-05-26:
hi latest 11.8.83.3874 (184.83.3874) has been put to stable... thank you very much....

closed 2022-05-05 by robbyki