fwupd/firmware-lenovo #103: Wrong firmware shown in Gnome Software

CodedOre, 2021-02-23:
[Original Tweet](https://twitter.com/CodedOre/status/1364141948047622144)

On my system, Gnome Software seems to show a wrong firmware update not meant for my hardware:

**My System:** Lenovo IdeaPad 720s-15IKB\
**System for the Update:** Lenovo ThinkPad X1 Yoga 3

[image]

---

Output of `fwupdmgr get-devices`:

```
81AC
│
├─SAMSUNG MZVLW256HEHP-000L2:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Zusammenfassung:    NVM Express Solid State Drive
│     Current version:    4L1QCXB7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:              030c853f-259a-57a0-b3fb-1c66100db94b
│                         aed4d3c0-fd97-5e46-a32f-ff35e0692f6d
│                         f87b9ac8-1cb3-5c0a-ae57-7144f211fe5e
│                         5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
│                         13f7a30c-48d7-563c-8157-408e3cf0985d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          4be1d82544d8b1332b6189d4368d1e5a03e4883c
│ │   Current version:    1.75.41643
│ │   Minimum Version:    1.75.41643
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   GUIDs:              e20bafd3-9914-4f4f-9537-3129e090eb3c
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       e9a595ff-a990-58cc-926d-669b8b018e76
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Zusammenfassung:  UEFI Revocation Database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─Touchpad:
│     Device ID:          a753a477a33fb4106b91d39d2b966621f9475584
│     Current version:    1.3.2533979
│     Bootloader Version: 54.0.0
│     Vendor:             Synaptics (HIDRAW:0x06CB)
│     GUIDs:              f124e67f-79b3-50b8-ac64-cab867f40d35
│                         ac8761e3-496d-5c5e-ad92-f1eae31a0561
│                         e8583d67-886b-58e9-8bae-e50e79b7a43a
│                         4d797423-a52b-5d82-b1b6-5857d318bfa5
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:          cfef24379752061bc5125871ad11cf4cf412ef77
│     Current version:    0
│     Vendor:             DMI:LENOVO
│     GUIDs:              6a537f6c-51e2-460f-ba28-f79d9306d236
│                         13382573-0c7c-5dda-8a5a-dc2dd059b8d5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─VMM3200:
      Device ID:          d762543f8c20f636e6fff031a000078d3e10c600
      Zusammenfassung:    Multi-Stream Transport Device
      Current version:    0.00.00
      Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:              1beb64bd-6dc8-5db2-84b9-f32964abc876
                          b57f0a20-424c-51e0-b91b-575f3d74ed3a
                          d6072785-6fc0-5f83-9d49-11376e7f48b1
      Device Flags:       • Updatable
```

hughsie, 2021-02-23:
I've demoted the firmware and notified the Lenovo team about this bug. Thanks!

rgolangh, 2021-02-24:
I just did an update on a T470s and it hangs after the reboot. I hope I didn’t get a wrong bios update and it bricked the machine. Any advice how to proceed?

hughsie, 2021-02-24:
@rgolangh what hardware do you have and what did the firmware say? Power down, remove the batteries and wait 30 minutes.

rgolangh, 2021-02-24:
Hardware is a Lenovo T470 model 20HGS22D0J
I didn’t read the details (next time!) that appeared in “Software” but there were 2 updates there. 
I approved the updates and rebooted, it showed a Lenovo screen that says it is updating and then went black screen. 
Force power off using the button doesn’t work.

rgolangh, 2021-02-24:
Thanks @hughsie - Removing the batteries and waiting 30m works.
My updates history shows the update from today (Feb 24th) 
```
20HGS22D0J
│
├─T470s Thunderbolt Controller:
│ │   Device ID:          f41d9760864f7dd8d08fa52f90d8fccb5aa87200
│ │   Previous version:   04.00
│ │   Update State:       failed
│ │   Update Error:       failed to get device after update: failed to wait for detach replug: device f41d9760864f7dd8d08fa52f90d8fccb5aa87200 did not come back
│ │   Last modified:      2019-11-27 09:50
│ │   GUID:               96c3f95f-611a-56f8-be95-0f912cb84657
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Hardware is waiting to be replugged
│ │ 
│ └─  New version:      20.00
│       License:          Unknown
│     
├─Embedded Controller:
│ │   Device ID:          03a6930fc4c833443a7502dde379b98b961c8c57
│ │   Previous version:   0.1.18
│ │   Update State:       success
│ │   Last modified:      2020-05-13 07:37
│ │   GUID:               e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      0.1.21
│       License:          Unknown
│     
├─Intel Management Engine:
│ │   Device ID:          cbede50560aaa37366326cd54923421517427f39
│ │   Previous version:   184.77.3664
│ │   Update State:       success
│ │   Last modified:      2021-02-24 06:37
│ │   GUID:               e9124c4a-fdff-42e5-b2ad-f745db345953
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T470s Corporate ME Update:
│       New version:      184.82.3838
│       Remote ID:        lvfs
│       Summary:          Lenovo ThinkPad T470s Corporate ME Firmware
│       License:          Proprietary
│       Size:             7.5 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N1WRM21W
│       Vendor:           Lenovo Ltd.
│       Description:      
│       Version 11.8.82.3838 (LVFS: 184.82.3838)
│       
│       Security issues fixed:
│       
│       • CVE-2020-8705
│     
└─System Firmware:
  │   Device ID:          84149fb78009cf27cc1dd5520911f46f8792dbfe
  │   Previous version:   0.1.38
  │   Update State:       failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2021-02-24 06:37
  │   GUID:               7a176688-0960-47ba-931b-7829849e8347
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T470s System Update:
        New version:      0.1.40
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T470s System Firmware
        License:          Proprietary
        Size:             9.4 MB
        Created:          2016-07-08
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:      
        Lenovo ThinkPad T470s System Firmware
        
        Security issues fixed:
        
        • CVE-2020-0593
        • CVE-2020-0592
        • CVE-2020-0591
        • CVE-2020-0590
        • CVE-2020-0588
        • CVE-2020-0587
        • CVE-2020-8698
        • CVE-2020-8696
      
```

CodedOre, 2021-02-25:
Just to notice it: It's back again...

[image]

Now it is set to the Consumer branch, but otherwise still wrong device.

hughsie, 2021-02-25:
I've demoted both firmwares to testing. I'll send an email to the uploader.

lijunwang1990, 2021-03-03:
@rgolangh you can try to download the ThinkPad T470s System Update FW file from https://fwupd.org/lvfs/ , use "fwupdmgr --allow-older --allow-reinstall install  [Name].cab" to update the FW, thank you.

closed 2021-09-17 by CodedOre