fwupd/firmware-lenovo #104: Tunderbolt 3 Workstation Dock 40AN issue - block whole ethernet LAN when Laptop in suspend mode

reirob, 2021-02-25:
Hello,
I come here because I discovered a new issue (see below) with Tunderbolt 3 Workstation Dock 40AN connected to Linux. As I was wondering if a Firmware update would resolve it. This is how I came here - but now I realize for Linux users there is no possibility, until now, to have a Firmware update. That's sad.

The issue: After 1 year that I dismissed the dock, because it was giving me trouble, I tried again. I discovered that the external screen, connected via Thunderbold cable to the dock, worked fine. I could as well charge a phone by plugging it to the dock. However the laptop could not see any USB devices or the Ethernet. At some point I tried reconnecting and I discovered that the dock has 2 power entries: 1x230w and 1x135w/65w. I was wondering, why it needs 2 power entries, but I found a spare 65w Thinkpad charger and plugged both the 230w and the 65w - and I was very happy to discover that USB and Ethernet now worked on Linux! I finally could use the dock! I still do not understand why on earth this dock needs 2 power cables, but that's not the main issue. Now comes the issue: When I put the laptop into suspend mode, then the dock seems to block the whole Ethernet LAN! And by the WHOLE ethernet LAN I mean not only the devices that are on the same switch, but the device that are beyond the 2nd router. And by block, I mean that if the devices had an Internet connection they lose it. How is this possible? As an intermediate solution I found out, that when this happens, I just need to unplug the 65w power plug and then the Ethernet is unblocked.

Just letting others know what happens and in case somebody wonders. Did not find anybody describe this issue yet. Thought maybe a new firmware would solve the issue, but with Linux it's not possible.

The dock is type 40AN, FRU PN: 03X753B
The laptop attached is a ThinkPad X1 Extreme, Gen 1.

mrhpearson, 2021-03-03:
Hi,
I hear you on the two power connectors (the TBT4 dock that is just coming out doesn't have that). I'm sure there's a good reason - but I don't know it and it caught me out too setting mine up the first time

Firmware updates - it's getting really close to being available (it's been a really long drawn out process...but it's really close, and if you've followed the other threads you'll know how annoying the whole process has been)

However - can you disable pause control on the port? I believe the dock is flooding your network with pause packets - because it thinks the link is up but it's buffers get full and everything unravels from there.
I don't know why the extra power supply makes a difference - I'll ask the dock team and see if they can explain. I have some notes from this issue about a year ago - I thought it was fixed but....I don't have the details handy.

Mark

reirob, 2021-03-04:
Hello Mark,

many thanks for answering on my issue. How do I disable pause control on the port?

I was in the laptop BIOS tried to find there something, but only have seen some Wake On LAN stuff, that I disabled, and strangely now my laptop does not go into suspend mode - hibernate still works, and when I hibernate the dock does NOT block the LAN. I think I need to find out which BIOS settings to revert, in order to enable suspend mode again.

If you tell me how to disable pause control, and once I manage suspend mode to work again, I will test if it still blocks the Ethernet.

Regards,
Robert

mrhpearson, 2021-03-17:
Hi Robert,

Sorry for the slow reply - I thought this was going to be really easy, and tried it using ethtool ('ethtool -a' & ethtool -A ) and....it failed miserably. I need to go dig into the kernel driver and figure it out. I've got a horrible suspicion I'm going to need to push a kernel patch for it). Suffice to say, the answer isn't as simple as I thought it was.
If you're able to configure pause on the link partner's side can you try that as a workaround.

Mark

reirob, 2021-03-17:
Hello Mark,

many thanks that you are still following up on this.

Do I interpret it right, that you are able to reproduce the same issue?

I am sorry, but I am not as deep in this technology, so I don't know what you mean by "configure pause on the link partner's side"? Is my assumption right that you might want me to tweak some ethernet pause parameter on the other devices that get blocked by the USB-C dock? I think that I will not be able, because one is a Video/TV appliance, the other is the router itself and the others that I get when I have visit, are Windows 10 computers, but they are not mine. My laptop that I plug into the USC-C dock, is the only Linux and ThinkPad device. If you want me to do something on it, I will voluntary try my best, but please provide instructions how to do so.

In the meantime, I am living with the workaround, as soon as I put my laptop off, I unplug the 65W charger from the USB-C dock.

Thanks a lot for your efforts to make Linux better!
Robert

mrhpearson, 2021-03-17:
Hi Robert,

I haven't reproduced it directly - I don't have the setup at home that let's me with my devices. I have had another user who reported the same and we traced it down to pause packets flooding the network and he was able to work around it this way. In all honesty I'm guessing a little, but it's an educated guess :)

You might be able to change the setting on the router itself - and it only needs to be done on the port connected to the dock. Unfortunately I can't provide guidance on configuring the router, and it's possible you may not be able to do so - I can't on mine, it doesn't allow access at that level.

Mark

joakim-tjernlund, 2021-05-07:
i wonder if this kernel patch fixes this?
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.10.y&id=d844aaa49ac8b3225e906a150f89d2b805ad8cfc