fwupd/firmware-lenovo #107: I can't update my Thinkpad X1 Yoga 3rd gen anymore

aitorpazos, 2021-03-10:
I used to update it successfully in the past over multiple versions. However, now I am stuck and whenever I try to run `fwupdmgr update`, my laptop restarts, goes to a blank screen for around a minute or two and then reboots again without having completed the upgrade.
`fwupdmgr` reports the following error in the report:
```
"UpdateState" : 3,
"UpdateError" : "failed to run update on reboot",
``` 

This is the output of `fwupdmgr get-topology`:
```
20LDCTO1WW
│
├─Embedded Controller:
│     Device ID:          12e31bc5df493718c50805292e594ebf744efc9d
│     Current version:    0.1.19
│     Minimum Version:    0.1.19
│     Vendor:             DMI:LENOVO
│     GUIDs:              6c6c8104-65e5-45c2-8e31-059df92ef3fd
│                         333ac7e4-2fdc-5e34-8589-166543bc8825 ← UEFI\RES_{6C6C8104-65E5-45C2-8E31-059DF92EF3FD}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          a9ca34786c44d8798eb70e2ee146c8e7a6b9c613
│     Current version:    184.70.3626
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              7c38c98a-c3e9-4c15-b0cd-c634d441cf7f
│                         c1a1bc56-3d5b-5ce3-b2b6-fb24f09a94fa ← UEFI\RES_{7C38C98A-C3E9-4C15-B0CD-C634D441CF7F}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─SAMSUNG MZVLB512HAJQ-000L7:
│     Device ID:          310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:            NVM Express Solid State Drive
│     Current version:    5L2QEXA7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:      S3TNNY0K325601
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801&REV_00
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                         47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         79d6cfae-a5a2-5936-9248-5aebd23480f7 ← SAMSUNG MZVLB512HAJQ-000L7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:          d3db6c5633d83a940028fc88d8102d1f1563e3a4
│     Current version:    0.1.40
│     Minimum Version:    0.0.1
│     Vendor:             LENOVO (DMI:LENOVO)
│     GUIDs:              fc5552dc-7882-409b-9b98-699361c74fc4
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│                         d377c5fb-f0f4-5829-8a2e-b796b30f258a ← UEFI\RES_{FC5552DC-7882-409B-9B98-699361C74FC4}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          84f61b1e7f3b385d1d60dd5cca4fe7be18490047
      Current version:    4784132
      Minimum Version:    4784132
      Vendor:             DMI:LENOVO
      GUIDs:              6a8ddeb9-d0d3-4095-b5f0-0f6f5a344622
                          6dee972a-e99c-55e0-9c4c-eb22d7e3f6c4 ← UEFI\RES_{6A8DDEB9-D0D3-4095-B5F0-0F6F5A344622}
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```

kmauleon, 2021-03-11:
hi @aitorpazos forwarded your concern to dev owner.  please wait for advise.  thank you very much...

aitorpazos, 2021-03-15:
After getting a package upgrade for `fwupd` (OpenSUSE Tumbleweed) the upgrade worked:
```
fwupd-1.5.6-1.1.x86_64 upgraded to fwupd-1.5.7-1.1.x86_64
```
Closing this ticket.

closed 2021-03-15 by aitorpazos