fwupd/firmware-lenovo #108: Thinkpad X1 Nano UEFI firmware 1.39 problems

adamjbc, 2021-03-26:
The X1 Nano firmware available for Linux (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2TET.firmware) has a newer version number (1.39) and older release date (3 March 2021) than the latest available for Windows on the Lenovo website (1.39, 8 March 2021: https://support.lenovo.com/gb/en/downloads/ds547748-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-x1-nano-gen1-type-20un-20uq).

When I installed the Linux version (via Ubuntu), my machine started to say "2nd ECSDA region was changed. Please update the ECFW" when it powered on, and shortly afterwards it said "System has self-healed by restoring the BIOS from backup, please upgrade to the latest BIOS."

I eventually recovered by disabling the relevant option in the UEFI/BIOS to allow me to install older firmware versions, wiping the OS, installing Windows, and updating using the latest firmware on the Lenovo website.

mrhpearson, 2021-03-26:
Thanks for the report.
I've raised this to the firmware team (internal bug reference LO-1103).
As a note - my collegue is running 1.39 with EC 1.26 so my guess is that they need to set the metainfo to ensure the BIOS update only takes place once EC is at the right version, but I'll wait for confirmation from the FW team.
Sorry you ended up having to revert to Windows - that sucks :(

adamjbc, 2021-03-26:
Thanks for looking in this. I'm fairly sure my EC version was less than 1.26 so that may well be it, although I don't have a record of what it was before the upgrade / attempted upgrade.

I was planning on switching to Windows for a bit anyway, as battery usage whilst suspended seemed a bit high and I wanted to compare. It seemed to improve a bit when I disabled the WWAN in /proc/acpi/wakeup so I'm wondering if the lack of a WWAN driver is meaning it can't suspend to ram properly. But I don't know anything about this, just googling and guessing!

Thanks again.

jbuencuerpo, 2021-04-03:
Hi, I have also a X1 nano, and it is reporting on the firmware update tab the 1.39 version. I guess it is better to wait until this is solved, right?
Thanks,

mrhpearson, 2021-04-05:
Probably safest - though can you confirm what version your EC is at? I need to chase the FW team for an update :/

jbuencuerpo, 2021-04-05:
UEFI-BIOS 1.36, Embedded Controller Version 1.26.

mrhpearson, 2021-04-06:
Feedback from the FW team is as long as you are at EC version 1.26 there shouldn't be a problem.
I've asked them to look at adding that limitation to the metainfo to prevent other users hitting the same problem
Mark

jbuencuerpo, 2021-04-09:
Done, everything working so far. Actually, I am on 1.41 now with EC 1.37.

zwiar, 2021-05-20:
My device manager is showing that my EC version is 1.39, but showing an error. Any suggestions how to fix? I'm with Win10 Pro.

Thanks!

telmich, 2022-01-25:
I have also been hit by this problem and the bios message appears on every reboot now

Siberuang007, 2024-04-23:
[image]

cannot solve my problem. I've tried installing EC 1.22 but my EC version stuck at ver 1.00 and my bios is 1.57, I cannot install new bios which is 1.62 and EC 1.48, until now my machine always say "2nd ECSDA region was changed. Please update the ECFW" when it powered on