fwupd/firmware-lenovo #109: Firmware update hangs on Thinkpad T470

sethkush, 2021-03-27:
Reposting [this](https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1919046) to the correct place.

I tried to upgrade my firmware today because several security updates are available, and after 4 hours my ThinkPad T470 was still stuck on the "Installing new firmware..." screen with a spinning line and no progress bar. I unplugged it and had to wait for the battery to die to stop it.

After plugging it in again it booted normally.

I tried to find a way to report bugs to LVFS, but I couldn't find anywhere to do that on their site. Sorry if this is the wrong place for this bug.

I've pasted fwupdmgr output below:

% fwupdmgr get-updates
• Thunderbolt Controller has the latest available firmware version
• Crucial CT525MX300SSD1 has no available firmware updates
• System Firmware has the latest available firmware version
• UEFI Device Firmware has the latest available firmware version
20HDCTO1WW
│
└─UEFI Device Firmware:
  │ Device ID: 698432792a49732713733556c3d55a27bb3d5ee7
  │ Current version: 184.70.3626
  │ Minimum Version: 184.70.3626
  │ Vendor: DMI:LENOVO
  │ Update State: success
  │ GUID: 1c0f9993-38e5-4a62-8dab-f2efee639b63
  │ Device Flags: • Internal device
  │ • Updatable
  │ • Requires AC power
  │ • Supported on remote server
  │ • Needs a reboot after installation
  │ • Device is usable for the duration of the update
  │
  ├─ThinkPad T470 Consumer ME Update:
  │ New version: 184.80.3746
  │ Remote ID: lvfs
  │ Summary: Lenovo ThinkPad T470 Consumer ME Firmware
  │ License: Proprietary
  │ Size: 2.2 MB
  │ Details: https://pcsupport.lenovo.com/de/en/search?query=N1QRN28W
  │ Vendor: Lenovo Ltd.
  │ Flags: is-upgrade
  │ Description: Version 11.8.80.3746 (LVFS: 184.80.3746)
  │
  └─ThinkPad T470 Consumer ME Update:
        New version: 184.77.3664
        Remote ID: lvfs
        Summary: Lenovo ThinkPad T470 Consumer ME Firmware
        License: Proprietary
        Size: 2.2 MB
        Vendor: Lenovo Ltd.
        Flags: is-upgrade
        Description: Version 11.8.77.3664 (LVFS: 184.77.3664) 2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR) Several security fixes and enhancements are on this release.

ProblemType: Bug
DistroRelease: Ubuntu 20.04
Package: fwupd 1.3.11-1~focal1
ProcVersionSignature: Ubuntu 5.8.0-44.50~20.04.1-generic 5.8.18
Uname: Linux 5.8.0-44-generic x86_64
ApportVersion: 2.20.11-0ubuntu27.16
Architecture: amd64
CasperMD5CheckResult: skip
CurrentDesktop: Regolith:GNOME-Flashback:GNOME
Date: Sat Mar 13 14:06:12 2021
InstallationDate: Installed on 2020-05-01 (316 days ago)
InstallationMedia: Ubuntu 20.04 LTS "Focal Fossa" - Release amd64 (20200423)
SourcePackage: fwupd
UpgradeStatus: No upgrade log present (probably fresh install)

mrhpearson, 2021-03-29:
@kmauleon - just wanted to check if this was one you'd seen before or not; otherwise I'll raise a bug to the FW team.
Thanks!
Mark

kmauleon, 2021-03-30:
hi @mrhpearson sorry I did not encounter this issue with my systems.  T470 is handled by ODM LCFC.... thank you