fwupd/firmware-lenovo #110: Missing Firmware for USB-C Dock (Gen2)?

j13m126, 2021-03-30:
Updates for the usb-c dock gen 2 went online (in testing) on november 19th of 2020. As of today there has been no visible progress and there is still one usb 3.1 hub unsupported (at least on my machine):

Devices with no available firmware updates: 
 • USB3.1 Hub
...

When I run fwupdmgr get-udpates I get 2 usb 3.x hubs:

│ │   Device ID:          91c9a317b30acf7c0cd967f2ce6925d518ef19a7
│ │   Zusammenfassung:    USB 3.x Hub
│ │   Current version:    13.24
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)

and

│ │     Device ID:        05285288b47a52d357895905caed1c00e37e22f0
│ │     Zusammenfassung:  USB 3.x Hub
│ │     Current version:  13.23
│ │     Vendor:           VIA Labs, Inc. (USB:0x17EF)

But on lvfs there are 2 devices with update version [13.23](https://fwupd.org/lvfs/devices/com.lenovo.Gen2.VLIUSBQ8.firmware) and [13.24](https://fwupd.org/lvfs/devices/com.lenovo.Gen2.VLIUSBQ7.firmware). Is that a bug or why is it not recognized correctly? 
Can you tell us why these haven't been labeled as stable yet?

Fedora 34 Workstation (BETA) with fwupd-1.5.8-1.

brianredbeard, 2021-04-13:
This is good to know.  I was just looking at the Gen2 dock and found a number of confusing results, e.g.:

https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadThunderbolt3DockGen2.MST.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadThunderbolt3DockGen2.USB2.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadThunderbolt3DockGen2.USB1.firmware

All three of these links point back to the same firmware file (`64a6203cbf9a0efc21e9e973fb9c79da8db4229754ba7ea51511ac104ba68a67-LenovoTRDock_Firmware_3168.cab`) .  Introspecting the file, one sees the following:

```bash
[~]$ cabextract -l  64a6203cbf9a0efc21e9e973fb9c79da8db4229754ba7ea51511ac104ba68a67-LenovoTRDock_Firmware_3168.cab
Viewing cabinet: 64a6203cbf9a0efc21e9e973fb9c79da8db4229754ba7ea51511ac104ba68a67-LenovoTRDock_Firmware_3168.cab
 File size | Date       Time     | Name
-----------+---------------------+-------------
      7648 | 21.01.2021 15:28:00 | Lenovo_Thunderbolt3_B0_Gen2_Patch_04-0E-92_Rev_0092_NVM-.ptc
       488 | 25.01.2021 06:13:48 | Lenovo_Thunderbolt3_B0_Gen2_Patch_04-0E-92_Rev_0092_NVM-.ptc.asc
    524288 | 28.12.2020 11:41:00 | Panamera_fw_tbt3_Ver5_05_005.fullrom
       488 | 25.01.2021 06:13:48 | Panamera_fw_tbt3_Ver5_05_005.fullrom.asc
       920 | 25.01.2021 06:13:48 | README.txt
    495616 | 12.11.2020 11:00:00 | TRDOCK_6200I2C.bin
       488 | 25.01.2021 06:13:48 | TRDOCK_6200I2C.bin.asc
     61784 | 10.11.2020 15:21:00 | TR_Dock_Tier1_VL820Q7_FW5163.bin
       488 | 25.01.2021 06:13:48 | TR_Dock_Tier1_VL820Q7_FW5163.bin.asc
     61784 | 08.10.2019 16:36:10 | TR_Dock_Tier2_VL820Q7_FW5133.bin
       488 | 25.01.2021 06:13:48 | TR_Dock_Tier2_VL820Q7_FW5133.bin.asc
      1960 | 25.01.2021 06:13:48 | audio.metainfo.xml
      1988 | 25.01.2021 06:13:48 | dp_hub.metainfo.xml
     10070 | 25.01.2021 06:13:48 | firmware.jcat
     23800 | 17.11.2020 16:31:00 | fr2033_app_v0.30_OneClick.hex
       488 | 25.01.2021 06:13:48 | fr2033_app_v0.30_OneClick.hex.asc
      1884 | 25.01.2021 06:13:48 | mcu.metainfo.xml
      1782 | 25.01.2021 06:13:48 | tbt.metainfo.xml
      1834 | 25.01.2021 06:13:48 | usbhub_tier1.metainfo.xml
      1857 | 25.01.2021 06:13:48 | usbhub_tier2.metainfo.xml
```

It appears that the six different `<component>.metainfo.xml` files are largely redundant as even the changelogs are identical (which seems odd for firmware covering such a wide array of components):

```bash
[/tmp/tmp.71IvNDxyro]$ grep -A5 "      <description>" *.xml
audio.metainfo.xml:      <description>
audio.metainfo.xml-        <p>This stable release fixes the following issues:</p>
audio.metainfo.xml-        <ul>
audio.metainfo.xml-          <li>Show version correctly</li>
audio.metainfo.xml-        </ul>
audio.metainfo.xml-      </description>
--
dp_hub.metainfo.xml:      <description>
dp_hub.metainfo.xml-        <p>This stable release fixes the following issues:</p>
dp_hub.metainfo.xml-        <ul>
dp_hub.metainfo.xml-          <li>Show version correctly</li>
dp_hub.metainfo.xml-        </ul>
dp_hub.metainfo.xml-      </description>
--
mcu.metainfo.xml:      <description>
mcu.metainfo.xml-        <p>This stable release fixes the following issues:</p>
mcu.metainfo.xml-        <ul>
mcu.metainfo.xml-          <li>Show version correctly</li>
mcu.metainfo.xml-        </ul>
mcu.metainfo.xml-      </description>
--
tbt.metainfo.xml:      <description>
tbt.metainfo.xml-        <p>This stable release fixes the following issues:</p>
tbt.metainfo.xml-        <ul>
tbt.metainfo.xml-          <li>Show version correctly</li>
tbt.metainfo.xml-        </ul>
tbt.metainfo.xml-      </description>
--
usbhub_tier1.metainfo.xml:      <description>
usbhub_tier1.metainfo.xml-        <p>This stable release fixes the following issues:</p>
usbhub_tier1.metainfo.xml-        <ul>
usbhub_tier1.metainfo.xml-          <li>Show version correctly</li>
usbhub_tier1.metainfo.xml-        </ul>
usbhub_tier1.metainfo.xml-      </description>
--
usbhub_tier2.metainfo.xml:      <description>
usbhub_tier2.metainfo.xml-        <p>This stable release fixes the following issues:</p>
usbhub_tier2.metainfo.xml-        <ul>
usbhub_tier2.metainfo.xml-          <li>Show version correctly</li>
usbhub_tier2.metainfo.xml-        </ul>
usbhub_tier2.metainfo.xml-      </description>
```

That said I'm assuming that despite the largely similar content directed at administrators that the definitions were composed by hand because of typos, e.g.:

```bash
[/tmp/tmp.71IvNDxyro]$ grep summary audio.metainfo.xml 
  <summary>Conexanrt Audio firmware for ThinkPad Thunderbolt 3 dock Gen 2_ThinkPad Thunderbolt 3 workstation Dock Gen 2</summary>
```

Finally, I'd review the security caveats on the LVFS page linked at the top of this comment.  For review:

[image]

If you're comfortable with these caveats, there _are_ directions in the file `README.txt` (contained _within_ the cab file):

```
[/tmp/tmp.71IvNDxyro]$ cat README.txt
README FIRST
============

You can install this archive on Linux using fwupd >= 1.4.5 on supported
hardware. Use this command to install:

    fwupdmgr install 64a6203cbf9a0efc21e9e973fb9c79da8db4229754ba7ea51511ac104ba68a67-LenovoTRDock_Firmware_3168.cab

You can check the version of fwupd using:

    fwupdmgr --version

LEGAL NOTICE
============

This firmware archive was signed by the LVFS on 2021-01-25T06:13:49 and was made available
from https://fwupd.org/downloads/.

The firmware file is provided "as is", without warranty of any kind, express or
implied, including but not limited to the warranties of merchantability,
fitness for a particular purpose and noninfringement.

In no event shall the authors or copyright holders of the LVFS be liable for
any claim, damages or other liability, whether in an action of contract, tort
or otherwise, arising from, out of or in connection with the firmware archive.
```

mkauzlar, 2021-08-18:
I think this is the firmware that needs to be added:
https://support.lenovo.com/us/en/downloads/ds539092-firmware-for-windows-10-7-32-bit-64-bit-thinkpad-usb-c-dock-gen-2

groovenectar, 2021-09-09:
> I think this is the firmware that needs to be added:
> https://support.lenovo.com/us/en/downloads/ds539092-firmware-for-windows-10-7-32-bit-64-bit-thinkpad-usb-c-dock-gen-2

And it's in an `.exe` file... So are Linux users SOL for updating the firmware on that dock using what's currently available?

groovenectar, 2021-09-09:
@j13m126 Is your dock [`40AS0090US`](https://www.lenovo.com/us/en/p/accessories-and-software/docking/docking_usb-docks-(universal-cable-docks)/40as0090us?bvstate=pg:2/ct:q)?

rmappleby, 2021-09-23:
Worth noting that there seems to be quite a lot of unhappiness related to updating of these docks from Windows users, as found here: https://forums.lenovo.com/t5/Displays-Options-and-Accessories/Can-t-update-firmware-for-displayport-on-ThinkPad-USB-C-Dock-Gen2/m-p/4506706?page=1
The reports aren't definitive, but there appear to be problems updating the display port hub firmware. Some users are indicating failure of the DP monitor outputs afterwards. At least one mention of a dock being RMA'd by Lenovo.
When I run 'fwupdmgr get-devices' while attached to mine, I don't even see any mention of a DP hub device. Mine is a 40AS0090UK. From this support page (https://support.lenovo.com/us/en/solutions/acc500106-thinkpad-usb-c-dock-gen-2-overview-and-service-parts) it looks like all the 40ASxxxxxx devices are the same docking station hardware, with different power supplies / mains cables.

Sprinterfreak, 2022-01-08:
Also looking for updateability because of major usb/network'ing instabilities.

MadCatX, 2022-01-14:
+1

I need to restart the dock whenever I connect in to my computer, otherwise the external display connected to DP just won't light up. fwupdmgr 1.7.4 doesn't list any updates for the dock.

mrhpearson, 2022-01-17:
I discussed this with the docking team and currently LVFS support for this dock is **not** in plan and they don't have the resources to deliver it. I'm working on some exercises to improve our dock support and get them more funding, but I don't know what will be approved or not. This request is part of it but I have limited data as to how many Linux users are using the USB-c docks so it's a hard issue to prioritize. We are doing LVFS updates for our TBT workstation docks (though still a lot of work to do there) but for the USB-c dock I'm afraid the best solution currently is to borrow a Windows laptop and update from there.

I know the same components show up because of the TBT dock support but the docking team do not recommend or support using those on the USB-c dock. They were quite horrified when I suggested it. The PCB layout is different and the risk of failed updates and other issues is too high.

If anybody happens, somehow, to have any telemetry on how much Linux use of docks there is (please, not just a list of "I do") let me know. It doesn't have to be Lenovo specific - just something that helps me make a business case. I'm hoping to use the number of updates on the TBT docks to push the program but they're not yet at the stage that they are helpful (for many reasons).

I'm closing the ticket as support is not currently planned. If that changes in the future I will try and remember to add an update here.

closed 2022-01-17 by mrhpearson

bhack, 2022-01-17:
I think that telemetry stats could be only collected when you are going to propose an update.

https://lvfs.readthedocs.io/en/latest/telemetry.html

hughsie, 2022-01-18:
> I think that telemetry stats could be only collected when you are going to propose an update.

Right, we can't collect this kind of information for other updates as it oversteps the privacy policy unfortunately.

bhack, 2022-01-18:
@hughsie Is there a way to send dummy update for this?

hughsie, 2022-01-18:
> Is there a way to send dummy update for this

Well, you could issue an update with the current firmware versions, but people would have to do `fwupdmgr reinstall` manually and then wait for them to submit a report on the CLI to get the data.

bhack, 2022-01-18:
> > Is there a way to send dummy update for this
> 
> Well, you could issue an update with the current firmware versions, but people would have to do `fwupdmgr reinstall` manually and then wait for them to submit a report on the CLI to get the data.

Probably we could find a way in the future to voluntary pseudo-anon send specific hw stats that could help hw vendors to prioritize resources.

In this way we could voluntary send not LVFS covered hw reports in the db instead of handling only tickets on github.

Another alternative, but not mutal esclusive, is to create an LVFS compliant hw logo/trademark  that  hw vendors could use trought a certification process or something lighter.

In this way we could easy buy LVFS compliant HW.

hughsie, 2022-01-18:
> we could find a way in the future to voluntary pseudo-anon

That's super hard to avoid bias and also respect privacy, the latter have to do by law.

> create an LVFS compliant hw logo/

I've been asked for that before, I'm not sure how it works tbh. There has to be legal teams involved as it would be a trademark used with permission, and there would have to be some kind f policy to deal with vendors using the mark and not actually uploading firmware. Something to ponder for sure.

bhack, 2022-01-18:
> That's super hard to avoid bias and also respect privacy, the latter have to do by law.

I think it could be really similar to what we have here: 

> The fwupd daemon stores the result of each attempted update in a local SQLite database. In the event there is a firmware update that has been attempted, we now ask the user if they would like to upload this information to the LVFS. Using GNOME this would just be a slider in the control center privacy panel, although this feature is currently unimplemented.

probably we could just avoid too much duplicates but I suppose that also now we are not sharing the serial number in the telemetry.

> 've been asked for that before, I'm not sure how it works tbh. There has to be legal teams involved as it would be a trademark used with permission, and there would have to be some kind f policy to deal with vendors using the mark and not actually uploading firmware. Something to ponder for sure.

I think that the Linux Foundation is big and strong enough to setup and handle this process.

w1ldrabb1t, 2023-09-02:
I would like to report that I'm running Ubuntu 20.04 and I was using ThinkPad USB-C Dock Gen2 dockstation successfully to get both ethernet network and also a few other external devices like an USB Dual Pre Amp for a mic. 

Out of a sudden (I guess after some automatic OS/firmaware updates) I stopped having ethernet network connectivity and the USB Dual Pre Amp connected via USB to the dockstation stopped working...

Any ideas for why this is happening?

```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal

ThinkPad USB-C Dock Gen2 PD Controller:
       Summary:          CCGx Power Delivery Device
       Current version:  0.0.34
       Vendor:           Cypress Semiconductor (USB:0x04B4)
       Install Duration: 2 minutes
       Device Flags:     • Updatable
                         • System requires external power source
                         • Device stages updates
                         • Device can recover flash failures
                         • Unsigned Payload

Devices that were not updated correctly:
 • Embedded Controller (0.1.44 → 0.1.47)
 • System Firmware (0.1.41 → 0.1.58)
 • Battery (1.1.9 → 1.2.10)
 • Intel Management Engine (240.41.2142 → 240.45.2411)
```

hughsie, 2023-09-03:
> I guess after some automatic OS/firmaware updates

We never do automatic firmware updates. Try downgrading the kernel, or if that doesn't work reinstalling the original version of your OS and updating the packages one by one.

w1ldrabb1t, 2023-09-03:
Sorry, I didn't mean that I did the updates via fwupd. It was all under Ubuntu's software updates interface telling me that I had some updates available. 

Should I try to fix the Devices that were not updated correctly list?
```
 • Embedded Controller (0.1.44 → 0.1.47)
 • System Firmware (0.1.41 → 0.1.58)
 • Battery (1.1.9 → 1.2.10)
 • Intel Management Engine (240.41.2142 → 240.45.2411)
```
How do I do that though?

> Try downgrading the kernel
the OS kernel? How do you suggest I do that? :thinking:

mrhpearson, 2024-08-29:
Not officially - but if there are issues we'll do our best to resolve them

bhack, 2024-08-29:
Do you mean internal issues or issues we could subscribe/upvote?