fwupd/firmware-lenovo #111: Unable to update bios on Lenovo Thinkpad T14 due to version mismatch

jolars, 2021-03-31:
**Describe the bug**

I am unable to apply firmware updates for my Lenovo Thinkad T14 Gen1 (Intel) using the latest version of fwupd since fwupd recognized the current version (0.1.13) of the UEFI update as newer than an available update (0.1.17), even though it is in fact newer. I was able to solve this by running fwupd through a Ubuntu 20.04.2 live USB session, where I believe the version of fwupd is fwupd 1.3.11-1 or maybe fwupd 1.3.9-4. 

On the earlier version of fwupd, I see the following:

```bash
├─System Firmware:
│ │   Device ID:           dad157a257b9b98712d7189883790cf4fb897464
│ │   Current version:     0.1.13
│ │   Vendor:              LENOVO (DMI:LENOVO)
│ │   GUID:                9d255c4b-2d88-4861-860d-7ee52ade9463
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Cryptographic hash verification is available
│ │                        • Device is usable for the duration of the update
```

whereas for the latest version I see

```bash
├─System Firmware:
│     Device ID:           dad157a257b9b98712d7189883790cf4fb897464
│     Current version:     65549
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                9d255c4b-2d88-4861-860d-7ee52ade9463
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
```

**Steps to Reproduce**

* Install the latest version of fwupd using snap. Run

```bash
$ fwupdmgr refresh
$ fwupdmgr get-updates
```
I also tried downloading the bios update directly from lenovo's webpage and applying it manually, at which point I got the following message:

```
Specified firmware is older than installed '0.1.17 < 65549'
```

**Expected behavior**

A new version of the firmware should be installed.

**fwupd version information**

```shell
$ fwupdmgr --version

client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.8
```

I first used the fwupd version from the official ppa repository. Then I installed fwupd using snap.

[... 554 lines cut ...]
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img -> initrd.img-5.8.0-48-generic
├── initrd.img-5.8.0-45-generic
├── initrd.img-5.8.0-48-generic
├── initrd.img.old -> initrd.img-5.8.0-45-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.8.0-45-generic
├── System.map-5.8.0-48-generic
├── vmlinuz -> vmlinuz-5.8.0-48-generic
├── vmlinuz-5.8.0-45-generic
├── vmlinuz-5.8.0-48-generic
└── vmlinuz.old -> vmlinuz-5.8.0-45-generic
```

**Additional questions**
- Operating system and version: **Ubuntu 20.10**
- Have you tried rebooting? **yes**
- Is this a regression? **yes**
- Are you using an NVMe disk? **yes**
- Is secure boot enabled? **no**
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? **not sure**

mrhpearson, 2021-03-31:
Fix was delivered by the FW team and I've just tested and it should be good now:
fwupdmgr refresh --force
And check that under get-devices it's now showing the FW version in triplet form.

Thanks for the report (and to @hughsie for the pointers)

jolars, 2021-03-31:
Great, thanks! Yes, it looks good now. I still have other devices from lenovo that don't have their versions in triplet form, however, but perhaps that is as it should be? See the `UEFI Device Firmware` entries below.

```
20S0CTO1WW
│
├─ThinkPad Thunderbolt 3 Dock:
│     Device ID:          8919427f5fd42d01cca3756260593c20393ad011
│     Current version:    15.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0108|TBT:0x0108)
│     GUIDs:              fab08d10-6524-5cb6-ab20-6d365c812f9e
│                         0f511908-5c9f-511e-9099-06823ec33614
│                         df675f10-53ac-59ca-bddd-0a86ee492920
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Thunderbolt host controller:
│     Device ID:          af473a9d5e7f74c46165dddadf8aa258a0910f41
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              982e2ab2-6207-554c-974d-6300e30cd597
│                         d6591a43-ce08-57d7-8f00-a4f0cb625e01
│                         d868740a-5d9b-56e0-88b6-c7f76a26d0fa
│                         c4bbfb26-644e-50ee-b0dc-1edbc4e6c033
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          cef4be503c0f5c91b6ef48ef57a0ff9dea7fedfe
│     Current version:    65544
│     Minimum Version:    65544
│     Vendor:             DMI:LENOVO
│     GUIDs:              359c1169-abd6-4a0d-8bce-e4d4713335c1
│                         7ca908cb-fb7d-5573-834c-33b8054e2b7a
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          0fef0a0c55f6442bffaebd774ae771341c89571b
│     Current version:    58.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              cd900697-9e98-53bb-9f13-fac06bc26f77
│                         f8d483f9-9284-51b6-9363-7e6d59c40a97
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          96631d5d4db1e7fb990a201e4f22ed300a4f9373
│     Current version:    224.45.1389
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              0d7baa94-7f7f-4896-b1d3-8a6c594ff635
│                         d75316e9-3b91-5113-8ddb-6f250898c9fd
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─KXG6AZNV1T02 TOSHIBA:
│     Device ID:          599ead362818be284e800cbc30d4c29f934086c3
│     Summary:            NVM Express Solid State Drive
│     Current version:    5108AGLA
[... 90 lines cut ...]
│   
├─UEFI Device Firmware:
│     Device ID:          bd385cf420dfb0f29cf40df683528d7e4354ff5a
│     Current version:    22552
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     GUIDs:              8b8be06b-10b1-4443-88e3-1226388da75a
│                         e472ce20-7af4-5975-b074-2eae118b3262
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─VMM3320:
      Device ID:          d762543f8c20f636e6fff031a000078d3e10c600
      Summary:            Multi-Stream Transport Device
      Current version:    3.12.00
      Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:              2775d312-4747-5efa-95a7-3cd1d66894af
                          d6072785-6fc0-5f83-9d49-11376e7f48b1
                          5a9c2eec-d04c-5cec-aaaf-2043b6ea8605
                          e4acf3d9-790d-51a7-8284-b38a0ccb0e91
      Device Flags:       • Updatable
```

mrhpearson, 2021-03-31:
Yeah - I believe those are devices that we don't have LVFS support for....I'm not actually sure what they are (I should dig into that...).

Mark

hughsie, 2021-03-31:
> I believe those are devices that we don't have LVFS support for

Agree.

> I should dig into that...

I would be most interested in the results! If there are specific GUIDs that a common across all models then we can add a better `Name` and `VersionFormat` in the quirk file. e.g. https://github.com/fwupd/fwupd/blob/master/plugins/uefi-capsule/uefi-capsule.quirk#L37

emummel20, 2021-06-29:
@hughsie FYI I am dealing with the same issue right now https://www.reddit.com/r/thinkpad/comments/oacbc6/getting_error_specified_firmware_is_older_than/

Lenovo apparently refusing phone support. Does there happen to be a solution here?

LinuxOnTheDesktop, 2021-09-10:
It looks like this issue should be closed.

closed 2021-09-10 by hughsie