fwupd/firmware-lenovo #112: Numerous metadata issues with Lenovo firmware

brianredbeard, 2021-04-01:
Analyzing the firmware for [N22RM](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22RM.firmware) devices, the state of the information in `firmware.metainfo.xml` is confusing _at **best**_.   

Thus far I've performed introspection on the following firmware versions:

 - 184.83.3874
 - 184.82.3838
 - 184.79.3722
 - 184.55.3510

I stopped after checking these versions and moved over to checking the firmware for _other_ devices (N27ET, N22TF, N2CRN).  The usage of metadata across the matrix of devices and their respective versions is inconsistent and largely not useful to the end user.

As an example, I'll use `firmware.metainfo.xml` from  184.83.3874

```xml
[~]$ cabextract -p -F firmware.metainfo.xml 0797df92b7c525d930dedc94f276575f1ba815d43d5f2e5b0e70a11e3b5b17bf-Lenovo-ThinkPad-T480s-CorporateMEFirmware-11.8.83.3874.cab
<?xml version="1.0" encoding="UTF-8"?>
<component type="firmware">
  <id>com.lenovo.ThinkPadN22RM.firmware</id>
  <name>ThinkPad T480s</name>
  <summary>Lenovo ThinkPad T480s Corporate ME Firmware</summary>
  <description>
    <p>Lenovo ThinkPad T480s Corporate ME Firmware N22RM17W, 11.8.83.3874</p>
  </description>
  <provides>
    <firmware type="flashed">39a39e45-38ce-4b05-809b-60351bf88ec4</firmware>
  </provides>
  <url type="homepage">http://www.lenovo.com/</url>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>LicenseRef-proprietary</project_license>
  <developer_name>Lenovo Ltd.</developer_name>
  <categories>
    <category>X-CorporateManagementEngine</category>
    <category>X-ManagementEngine</category>
  </categories>
  <custom>
    <value key="LVFS::VersionFormat">triplet</value>
    <value key="LVFS::UpdateProtocol">org.uefi.capsule</value>
  </custom>
  <releases>
    <release version="184.83.3874" date="2016-07-08" urgency="high" tag="N22RM17W">
      <checksum type="sha1" filename="firmware.bin" target="content">2042b6e79b43a0af9792483e14e9536a35498b6f</checksum>
      <checksum type="sha256" filename="firmware.bin" target="content">f6f554f771dbfeedd83715506ade0148d992929fdd954949f41301a41af27c65</checksum>
      <description>
        <ul>
          <li>8 Intel Platform Update 2021.1</li>
        </ul>
        <p>Problem Fixes</p>
        <ul>
          <li>Mitigated the following security vulnerabilities:</li>
        </ul>
      </description>
      <url type="details">https://pcsupport.lenovo.com/de/en/search?query=N22RM17W</url>
      <size type="installed">7515520</size>
      <issues>
        <issue type="cve">CVE-2020-24588</issue>
        <issue type="cve">CVE-2020-24587</issue>
        <issue type="cve">CVE-2020-24586</issue>
      </issues>
    </release>
  </releases>
</component>
```

In this example the `date=` field is set to "2016-07-08".  This is actually the date for _every_ payload for N22RM devices.  As a user, I would expect this to reflect one of the following:
  - The binary artifact build date
  - The binary artifact release date
  - The advisory release date
  - The advisory discovery date

None of these seem to be the value used.  It should also be noted that the "release dates" above should be the _manufacturer_ release date, not the LVFS release date.  The LVFS release date is already exposed via the `fw.release_ts` property and is outside the scope of `firmware.metainfo.xml`.

This data is then exposed to the user as follows:

```
└─Intel Management Engine:
  │   Device ID:          f31a892e11e7c70aedbbb03610f376bb7cd2b2d4
  │   Current version:    184.82.3838
  │   Minimum Version:    184.82.3838
  │   Vendor:             DMI:LENOVO
  │   GUIDs:              39a39e45-38ce-4b05-809b-60351bf88ec4
  │                       e6e2469e-013d-564c-917a-5938b9277c49
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480s Corporate ME Update:
        New version:      184.83.3874
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T480s Corporate ME Firmware
        License:          Proprietary
        Size:             7.5 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N22RM17W
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:      
        • 8 Intel Platform Update 2021.1
        
        Problem Fixes
        
        • Mitigated the following security vulnerabilities:
```

As you can see, this becomes ambiguous because "created" would imply when the patch was created, which is obviously not the case.

The sole exception to this strategy appeared to be payloads for N2CRN devices.  For these devices the `date=` field for payloads was approximately one month before the LVFS release date (ergo I assume it is one of the values noted above).  This aids in informing the user that there is not a rollback attack underway (Note: this doesn't speak to the trustworthiness of the metadata, solely that there are not obvious errors.  Put differently: As a user, why should I expect that an update won't cause subsequent issues if insufficient QA was performed on the release of the firmware itself?)

Beyond this there are quite a few situations where the `tag=` is completely missing or has no information published via the URL in the metadata (like the example above) or where fields have not been filled out completely (like the description above).  All of these errors act in concert to erode user confidence in the application of security patches.

mrhpearson, 2021-04-01:
Hi,

Thanks for the report - I'll pass those details on to the firmware team. Realistically I don't think you're going to get all the already done releases fixed - but they can ensure things are correct for future releases.

As a note different firmware teams for each platform provide these updates - hence why you may see some differences between them (like the date field set correctly for N2CRN). Some teams are also more experienced at this than others.

For the tag= field; do you mean <release tag= >? My understanding is that's an optional tag and a nice to have - let me know if I'm missing something obvious (I'm somewhat new to this myself and still learning)

Mark

hughsie, 2021-04-02:
> My understanding is that's an optional tag and a nice to have

Yes and no. For some Lenovo firmware it's enforced as required (e.g. ME) and for other firmware it's nice-to-have (e.g. system BIOS). I think longer term it'll be required for all firmware as one of the pain-points of real users is matching up the Lenovo tag with the version number.

joakim-tjernlund, 2021-05-11:
One thing I miss in most Lenovo FW releases is the lack of change notes.

brianredbeard, 2021-09-22:
More grist for the mill

Just as an FYI, the `created` field still shows up as a date in the past.  I had forgotten that i mentioned that previously and came here to note it.  With the exception of seeing `Current version`:  `0.1.45` the lack of usable or actionable information provided in the information _about_ the update really had my hackles up about a potential roll-back attack. If the urgency is "high" but there's nothing telling me _why_ it's considered urgent that's a huge red flag.

```bash
20L8S2N800
│
└─System Firmware:
  │   Device ID:          e7cc234e82d08b6a14e246beb6b5f8142091e570
  │   Current version:    0.1.45
  │   Minimum Version:    0.0.1
  │   Vendor:             LENOVO (DMI:LENOVO)
  │   GUIDs:              ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │                       2b79e0a0-0801-5fc9-a7a5-90374e924a9b ← UEFI\RES_{EBFE8DF8-DEE7-4692-A721-CBCF5095C5CF}
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480s System Update:
        New version:      0.1.47
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T480s System Firmware
        License:          Proprietary
        Size:             9.5 MB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:      
        Lenovo ThinkPad T480s System Firmware
        
        The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
 
```