fwupd/firmware-lenovo #113: X390 fails to install Intel ME and System firmware updates

olbat, 2021-04-08:
I'm running into the same problem mentioned in this issue https://github.com/fwupd/firmware-lenovo-thinkpad/issues/31 ("Press key to continue" message on reboot and update not installed) when trying to install the following updates:
- _Intel Management Engine (192.47.1524 → 192.72.1757)_
- _System Firmware (0.1.51 → 0.1.71)_

My laptop runs an up-to-date Arch Linux, here are a few more details about my system:

```
$ fwupdmgr --version
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.6
daemon version:	1.5.8
```

```
$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • INTEL SSDPEKKF512G8L
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • Embedded Controller
________________________________________________

Devices that were not updated correctly:

 • Intel Management Engine (192.47.1524 → 192.72.1757)
 • System Firmware (0.1.51 → 0.1.71)
```

```
$ fwupdmgr get-devices
20Q0CTO1WW
│
├─Thunderbolt host controller:
│     Device ID:          1f274b5601b0724bba982610ca3523eb966cb6ee
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              ea3c7ac8-d937-568a-aab5-e053bd6af9ce
│                         3d3f953a-3408-56fa-81f3-042d73c59c16
│                         e773c51e-a20c-5b29-9f09-6bb0e0ef7560
│                         f1793149-6c12-5a7f-95db-9fefd116748d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          09ce7908b2d90a3b53473c3f328410590a527f29
│     Current version:    0.1.15
│     Minimum Version:    0.1.15
│     Vendor:             DMI:LENOVO
│     GUIDs:              be9a5f93-930b-40bf-8864-815268ea4ea3
│                         98129d04-2653-5233-89f1-8aa461e54a25
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─INTEL SSDPEKKF512G8L:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:            NVM Express Solid State Drive
│     Current version:    L11P
│     Vendor:             Intel Corporation (NVME:0x8086)
│     GUIDs:              f91a5c60-8696-539e-9d0a-57d194f74ac4
│                         396fb4fd-7656-564f-88b5-381e83ca30c0
│                         79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                         68db11e5-b0cf-5bc9-a94e-17e28496e505
│                         87fd9fec-f447-5ec6-a1cd-d55fac0d3d30
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          07edea1cf12d20814225b44de22b33758db1a1c8
│     Current version:    192.47.1524
│     Minimum Version:    192.47.1524
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUIDs:              ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
│                         056f0b06-1562-50f4-aea6-0d52f6fd4be5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          ca8e899e46f1da1776d16410bc92f2cf0d963433
│ │   Current version:    0.1.51
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              6300ba52-d9ea-4ca0-88f9-f89ef73cc07b
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       df8fecf2-0e94-5191-9b0b-cc3aee5da5c0
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341
│                         5971a208-da00-5fce-b5f5-1234342f9cf7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
└─UEFI Device Firmware:
      Device ID:          b1825e2b03a9ebdea247a74f1f3ddee2a21ac565
      Current version:    16842760
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      GUIDs:              24aac206-3921-45c0-a79e-0a607b11aef0
                          3a511605-fb00-56af-8a7f-1a6689d53ad2
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
________________________________________________

Devices that were not updated correctly:

 • Intel Management Engine (192.47.1524 → 192.72.1757)
 • System Firmware (0.1.51 → 0.1.71)
```

```
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-6300ba52-d9ea-4ca0-88f9-f89ef73cc07b-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ef5cdc85-9cf6-469d-9cb7-920b7dd6672b-0
```

```
$ find /boot/ -name 'fw*'
/boot/EFI/arch/fw
/boot/EFI/arch/fw/fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
/boot/EFI/arch/fw/fwupd-ef5cdc85-9cf6-469d-9cb7-920b7dd6672b.cap
/boot/EFI/arch/fw/fwupd-6300ba52-d9ea-4ca0-88f9-f89ef73cc07b.cap
```

Is there anything I can do to solve this issue?

kmauleon, 2021-05-07:
hi @olbat will close this issue as well for now... kindly reopen if anything... thank you very much

olbat, 2023-12-23:
It's been quite a while and the issue is still not resolved. Is there anything I can do to fix it?

kmauleon, 2023-12-27:
@olbat please try to install one firmware at a time.... 

update system firmware
reboot
update MEFW
reboot

olbat, 2024-03-16:
It worked, thank you so much 👍

kmauleon, 2024-03-18:
great!  closing this ticket...

closed 2024-03-18 by kmauleon