fwupd/firmware-lenovo #114: Lenovo T15 Gen I failed to update Intel Management Engine on Fedora 34

snaggen, 2021-04-13:
I got a notification about new firmware on my Lenovo Thinkpad T15 running Fedora 34, so I ran 'fwupdmgr update' to update it and then rebooted when I got asked. When the system got up, the update failed. 

fwupdmgr get-history shows:
`└─Intel Management Engine:
  │   Enhets-ID:          96631d5d4db1e7fb990a201e4f22ed300a4f9373
  │   Föregående version: 224.45.1389
  │   Uppdateringstillstånd:Misslyckades
  │   Uppdateringsfel:    failed to run update on reboot
  │   Senast ändrad:      2021-04-13 12:13
  │   GUID:               0d7baa94-7f7f-4896-b1d3-8a6c594ff635
  │   Enhetsflaggor:      • Intern enhet
  │                       • Uppdateringsbar
  │                       • Systemet kräver extern strömkälla
  │                       • Stöds på fjärrserver
  │                       • Kräver en omstart efter installation
  │                       • Enhet är oanvändbart under tiden som uppgraderingen pågår
  │ 
  └─ThinkPad T14 Gen 1/ P14s Gen 1/ T15 Gen 1/ P15s Gen 1/ T14 Gen 1 Healthcare Edition Consumer ME Update:
        Ny version:       224.48.1605
        Fjärrkälla-id:    lvfs
        Sammanfattning:   Lenovo ThinkPad T14 Gen 1/ P14s Gen 1/ T15 Gen 1/ P15s Gen 1/ T14 Gen 1 Healthcare Edition Consumer ME Firmware
        Licens:           Proprietär
        Storlek:          3,1 MB
        Skapad:           2021-04-09
        Urgency:          Hög
        Detaljer:         https://pcsupport.lenovo.com/de/en/search?query=N2XRN09W
        Tillverkare:      Lenovo Ltd.
        Flaggor:          is-upgrade
        Beskrivning:      
        • 0 Intel Platform Update
        
        Version 14.0.48.1605 (LVFS: 224.48.1605)
        
        Problem Fixes
        
        • Intel CSME IPU 2020.2:
        • Mitigated the following security vulnerabilities under issues.
`

hughsie, 2021-04-13:
What does `efibootmgr` say? Is "boot order lock" turned on in the BIOS?

snaggen, 2021-04-13:
$ efibootmgr
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022,0023,0002
Boot0000* Windows Boot Manager
Boot0001* Fedora
Boot0002* Linux-Firmware-Updater
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Regulatory Information
Boot0015  ThinkShield secure wipe
Boot0016  Startup Interrupt Menu
Boot0017  Rescue and Recovery
Boot0018  MEBx Hot Key
Boot0019* USB CD
Boot001A* USB FDD
Boot001B* NVMe0
Boot001C* NVMe1
Boot001D* ATA HDD0
Boot001E* ATA HDD1
Boot001F* USB HDD
Boot0020* PXE BOOT
Boot0021* LENOVO CLOUD
Boot0022  Other CD
Boot0023  Other HDD
Boot0024* USBR BOOT CDROM
Boot0025* USBR BOOT Floppy
Boot0026* ATA HDD
Boot0027* ATAPI CD

Boot Order Lock is disabled (just rebooted and checked to be sure). I have previously updated Intel ME using fwupdatemgr, and I also used it last week under F34 to successfully update the System Firmware.

snaggen, 2021-04-13:
Running efibootmgr right after an update attempt will have point to the Linux_Firmware-Updater: BootCurrent: 0002
I also noticed that the boot sequence displayed the grub menu this time, so it was clearly not running the usual Fedora boot. The update history still shows "failed to run update on reboot".

snaggen, 2021-06-06:
I see the same issue with the new System Firmware update. 
 • System Firmware (0.1.19 → 0.1.20)  downloads, reboot, but nothing is installed.

snaggen, 2021-06-06:
Could this MR be relevant, https://github.com/rhboot/shim/pull/374
I would be happy to test that PR if someone provided a rpm package for Fedora 34

snaggen, 2021-06-10:
I disabled secure boot and tested the update again, and now it worked. So this is most likely the shim issue above and not a problem with fwupd or the actual firmware.

closed 2021-06-10 by snaggen