fwupd/firmware-lenovo #116: Unable to update firmware on T580 running version 1.22

lasombra, 2021-04-30:
**Describe the bug**
The UEFI firmware 0.1.22 for my Lenovo Thinkpad T580 can't be updated to version 0.1.27. The following error message is returned:
```
modok :: ~ % sudo fwupdmgr install N27ET41W.cab
Decompressing…           [***************************************]
Not compatible with firmware version 0.1.22, requires >= 0.1.64
```
I don't understand how it requires a larger, unavailable firmware version though.

**Steps to Reproduce**
Install UEFI firmware 0.1.27 from downloaded from Lenovo's website or from https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN27ET.firmware.

**Expected behavior**
Firmware update gets prepared to be installed after a reboot.

**fwupd version information**
Please provide the version of the daemon and client.
```
modok :: ~ % fwupdmgr --version 
client version: 1.5.9
compile-time dependency versions
        gusb:   0.3.6

daemon version: 1.5.9
```

Please note how you installed it (apt, dnf, pacman, source, etc): dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.
```
modok :: ~ % fwupdmgr get-devices --show-all-devices
20LAS3NJ0S
│
├─Intel Management Engine:
│     Device ID:          b7dd95c2ffb9bc7b710575d6222c9eff415f0921
│     Current version:    184.83.3874
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     Last modified:      2021-04-30 08:42
│     GUIDs:              ae82a157-cc81-419a-88e4-a983a76075db
│                         a2bca6e6-43f2-5782-af9b-a957ea644511
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel(R) Core™ i7-8650U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000e0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         54bdc429-0179-5019-9cb6-8a343324c97c
│     Device Flags:       • Internal device
│
├─LENSE30512GMSP34MEAT3TA:
│     Device ID:          be29aadda98b637d33b6b740ff7b367f61886a31
│     Summary:            NVM Express Solid State Drive
│     Current version:    2.5.0412
│     Vendor:             Lenovo (NVME:0x17AA)
│     GUIDs:              c483fc41-ad0a-5336-8570-a3277e56e354
│                         9b4e3e73-ffd8-5440-b673-9558c98af843
│                         124c38ac-0100-5a50-aac8-89602d99769f
│                         70cc283d-bc6c-5b87-9977-17c5f95c3168
│                         0a2bdb12-23de-5728-829f-b1674e12dde0
[... 204 lines cut ...]
├── initramfs-0-rescue-524e0a63b3ea4350a8039e80705c5dac.img
├── initramfs-5.11.15-200.fc33.x86_64.img
├── initramfs-5.11.16-200.fc33.x86_64.img
├── initramfs-5.11.17-200.fc33.x86_64.img
├── loader
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.31
├── System.map-5.11.15-200.fc33.x86_64
├── System.map-5.11.16-200.fc33.x86_64
├── System.map-5.11.17-200.fc33.x86_64
├── vmlinuz-0-rescue-524e0a63b3ea4350a8039e80705c5dac
├── vmlinuz-5.11.15-200.fc33.x86_64
├── vmlinuz-5.11.16-200.fc33.x86_64
└── vmlinuz-5.11.17-200.fc33.x86_64

11 directories, 98 files
```
**Additional questions**
- Operating system and version: Fedora 33
- Have you tried rebooting? Yes
- Is this a regression? Yes, it worked until 1.22 was installed
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Lenovo system w/o Boot Order Lock

hughsie, 2021-04-30:
@lasombra I think the uploader made a mistake. @mrhpearson could you file an internal issue and get the requirement of https://fwupd.org/lvfs/firmware/6524 set to the correct value please. Thanks!

mrhpearson, 2021-04-30:
Internal ticket raised (For my reference LO-1174). A warning that both China and Japan have holidays right now so there might be a bit of a delay getting this fixed - sorry :(  
This is definitely an issue that could and should have been caught before moving the firmware to stable - I'll be triggering an investigation as to what happened because that's not acceptable.

hughsie, 2021-04-30:
> This is definitely an issue that could and should have been caught before moving the firmware to stable

Do you think the LVFS should have forbidden promoting it if the requirement was bigger than the firmware version?

hughsie, 2021-04-30:
> forbidden promoting it if the requirement was bigger

Answering my own question: https://gitlab.com/fwupd/lvfs-website/-/merge_requests/953

> because that's not acceptable

Well, it's not a disaster as we wouldn't have downloaded or offered the update as the requirements wouldn't have been fulfilled. It's just obviously never been tested which is the bit to get upset about I guess. Thanks!

mrhpearson, 2021-04-30:
Yeah - I'm in agreement  on all the above

mrhpearson, 2021-07-20:
I believe this issue can be closed - I'd completely missed the update from the FW team - but they've fixed the original issue and since uploaded a newer version that also appears correct.

closed 2021-07-20 by mrhpearson