fwupd/firmware-lenovo #117: T460s Dead during Firmware Update

Cameron-Cn, 2021-05-13:
Howdy!

(I can also file an error report elsewhere, this was the only location I could track down to make this sort of report. Sorry if this is the wrong spot!)

This is just to put in a notice for what I suspect is a firmware failure during update of a Lenovo ThinkPad T460s.

Running openSUSE Tumbleweed with GNOME 40, updated using a TTY with `sudo fwupdmgr update`. I have used this method in the past without issue, and I super appreciate the FWUPD devs for all the work.

The machine applied the first firmware update without issue, but on the second reboot it died and has since been unable to POST.

I wish I could provide more information, but the machine has zero responsiveness, even after removal of internal batteries (and CMOS).

hughsie, 2021-05-13:
> but on the second reboot it died and has since been unable to POST

Super uncool. Did you try removing all batteries (including internal ones and CMOS) for 30 minutes and then try again? Do you remember what firmware update it tried to apply, or how it failed? Or the old firmware version? When you say "died", do you mean the LEDs and fans are completely turned off? Did anything interrupt the update? Did you schedule two updates to be installed at the same time, and if so do you remember what they were? Any additional data at all would be great.  Thanks.

Cameron-Cn, 2021-05-13:
@hughsie 

The two updates (I am working from memory of GNOME software's display here, which I assume are the same as the `fwupdmgr` pair) were Intel Managment Engine and a "Firmware" (BIOS, I assume again).

They updated one at a time, I believe the BIOS was second as it displayed a LENOVO logo during the process whereas the first one had an "INTEL...WAITING" message before the reboot.

The old firmware version would have been current as of December 2020, the last time I update firmware on that machine.

"Dead" here means the machine no longer POSTs, although the LEDs on the keyboard and the power button will flare on after the power button is pressed. (Video available here: https://imgur.com/a/gkPZLJs)

I did nothing that I am aware of to interrupt the update, it was on a desk left for the update plugged in to power and ethernet, and when I checked on it the screen was off--I had thought maybe it turned itself off and didn't reboot correctly, but then it would not power on.

+ I did not try removing CMOS for 30 minutes--I will try that now and get back to you if the machine is able to POST/recover.

Hopefully it's something I did wrong! And can fix :)

hughsie, 2021-05-13:
@mrhpearson help!!

Cameron-Cn, 2021-05-13:
Update:

After removing the batteries (including CMOS) for 30 minutes, the machine is in the same state as before.

Steps:
1. Removed internal batteries, disconnected CMOS
2. Let sit for ~30 minutes
3. Reconnected CMOS, plugged in AC power
4. Attempted to boot (not successful)
5. Re-inserted batteries, attempted boot (again no success)

I should add now that this was merely my testing machine, so while I am sad to see it go it's not of dire importance.

I'm totally willing to perform all manner of tests or hardware tinkering (provided some guidance or YouTube video) if it could help triangulate an issue.

@hughsie 
Thanks for your time with this!

hughsie, 2021-05-13:
@mrhpearson if it helps, the latest T460s update went out 4 months ago, and has been deployed ~7k times, with 100.0% reported success -- although notable in cases like this the machine is _too dead_ to actually send a failure report.

Cameron-Cn, 2021-05-13:
It is entirely possible (I would say likely even) that this is a hardware failure of some sort, as I've never encountered an issue with `fwupdmgr`.

I am not the machine's original owner, so I am unsure how many paces it has been through. I believe the 6-series came out in 2016, and if it was a corporate machine it could well have been near EoL anyway.

markrages, 2021-05-14:
Me too

My T460s  has Ubuntu 18.04, which tried installing two "firmware"updates through the Ubuntu software store this afternoon. 

It is now bricked. The power button doesn't work. The "FnLk", "Mute" and power light are on.

I haven't attempted any mitigation or recovery efforts yet.

[image]

hughsie, 2021-05-15:
> This issue was fixed by the reset button through the hole in the bottom.

Still scary! @mrhpearson this shouldn't be required, right?

mrhpearson, 2021-05-21:
Sorry - only just became aware of this one :(
Absolutely - reset button shouldn't be needed. I've raised this with the FW team (LO-1426 for my reference) and will see what we can get you support. FW updates obviously shouldn't brick your device. Sorry!
Mark

lkiesow, 2021-06-28:
I can add to that: Same experience with the latest firmware update on my T470s: Stuck in a black screen after reboot. Reset button helped but obviously shouldn't be necessary. So, that seems to happen more often.

ChiWei-Chen, 2021-07-27:
Sorry for the late @Cameron-Cn . I'm working with @mrhpearson and got feedback from BIOS expert.
It seems that CMOS battery removal did not recover the system. So I think that motherboard(SPI ROM?) replacement will be required. Is it possible to reach [customer support](https://www.lenovo.com/us/en/contact) in your area to repair your system?
Thank you.

@lkiesow , the problem seems different from the system get bricked(dead).
Could you provide the instructions and firmware versions before and after the firmware update? The BIOS experts need this information for investigating. Thank you!
Chi-Wei

Cameron-Cn, 2021-07-29:
> Is it possible to reach [customer support](https://www.lenovo.com/us/en/contact) in your area to repair your system?
> Thank you.

The machine is not able to be seen by customer support (being far out of band now). I was able to trade it to a friend who does these sort of hardware repairs, who then fixed it (I can ask how) and sold it off.

In short, I no longer have the machine in question, having decided hardware issues of this sort were beyond my skill!

Thank you for your help!!

mrhpearson, 2021-10-20:
We were unable to reproduce so closing issue.

closed 2021-10-20 by mrhpearson