fwupd/firmware-lenovo #118: Firmware Capsule Is Not Getting Loaded on a Thinkpad T480

kendoori, 2021-05-18:
**Describe the bug**
A clear and concise description of what the bug is.

When I run fwupdmg I am told: "Upgrade available for UEFI Device Firmware." I then apply the update. When I reboot, it looks like the capsule is being loaded. I see this screen, and then it goes to the normal login screen

I keep getting prompted to reapply this update.

**Steps to Reproduce**
Steps to reproduce the behavior.

sudo fwupdmgr update

`Upgrade available for UEFI Device Firmware from 184.79.3722 to 184.83.3874
20L50011US must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]`

[image]

If I respond Yes and reboot, the screen displays "found update, adding new capsule and then boots into the system

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

After applying update, the same update shouldn't be presented again.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
`client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
`

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
`fwupdmgr get-devices --show-all-devices
20L50011US
│
├─Thunderbolt Controller:
│     Device ID:           36cf863540d97a0506a091d24cf4f1c5d848183e
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               7d131734-64ed-5865-a46e-a67cfb9b5a46
│                          e45313fd-8d5c-54e9-b175-1f15ffe6b05c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.83.3874
│     Bootloader Version:  11.8.83.3874
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
[... 464 lines cut ...]
├── initrd.img-5.8.0-53-lowlatency
├── initrd.img.old -> initrd.img-5.4.0-73-generic
├── System.map-5.4.0-26-generic
├── System.map-5.4.0-73-generic
├── System.map-5.4.0-73-lowlatency
├── System.map-5.8.0-50-generic
├── System.map-5.8.0-50-lowlatency
├── System.map-5.8.0-53-lowlatency
├── vmlinuz -> vmlinuz-5.8.0-53-lowlatency
├── vmlinuz-5.4.0-26-generic
├── vmlinuz-5.4.0-73-generic
├── vmlinuz-5.4.0-73-lowlatency
├── vmlinuz-5.8.0-50-generic
├── vmlinuz-5.8.0-50-lowlatency
├── vmlinuz-5.8.0-53-lowlatency
└── vmlinuz.old -> vmlinuz-5.4.0-73-generic
`

**Additional questions**
- Operating system and version: Mint 20.1 Cinnamon
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? No
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Disabled

mrhpearson, 2021-05-21:
I've raised a ticket with the FW team (LO-1247 for my reference). Agreed with Mario that it looks like a firmware issue - I'll get some feedback as to why the update isn't taking place.

lijunwang1990, 2021-05-27:
Hi @kendoori ,could you please try download the package form LVFS site.  and use below command to update again?  if you still can not update, we will upload 11.8.86.3909 recently. you can try ro update to this version directly. thank you.

fwupdmgr --allow-older --allow-reinstall install  [Name].cab
https://fwupd.org/lvfs/firmware/6718

ChiWei-Chen, 2021-05-27:
Thanks, @lijunwang1990.
Hi @kendoori, if you cannot access the link provide by Lijunwang, please try this one, thanks!
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24RM.firmware

kendoori, 2021-05-27:
I downloaded the firmware and updated as suggested. After reboot, I ran: 
`
[image]
`
The same firmware update is being offered.

lijunwang1990, 2021-06-01:
Hi @kendoori , can you check your ME version under setup? you can press F1 before boot to OS ,the ME version will be displayed on the main page. and use "FWupdmgr get-devices" to check the ME version under OS.

kendoori, 2021-06-01:
`fwupdmgr get-devices
20L50011US
│
├─Thunderbolt Controller:
│     Device ID:           36cf863540d97a0506a091d24cf4f1c5d848183e
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               7d131734-64ed-5865-a46e-a67cfb9b5a46
│                          e45313fd-8d5c-54e9-b175-1f15ffe6b05c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─SanDisk SDSSDXPS960G:
│     Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:             ATA Drive
│     Current version:     X212T0RL
│     GUIDs:               22be2cc6-990f-583a-98d3-f89e332bcb93
│                          bc497ac3-efa7-5d3a-98ca-7777bb59fad1
│                          a511a2d6-df8d-5291-90cc-49dc6ab6fe76
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           48477ba14bd9bd0c97323a07caa54b7766f917c2
│     Current version:     0.1.36
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           5decfff65e3391259bca4003a090e59bf50e3bea
│     Current version:     184.79.3722
│     Minimum Version:     184.79.3722
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                dd31d983-cf3b-4c84-acc9-70caa94e827b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           c19580db344e418cf2b23297bcf46d75c06e432a
│     Current version:     0.1.21
│     Minimum Version:     0.1.21
│     Vendor:              DMI:LENOVO
│     GUID:                b87a926d-189c-49a1-b1ff-921099b1de89
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           b1825e2b03a9ebdea247a74f1f3ddee2a21ac565
│     Current version:     1.3.1084
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                24aac206-3921-45c0-a79e-0a607b11aef0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           a567cff02396e1038d53d7433ef3dbcd9558aebe
│     Current version:     1.1.5
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                d4583ddb-bf0a-4801-aefd-951930545cb9
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           82bae8448b308a038b50466dd710c0cd237e3464
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.01_B0019
      Bootloader Version:  BOT01.02_B0014
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
`

lijunwang1990, 2021-06-03:
Hi @kendoori , sorry, I also need check the MEFW version in setup. please reboot the system and keep pressing "F1" before Lenovo Logo displayed. you will enter into setup menu.like this:
[image]

kendoori, 2021-06-03:
[image]

lijunwang1990, 2021-06-08:
Hi @kendoori ,someone fixed this issue use below commond, please have a try:

$ sudo fwupdmgr downgrade
This should give you a list of your recent updates. Just downgrade the looping updates. Don't worry though. Once you've downgraded, just go ahead and run
$ sudo fwupdmgr update
And that should be the end of your troubles. Or at least it was the end of my troubles. No more updates which can't be installed. fwupdmgr --help for other options.

kendoori, 2021-06-08:
I followed the procedure, but after downgrading, and running update, I still get this
[image]

If I try to trigger a downgrade again, it says none are available.
[image]

ChiWei-Chen, 2021-11-26:
Hello @kendoori , I worked with @lijunwang1990 and @mrhpearson , I got a trial BIOS from them. If you still encounter this problem and want to try this BIOS, could you drop me a mail to "cchen50 @lenovo.com"? (without space)
I will send you a link to download it.
Thank you

ChiWei-Chen, 2021-11-30:
@kendoori Thanks for the mail reply.
As discussed in the mail, the system updated ME successfully with the trial BIOS, could you provide output of “fwupdmgr get-devices” for @lijunwang1990 's reference? 
Thanks

kendoori, 2021-11-30:
`fwupdmgr get-devices
20L50011US
│
├─Embedded Controller:
│     Device ID:          c19580db344e418cf2b23297bcf46d75c06e432a
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     GUIDs:              b87a926d-189c-49a1-b1ff-921099b1de89
│                         646e8b00-c980-5327-b889-b518b0a1cc1c
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          5decfff65e3391259bca4003a090e59bf50e3bea
│     Current version:    184.79.3722
│     Minimum Version:    184.79.3722
│     Vendor:             DMI:LENOVO
│     GUIDs:              dd31d983-cf3b-4c84-acc9-70caa94e827b
│                         49b00a7c-b9ac-5782-aaf9-76d3aa53f003
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─SanDisk SDSSDXPS960G:
│     Device ID:          64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:            ATA Drive
│     Current version:    X212T0RL
│     Vendor:             SanDisk (ATA:0x15B7, OUI:001b44)
│     GUIDs:              22be2cc6-990f-583a-98d3-f89e332bcb93
│                         bc497ac3-efa7-5d3a-98ca-7777bb59fad1
│                         a511a2d6-df8d-5291-90cc-49dc6ab6fe76
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          48477ba14bd9bd0c97323a07caa54b7766f917c2
│ │   Current version:    0.1.41
│ │   Minimum Version:    0.0.1
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   GUIDs:              dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       7924fa68-3277-5fd5-aee1-ce5985707c3b
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  238
│       Minimum Version:  238
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341
│                         5971a208-da00-5fce-b5f5-1234342f9cf7
[... 52 lines cut ...]
│                         279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:       • Updatable
│                         • Supported on remote server
│   
└─WDC PC SN520 SDAPMUW-128G-1001:
      Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
      Summary:            NVM Express Solid State Drive
      Current version:    20370001
      Vendor:             Sandisk Corp (NVME:0x15B7)
      GUIDs:              d03f7b57-df39-578a-80d2-df439b4682ca
                          4581e034-bf68-5440-9445-c4b251ba45b5
                          c0ff0f18-855d-5415-af06-a36723074fc9
                          c07858fa-63ab-586e-aec8-b5da8d637b21
                          c6efcf90-660c-5b41-9ac5-721c8dc4f4d6
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
Idle…                    [***************************************]
Update failure is a known issue, visit this URL for more information: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
Successfully uploaded 1 report
`

ChiWei-Chen, 2021-12-02:
@superm1 , BIOS replied BIOS code will compare the GUID first. If the GUID is the same as MEFW(Vpro/Nonvpro), BIOS will update the new version. But they foun an excrescence GUID on this @kendoori 's log. Normally there should be only one GUID.
Do you know where the excrescence GUID comes from? Could we have your advice is there something we can check in fwupd?
Thanks.

ChiWei-Chen, 2022-02-18:
@superm1 , could we have your advice/clarification on the comment from @lijunwang1990?

ChiWei-Chen, 2022-04-28:
Hi @kendoori 

I have a trail BIOS and ME for T480. If you're willing to test that it would be appreciated - send me or @mrhpearson  an email. Thanks!
(cchen50 at lenovo dot com) (mpearson at lenovo dot com)

Installation will be by fwupdmgr and it will involve updating the BIOS and then trying a ME FW update.

Thanks

lijunwang1990, 2022-04-28:
您好，邮件已收到

ChiWei-Chen, 2022-08-23:
Hi @kendoori , New BIOS for T480 is released, could you give it a try and let me know how it works? Thanks!
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24ETXXP.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24ETXXW.firmware

lijunwang1990, 2022-08-23:
您好，邮件已收到

ChiWei-Chen, 2022-10-11:
This fix already verified pass in #140.

closed 2022-10-11 by ChiWei-Chen

lijunwang1990, 2022-10-11:
您好，邮件已收到