fwupd/firmware-lenovo #119: ACPI errors when booting ThinkPad T14 Gen 1 (20S0000FMX)

joakim-tjernlund, 2021-05-19:
```
[    3.593344] ACPI Error: Needed [Integer/String/Buffer], found [Package] 000000004efcfed4 (20200925/exresop-469)
[    3.593349] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [OpcodeName unavailable] (20200925/dswexec-431)
[    3.593351] ACPI Error: Aborting method \ADBG due to previous error (AE_AML_OPERAND_TYPE) (20200925/psparse-529)
[    3.593355] ACPI Error: Aborting method \_SB.HIDD._DSM due to previous error (AE_AML_OPERAND_TYPE) (20200925/psparse-529)
[    3.593361] ACPI: \_SB_.HIDD: failed to evaluate _DSM (0x3003)
```

I see this error on several Lenovo laptops but this one is what I have handy ATM.
Ideally the Lenovo FW should be fixed to address this error.

hughsie, 2021-05-19:
This really isn't the place for general Lenovo firmware problems I'm afraid. If we started collecting these kind of issues here it would make actual firmware update problems harder to track. Sorry.

closed 2021-05-19 by hughsie

joakim-tjernlund, 2021-05-19:
Sure, I get it.

AlMa0r, 2023-08-04:
@hughsie Similar issue here on a fully upgraded Lenovo ThinkPad T14s Gen 1 type 20T1 (in particular, not the one above) with Intel Core i7-10610U. The BIOS version is 1.26, released 12/14/2022. `fwupdmgr` doesn't suggest anything to update/upgrade. The journal contains the following:
```
Aug 04 00:34:14 MyPC kernel: ACPI Error: Needed [Integer/String/Buffer], found [Package] 000000007577f2c1 (20220331/exresop-469)
Aug 04 00:34:14 MyPC kernel: ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [OpcodeName unavailable] (20220331/dswexec-431)
Aug 04 00:34:14 MyPC kernel: ACPI Error: Aborting method \ADBG due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
Aug 04 00:34:14 MyPC kernel: ACPI Error: Aborting method \_SB.HIDD._DSM due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
Aug 04 00:34:14 MyPC kernel: ACPI: \_SB_.HIDD: failed to evaluate _DSM b356ecee-4244-8f40-a792-4edd4d758054 (0x3003)
```
The first four lines appear red in the journal, and the fifth line appears orange. A Debian guy in http://bugs.debian.org/1026965 suggested we report here at https://github.com/fwupd/firmware-lenovo .  Please consider reopening.