fwupd/firmware-lenovo #121: Black screen after starting `fwupdx64.efi` on Lenovo P70 (Arch Linux)

sedrubal, 2021-05-20:
#### Describe the bug
When I start a firmware update (Intel Management Engine) with `fwupdx64.efi`, and reboot into the UEFI updater, it results in a black screen. When I press the power button of my laptop I can hear a beep code (long long short short short). I guess it is signaling, that I should not interrupt the process. The fans are ventilating. After waiting some minutes, nothing happens. When I restart using <kbd>Ctrl</kbd> + <kbd>Alt</kbd> + <kbd>Backspace</kbd> and run `fwupmgr` it tells me, that the updater did not run.

#### Steps to Reproduce
- Use a Lenovo P70 (20ER003SGE) with Arch Linux
- `fwupdmgr update`
- update `Intel Management Engine` from `184.80.3746` to `184.83.3874`
- press <kbd>y</kbd> when it asks to reboot
- After rebooting, you can read `Resetting`
- Then the device reboots again, fans turn on and the screen is black

#### Expected behavior
At this point I expect the fwupmgr UEFI updater to display a progress bar on the screen (as it did on previous updates) and the Intel ME gets updated. When it is done, the device should reboot into the system.

#### fwupd version information

<details>
<summary><code>$ fwupdmgr --version</code></summary>
<pre>
$ fwupdmgr --version
client version: 1.6.0
compile-time dependency versions
        gusb:   0.3.6

daemon version: 1.6.0
</pre>
</details>

Source: `pacman`

#### fwupd device information

<details>
<summary><code>$ fwupdmgr get-devices --show-all-devices</code></summary>
<pre>
$ fwupdmgr get-devices --show-all-devices
20ER003SGE
│
├─GM107GLM [Quadro M600M]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a2
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              d1079796-02ba-50d9-a8ce-c76519d255b5
│                         768699a7-ad1c-5ce2-9c6b-ae097ba2229c
│                         34e50bcd-a9d7-5468-b2ee-e77de378dee7
│                         1ff39d65-6601-5e0e-8f67-355725c82003
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─HD Graphics 530:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    06
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              84a4dbbc-4a5f-57e7-980a-775bfb35ba11
│                         49ab06cd-bdd1-5e47-a843-9ace3766777d
│                         fb0e4398-241c-5c36-ab78-398aa4ed66e5
│                         be415cfb-b50d-5724-9483-7a56f16d0d70
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─INTEL SSDSCKKF256H6L:
│     Device ID:          5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:            ATA Drive
│     Current version:    LBFL11P
│     Vendor:             Intel (ATA:0x8086|OUI:5cd2e4)
│     GUIDs:              a6af6710-9837-5f0d-8ca2-405f259ce5f6
│                         38bae1a4-d6a2-5ab4-84a2-d90d93084106
│                         983c4c4a-a661-57ac-a5a4-a8bb8b37b2ed
│     Device Flags:       • Internal device
│                         • Updatable
[... 484 lines cut ...]
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── vmlinuz-linux
└── vmlinuz-linux-lts

11 directories, 341 files
</pre>
</details>

**Additional questions**
- Operating system and version: **Arch Linux (up to date)**
- Have you tried rebooting? **Yes**
- Is this a regression? **?**
- Are you using an NVMe disk? **No**
- Is secure boot enabled? **No** (/ I tested it with both)
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? I'll check soon and update this information.

kmauleon, 2021-05-21:
hi @sedrubal can you please send screenshot of BIOS settings... also please post result of below command... thank you very much...
**fwupdmgr get-devices**

sedrubal, 2021-05-21:
I updated the description above and can confirm, that `Boot Order Lock` is turned off. Further BIOS settings can be found here:

[image]
[image]

I already posted the output of `fwupdmgr get-devices --show-all-devices` above, but here again without `--show-all-devices`:

```
$ fwupdmgr get-devices
20ER003SGE
│
├─INTEL SSDSCKKF256H6L:
│     Device ID:          5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:            ATA Drive
│     Current version:    LBFL11P
│     Vendor:             Intel (ATA:0x8086|OUI:5cd2e4)
│     GUIDs:              a6af6710-9837-5f0d-8ca2-405f259ce5f6
│                         38bae1a4-d6a2-5ab4-84a2-d90d93084106
│                         983c4c4a-a661-57ac-a5a4-a8bb8b37b2ed
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          118ec9ef6f590ae68fae0026096cb21882481627
│     Current version:    184.80.3746
│     Minimum Version:    184.80.3746
│     Vendor:             DMI:LENOVO
│     Update State:       Transient failure
│     Update Error:       Secure boot is enabled, but shim isn't installed to the EFI system partition
│     Last modified:      2021-05-21 09:10
│     GUIDs:              9b0578cf-f035-4fe8-b6ae-42591c946c36
│                         d7b47f59-ab20-51b8-9261-d68741f61e09
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Samsung SSD 850 EVO 500GB:
│     Device ID:          64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:            ATA Drive
│     Current version:    EMT02B6Q
│     Vendor:             Samsung (ATA:0x144D|OUI:002538)
│     GUIDs:              4f91c57f-0080-5f61-95b8-c54899cc39cb
│                         086811eb-c412-5a0b-a319-7c37d3dd3e46
│                         967c50bf-25a0-5394-b21d-5939967dade6
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:          584bb826ddd562b072efe306596b05d0ae9a29f5
│     Current version:    0.2.32
│     Minimum Version:    0.2.19
│     Vendor:             LENOVO (DMI:LENOVO)
│     GUIDs:              df91783d-a03d-485a-9a8e-a3f39067f18e
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         603eb043-4f1d-5538-9d3a-9923b4d07b34
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          422d67cf5a4e6aeea9f92945ea2aafc1cbe6a317
      Current version:    65553
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      GUIDs:              64f10525-b16b-4fd6-a7e3-6e04ad32b64a
                          0ece6108-781e-5b2a-be3d-c00805f787a4
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

kmauleon, 2021-05-21:
hi @sedrubal can you please try following settings as attached images for _SecureBoot and Startup_.
[image]
[image]

Then please try to execute the following again...
**fwupdmgr --force refresh
fwupdmgr update**

sedrubal, 2021-05-21:
I disabled Secureboot before for testing and it did not change anything. Now I additionally enabled "Legacy Boot" and "CSM Support" and after running `fwupdmgr --force refresh && fwupdmgr update` `fwupdx64.efi` started successfully and I could see the update screen as expected. After ~1 minute the system rebooted - everything as expected.

But when I run `fwupdmgr` again, it still says `Upgrade available for Intel Management Engine from 184.80.3746 to 184.83.3874`. So it seems that the update did not happen. I repeated the step about 4 times but the result was the same.

Apart from the non-functional Intel ME update I'm wondering if fwup requires `CSM Support` to be turned on. As soon as I enable secure boot, the BIOS is setting `UEFI/Legacy Boot` to `UEFI only` and `CSM Support` to `No`. AFAIK fwup works with secure boot as long as `fwupdx64.efi` is properly signed. But why did the update screen appear, when I enabled `CSM Support`?

kmauleon, 2021-05-24:
hi @sedrubal sorry I can not answer your question as I am not into BIOS and fwupd technicalities.  Maybe @hughsie can help to answer... BTW, what's the ME version in BIOS now?  I am asking because there is an existing issue in BIOS that ME is already updated in BIOS but fwupd still detects lower version thus prompting the user to update again.  Please advise.  Thank you very much...

sedrubal, 2021-05-25:
Thank @kmauleon the BIOS displays `11.8.83.3874` which I guess is equivalent to `184.83.3874`, which is the version, fwupd tries to install. `11.8.83.3874` is also the latest Intel ME version I can find on the support website of Lenovo. So it seems, the upgrade was successful and I have a workaround for this issue.

closed 2021-05-25 by sedrubal

kmauleon, 2021-05-26:
hi @sedrubal thank you very much for the confirmation and you are most welcome!

sedrubal, 2021-05-29:
@kmauleon just want to note one thing: Using [mei-amt-check](https://github.com/mjg59/mei-amt-check) (which is querying the ME version via `/dev/mei0`) gives me the right version `11.8.83`, while `fwupdmgr get-devices` gives me `184.80.3746`.

kmauleon, 2021-05-31:
hi @sedrubal this is a known issue and our BIOS DEV can't debug without seeing the content of ESRT table.  I can't get anyone to run a tool on their system... so i'm not sure how we can resolve the issue.  thank you very much for the information

sedrubal, 2021-06-06:
@kmauleon if you tell me the issue and the tool, I have to run, I can maybe provide some information. This however is the content of `/sys/firmware/efi/esrt/`:

| Entry  | fw_class                             | lowest_supported_fw_version | last_attempt_version | last_attempt_status | fw_type | fw_version | capsule_flags |
|--------|--------------------------------------|-----------------------------|----------------------|---------------------|---------|------------|---------------|
| entry0 | df91783d-a03d-485a-9a8e-a3f39067f18e | 131091                      | 0                    | 0                   | 1       | 131104     | 0x0           |
| entry1 | 9b0578cf-f035-4fe8-b6ae-42591c946c36 | 3092254370                  | 3092254370           | 0                   | 2       | 3092254370 | 0x8010        |
| entry2 | 64f10525-b16b-4fd6-a7e3-6e04ad32b64a | 1                           | 0                    | 0                   | 2       | 65553      | 0x0           |

kmauleon, 2021-06-07:
hi @sedrubal thank you very much for your support... as I am not allowed to share the tool publicly, can you please contact me thru my lenovo email kmauleon@lenovo.com  ... thank you very much for your help in advanced...