fwupd/firmware-lenovo #128: Unable to update Thinkpad T14s AMD firmware: "failed to write data to efivarfs: Error writing to file descriptor: Invalid argument"

woozong, 2021-06-06:
Running Fedora 34 here. When trying to update the system firmware of my T14s AMD, I get the following error: 
> failed to write data to efivarfs: Error writing to file descriptor: Invalid argument
The report looks like this: 
`{
  "ReportVersion": 2,
  "MachineId": "a9fc92364ac5240804bba0ebaa9c4b9cc9ecc95d26887a4a62c1b453fd890e72",
  "Metadata": {
    "DistroId": "fedora",
    "DistroVersion": "34",
    "DistroVariant": "workstation"
  },
  "Reports": [
    {
      "Checksum": "6b8254494ab25c7ea12c214f71fedc16bc047b91",
      "UpdateState": 3,
      "UpdateError": "failed to write data to efivarfs: Error writing to file descriptor: Invalid argument",
      "Guid": [
        "428ba8ba-40c8-4446-8d7a-394d6c28b26a"
      ],
      "Plugin": "uefi_capsule",
      "VersionOld": "0.1.5",
      "VersionNew": "0.1.32",
      "Flags": 574619947,
      "Created": 1622983724,
      "Modified": 1622984968,
      "Metadata": {
        "HostSku": "LENOVO_MT_20UH_BU_Think_FM_ThinkPad T14s Gen 1",
        "BootMgrDesc": "legacy",
        "RuntimeVersion(org.freedesktop.gusb)": "0.3.7",
        "CompileVersion(com.redhat.fwupdate)": "12",
        "CompileVersion(org.freedesktop.fwupd)": "1.5.9",
        "SecureBoot": "Disabled",
        "LinuxLockdown": "none",
        "EspPath": "/boot/efi",
        "UEFIUXCapsule": "Enabled",
        "DistroVersion": "34",
        "RuntimeVersion(org.kernel)": "5.12.8-300.fc34.x86_64",
        "MissingCapsuleHeader": "False",
        "CompileVersion(org.freedesktop.gusb)": "0.3.6",
        "RuntimeVersion(com.dell.libsmbios)": "2.4",
        "DistroId": "fedora",
        "CpuArchitecture": "x86_64",
        "BootTime": "1622983687",
        "HostVendor": "LENOVO",
        "EfivarNvramUsed": "135582",
        "RuntimeVersion(org.freedesktop.appstream-glib)": "0.7.14",
        "HostProduct": "20UHCTO1WW",
        "RuntimeVersion(com.redhat.fwupdate)": "12",
        "DistroVariant": "workstation",
        "HostFamily": "ThinkPad T14s Gen 1",
        "RuntimeVersion(org.freedesktop.fwupd)": "1.5.9",
        "KernelVersion": "5.12.8-300.fc34.x86_64"
      }
    }
  ]
}`

Let me know if I can provide additional information to help debug this.

hughsie, 2021-06-06:
@superm1 do you think this could be a case of SPI-out-of-space? It's a weird one.

superm1, 2021-06-08:
Being Lenovo, the first thing to check - is Boot Order lock enabled? 

BTW @mrhpearson can the `think-lmi` driver merging in 5.14 tell whether it's enabled or not?  If so, I think we really want a plugin or this in fwupd.