fwupd/firmware-lenovo #130: ThinkBook 15-IIL 20SM - UEFI Update DJCN25WW

CyberDomovoy, 2021-06-10:
**Describe the bug**
After trying to update UEFI using fwupdate and rebooting computer:
. fwupd seemed to run, but very quickly, didn't have time to read anything.
. Computer rebooted
. Computer won't boot
It stays stuck on the Lenovo logo.
Trying to get in UEFI settings leads to a black screen.
Trying to get to boot menu leads to a black screen.
Using the "Novo button hole" leads to a menu, where the only working item is "Lenovo Diagnostics".

You can see my report on the Lenovo forums [here](https://forums.lenovo.com/t5/ThinkBook-Notebooks/ThinkBook-15-IIL-20SM-BIOS-EFI-Update-DJCN25WW-No-POST-stuck-on-logo/m-p/5082867).

**Steps to Reproduce**
. Download UEFI update from Lenovo: [djcn25ww.exe](https://download.lenovo.com/consumer/mobiles/djcn25ww.exe)
. Extract firmware using innoextract:
```shell
innoextract -e djcn25ww.exe
```
. Prepare firmware for update with fwupdate:
```shell
/usr/lib/fwupd/fwupdate -a <UEFI_GUID> <path_to_bios.cap>
```
. Reboot

**Expected behavior**
After reboot, fwupd should update the UEFI, then reboot, and the computer should boot the OS as usual, with the updated UEFI.

**fwupd version information**
Since the machine won't boot, i can't run these commands.
But the installed OS was LMDE4.
On another machine with same OS:

Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
```
client version:	1.2.13
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.13
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
```shell
apt install fwupd
```
I had some errors about some tpm2 commands not being found, fixed with:
```shell
apt install -t testing tpm2-tools
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
Since i can't boot that device, i'm unable to provide that information.

**System UEFI configuration**
Please provide the output of the following commands:
```shell
$ efibootmgr -v
```
Since i can't boot that device, i'm unable to provide that information.
I _can_ tell you that i checked that fwupd was setup as UEFI boot for next reboot before i rebooted.

```shell
$ efivar -l | grep fw
```
Since i can't boot that device, i'm unable to provide that information.

```shell
$ tree /boot
```
Since i can't boot that device, i'm unable to provide that information.
I _can_ tell you that fwupd UEFI tool was present at the right place, same thing for the firmware .cap.

**Additional questions**
- Operating system and version: Linux Mint Debian Edition 4
- Have you tried rebooting? Not applicable
- Is this a regression? Unknown, first time i try fwupd
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Unknown, that option is nowhere to be found in the UEFI settings (as far as i can remember, since i can't go there anymore).

I am well aware that the procedure i describe here is not the _usual_ way to update the UEFI. But since Lenovo only distribute that update as a windows executable, as far i know it is the only way to apply such updates under linux.

Since that machine is still under warranty, it is going to a Lenovo repair center.
When i'll get it back, i'll be happy to provide you with any information you need.
Although, i won't try updating that UEFI again unless i'm confident it would go without trouble (i need that computer for work, and somehow i doubt Lenovo will accept me sending it back to the repair center for the same reason i just did).

CyberDomovoy, 2021-06-10:
> Well, actually they upload firmware to the LVFS. I'll reassign your bug and see what Mark says.

Yes, but that one is not on LVFS, unless i missed it.

mrhpearson, 2021-06-10:
I don't believe the Thinkbook 15 is Linux certified - so no LVFS firmware updates I'm afraid. Looks like they don't have the 'bootable CD' option on the support site which is often a good alternative :(

I've seen the trick referenced of extracting the UEFI capsule before - and it's neat - but never tried it myself and from a Lenovo point of view it's not supported so I'll never get anybody on the Lenovo firmware team to look into it. I will see if I can find the Thinkbook team and if they have any suggestions/alternatives.

Mark

CyberDomovoy, 2021-06-10:
> I will see if I can find the Thinkbook team and if they have any suggestions/alternatives.

Nice, thanks for your efforts.

thefoxie, 2021-06-18:
I've bought thinkbook 14 recently because it was the only laptop with ryzen without nvidia gpu available at the moment. Wanted maxxed out T14s gen2 or even gen1, but I couldn't find any. I don't use laptop often and when I go to office I plug it into thinkpad dock so it doesn't even matter and thinkbook is fine. It'd be nice if they supported it in lvfs.

mrhpearson, 2021-07-29:
I've started some conversations - but it's going to be a slow burner.

I'm going to close the bug as there isn't a solution until the Thinkbooks are in our Linux certification program. Let me know if any objections

closed 2021-07-29 by mrhpearson

mrhpearson, 2022-02-03:
Hi - no idea on why it broke I'm afraid.

I'm not allowed to comment on future products but hopefully the Thinkbooks get added in the future - right now none of them are listed. 
The Linux program is growing a little every year but it really happens by customer demand. Once the product teams start seeing customer demand they come to my team to understand how they can deliver Linux - so when you get your customer survey make sure it's listed that really you wanted Linux support for that system ;)

For AMD platforms - the T and P series all have LVFS support.

Mark