fwupd/firmware-lenovo #131: 1.32 BIOS update for T14s not successfull, falls back to 1.29

GITMichaelK, 2021-06-12:
On my T14s AMD I run fwupd to get 1.32 before I've read this here. These have been the results:

> $ fwupdmgr update
> Devices with no available firmware updates:
>  • Integrated Camera
>  • Samsung SSD 970 EVO Plus 2TB
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
> Devices with the latest available firmware version:
>  • Prometheus
> Upgrade available for System Firmware from 0.1.29 to 0.1.32
> 20UH001QGE must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
> Herunterladen …          [***************************************] Weniger als eine Minute verbleiben…
> Entpacken …              [***************************************]
> Authentifizierung …      [***************************************]
> Authentifizierung …      [***************************************]
> System Firmware wird aktualisiert …******************************]
> Einplanen …              [***************************************]
> Successfully installed firmware
>  • UEFI dbx
> 
> A Re-Boot is necessary ...

I did the Re-Boot: It produces a black screen, ventilation running at hight speed. After more than five minutes without any action a re-boot was forced by power button.  
After that: 

> fwupdmgr get-devices  
> Devices NOT actualized:  
>  • System Firmware (0.1.29 → 0.1.32)
> 

The T14s runs, but during start-up t**he red LENOVO-flag appears twice and takes a long time**, esp. the second one. **Is there any way to get rid of this behavior?**

michaellass, 2021-06-13:
Also see #129. This update tends to fail and even brick devices.

mrhpearson, 2021-06-14:
I've been trying to collate all the failures for this (there are too many - it's bad).
Do you have secure boot enabled? 
Thanks
Mark

GITMichaelK, 2021-06-14:
> Do you have secure boot enabled?
Definetly NOT!
Thanks for you work!

ozls, 2021-06-21:
My Thinkpad T14 Gen 1 AMD was just completely bricked by the 1.32 BIOS update (from 1.30 as far as I know). Same as @GITMichaelK, black screen and ventilation going strong, except mine was not able to fall back when I turned it off and I had to send it to Lenovo for repair. I think it's the same issue. What can I do to help? What information is needed? Moreover, can anyone tell me what to do to prevent this kind of behaviour in the future? Was I given a choice I would have prefered the update fail rather than this happen...

hughsie, 2021-06-21:
@ozls I'm pulled all the T14 updates now. Lenovo are investigating.

j-silver, 2021-06-21:
Has Lenovo withdrawn this update? It's no longer available.

hughsie, 2021-06-22:
> Has Lenovo withdrawn this update

I demoted them back to testing.

andersonjwan, 2021-06-25:
The unsuccessful update happened to me as well; however, I went into the BIOS, disabled secure boot, and tried again, and it seemed to work.

The downside is that I am noticing more issues within my laptop generally with more kernel errors and docking station port issues than before. Seeing that the BIOS v0.1.32 has been pulled from fwupd.org, should I consider downgrading back to v0.1.30? Is downgrading even a reliable and safe option?

michaellass, 2021-06-25:
> The downside is that I am noticing more issues within my laptop generally with more kernel errors and docking station port issues than before. Seeing that the BIOS v0.1.32 has been pulled from fwupd.org, should I consider downgrading back to v0.1.30? Is downgrading even a reliable and safe option?

I'm only a user, so take my advice with caution. Before downgrading I would attempt cutting the power and using the reset/recovery hole on the bottom on the machine. Mine was in some weird state after the update as well, presumably with some parts of the hardware stuck. After the full reset using that little hole on the bottom, my machine works well on Linux and also with the USB-C dock.

If that does not help: Are those some specific kernel errors? Then I can take a closer look on my logs.

andersonjwan, 2021-06-25:
> I'm only a user, so take my advice with caution. Before downgrading I would attempt cutting the power and using the reset/recovery hole on the bottom on the machine. Mine was in some weird state [...]

I'll look into it. Thank you.

Also, if this doesn't work, do you happen to know if downgrading is a potentially fatal move for the machine (i.e. could it effectively break it), or is it considered a safe operation?

>If that does not help: Are those some specific kernel errors? Then I can take a closer look on my logs.

There is nothing too specific that is of interest; however, if any further pop-up, I'll be sure to supply them here.

mrhpearson, 2021-06-25:
@andersonjwan - you can directly send me the kernel errors (markpearson at lenovo) and dock issues. There were some other fixes included in 1.32 and it wasn't supposed to make things worse.

There is a new update coming shortly - it's under test right now but expected early to mid July, to address the black screen issues during update. It won't solve the issues with secure boot enabled as those are due to the shim and are being worked on separately by the community (they're not to do with the firmware itself) but should make the updates less scary.

andersonjwan, 2021-06-26:
@mrhpearson, I sent you one of the same kernel logs that are caught with details.

And thank you. If you don't me mind asking, do you suggest downgrading the BIOS to my previous verison (v0.1.30) or simply keeping it as is if it boots?

miquecg, 2021-06-28:
> There is a new update coming shortly - it's under test right now but expected early to mid July, to address the black screen issues during update. It won't solve the issues with secure boot enabled [...]

Just to be clear since I couldn't update to 1.32. With the upcoming fix would it be necessary to have Secure Boot enabled before the update happens?

mrhpearson, 2021-06-28:
@andersonjwan - replied to your email off thread. Thanks. My inclination would be to not downgrade the BIOS firmware unless you're convinced it's the source of your issues - it does have a number of other fixes in it which are important.

@miquecg  - the secure boot issue is separate and you'll need Secure Boot *disabled* until that is fixed upstream in shim. It's not directly related to the Lenovo firmware or firmware updater.

miquecg, 2021-06-28:
Thanks @mrhpearson. I always had it disabled. Would you mind sharing what is preventing the upgrade in those cases where Secure Boot was not enabled?

RX14, 2021-07-06:
I just experienced something similar to this. T14 Gen 1 AMD, Secure Boot disabled, updating from 0.1.30 to 0.1.32 using fwupd.

I booted the laptop with dock attached with 2 external screens. The laptop booted with the normal splash screen, then screen flashed the words "settings reset" or "config reset" or something similar (can't recall the exact wording), and then cleared the screen. I waited 10 minutes then tried pressing spacebar and enter a few times. Before this the fans were spinning but slowly. Entirely unsure whether it was a coincidence, but after pressing enter/spacebar the fans spun up further. Waited another 20 minutes with no change, still black screen, so pressed the power button. The laptop immediately powered off, without having to hold the button to do a reset. After this, laptop booted normally with BIOS settings including UEFI intact, but on the old firmware version.

I'm reporting this because it fits *some* of the symptoms of other reports, but not all, so might be useful to some.

miquecg, 2021-07-09:
@RX14 I just went updating with the iso file provided on Lenovo downloads site for my device. See this comment on another issue: https://github.com/fwupd/firmware-lenovo-thinkpad/issues/74#issuecomment-751685580

j-silver, 2021-07-28:
T14 running Fedora 34

Today I found an update to 0.1.34. My firmware was still at 0.1.30. Updated without problems on reboot.

mrhpearson, 2021-07-29:
1.34 has been released - recommend updating with secure boot disabled as the shim/secure boot issue may hit you.
Any objections if I close this thread and we track any issues with 1.34 (hopefully none!) separately?

GITMichaelK, 2021-07-29:
Thank you @mrhpearson,
one question: I'm using the system since my OP; the old download may be still stored "somewhere". What is the recommended procedure to go with 1.34? Secure boot is disabled here!

mrhpearson, 2021-07-29:
You should just be able to do the update using fwupd - there are a few success reports up and so far (touch wood) no reports of failures. I tried it on my systems with no problems.
Happy to leave this open for a bit longer. As a heads up I'm on PTO next week and going to be offline (no internet - yay!) so responses will be delayed. I'll ask the team to keep an eye out for anything and hughsie knows how to reach Lenovo folk if anything blows up.

flortsch, 2021-07-31:
Unfortunately, I have the same problem with the 1.34 update as I had with 1.32 before (which I then flashed with the iso provided by Lenovo). After updating with fwupd, the Lenovo screen pops up with "Reset system". The system then reboots and stays in a black screen where nothing happens, and where I have to force reboot the machine.

michaellass, 2021-07-31:
> Unfortunately, I have the same problem with the 1.34 update as I had with 1.32 before (which I then flashed with the iso provided by Lenovo). After updating with fwupd, the Lenovo screen pops up with "Reset system". The system then reboots and stays in a black screen where nothing happens, and where I have to force reboot the machine. Secure boot is turned off.

Same here. Still using the last shim release without the fix but secure boot is disabled. I sent a failure report via fwupd afterwards.

GITMichaelK, 2021-07-31:
@ flortsch, michaellass: Thank you for the warnings!
@ mrhpearson: I'll stay away from it actually, unless you recommend different. What, if my system would not start at all after trying ...? Secure boot is turned off here, too.  
Please let us know about next steps.

mrhpearson, 2021-08-09:
Sorry for the slow reply - I was on vacation. I don't know why you're seeing these issues but will raise them with the FW team. I did a bunch of testing myself and haven't seen a problem.

GITMichaelK, 2021-08-17:
This is to confirm, that today using fwupd, my system (T14s Gen. 1 (AMD)) was successfully updated to BIOS 1.34

michaellass, 2021-08-18:
After the latest failed update I reported (https://github.com/fwupd/firmware-lenovo-thinkpad/issues/131#issuecomment-890325274), I built and installed shim from git (https://github.com/rhboot/shim/commit/1b30c2b9e5ee7d3e305a28a92805152d5cbfc9cb) and retried the 1.34 update via fwupd. And I am very glad to report success!

Brief summary:
* On reboot I was greeted with a friendly update screen instead of a long-taking black screen (sth. like "Please wait while we install an update").
* Afterwards there was a typical firmware flashing screen (erasing, verifying, flashing, verifying) which went though all steps without issues.
* Then there was the "BIOS Self Healing backup progressing" screen which went horribly wrong for me on the 1.32 update (https://github.com/fwupd/firmware-lenovo-thinkpad/issues/129#issuecomment-860204653). This time it went from 0 to 100% without issues and fairly quickly.

So it looks to me like whatever the issue was with the 1.32 update is fixed. What remains is the broken shim release. I hope there will be a release soon that includes the latest fixes. Then firmware updates for the T14s should work reliably again.

I finally have some confidence again in the firmware state of my machine...

Colfenor, 2021-09-14:
as an alternative: shouldn't it be possible to flash the bios from 1.30 to e.g. 1.32 with a windows 10 booted from usb-stick ? has someone tried this out ?

flortsch, 2021-09-14:
Never tried it with Windows, but before fwupd, I flashed BIOS updates on my Thinkpad with geteltorino (which I also did after the update to 1.32 failed with fwupd).

Colfenor, 2021-09-14:
> Never tried it with Windows, but before fwupd, I always flashed BIOS updates on my Thinkpads using geteltorito (which I also did on my T14s after the update to 1.32 failed with fwupd).

Thanks for that insight ! If u have time, could you elaborate on how you upgraded with `geteltorito` (I guess via usb flash drive) ? I'd like to upgrade the bios from 1.30 -> 1.32 however feel a bit reluctant to do so after reading so many hassles/troubles arose

Colfenor, 2021-10-01:
> Yeah, you basically use geteltorito to extract an image from the Lenovo BIOS ISO file and copy it onto a usb flash drive.
> 
> Here are two guides: https://wiki.archlinux.org/title/Flashing_BIOS_from_Linux#Bootable_optical_disk_emulation https://thinkwiki.de/BIOS-Update_ohne_optisches_Laufwerk_unter_Linux#El_Torito_Image_auf_USB-Stick_.28nur_UEFI.29

thanks a lot this worked for me :) however did you experience this issue after upgrading to 1.32 ?
 
https://bbs.archlinux.org/viewtopic.php?id=269705

flortsch, 2021-10-01:
> thanks a lot this worked for me :) however did you experience this issue after upgrading to 1.32 ?
>
> https://bbs.archlinux.org/viewtopic.php?id=269705

Did not notice any TSC clocksource related problems on my machine after upgrading so far, but I will check it out :)

mrhpearson, 2022-12-06:
Catching up on issues...
Please raise a separate ticket (with details) for the T14 (G2?)
I think looking through the other comments that the update worked so I'm going to close this. Please let me know if I'm missing anything.

closed 2022-12-06 by mrhpearson