fwupd/firmware-lenovo #132: Lenovo p1gen2 fails to upgrade Intel Management Engine and System Firmware

fidencio, 2021-06-15:
**Describe the bug**
A clear and concise description of what the bug is.

`$ fwupdmgr upgrade` tells me I have the following upgrades.
 • Intel Management Engine (192.47.1524 → 192.72.1757)
 • System Firmware (0.1.37 → 0.1.38)

I apply those, reboot, and those are not applied.

**Steps to Reproduce**
Steps to reproduce the behavior.

1. `fwupdmgr upgrade` on a p1gen2, using up-to-date Fedora 34
2. reboot to complete the update
3. Run `fwupdmgr upgrade` again to notice that nothing has changed

**Expected behavior**
The update should be completed.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20QUS10L16
│
├─ThinkPad Thunderbolt 3 Dock:
│ │   Device ID:          69ef4e2517425dd61199e356bcd62c9e2ee08993
│ │   Current version:    50.00
│ │   Vendor:             Lenovo (THUNDERBOLT:0x0108|TBT:0x0108)
│ │   GUIDs:              45c1814e-eb83-5b03-8997-ef8d4d8ade29
│ │                       03b97e38-f4b8-5a19-917c-28f95ca29768
│ │                       75039175-c354-5df9-a798-5da41461c007
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │ 
│ ├─USB3.1 Hub:
│ │ │   Device ID:        7e452eb6c025449c23b67d34f76de2b1fdf4c815
│ │ │   Summary:          USB 3.x Hub
│ │ │   Current version:  51.13
│ │ │   Vendor:           VIA Labs, Inc. (USB:0x17EF)
│ │ │   Install Duration: 15 seconds
│ │ │   GUIDs:            8ee94f0e-9b44-596a-bdd9-6f90401664cc
│ │ │                     4ad676ff-8529-5cb0-930d-e3a1da2c60cb
│ │ │                     8aab7e9b-2ae8-559b-9be0-49bf5a647a2a
│ │ │                     a9866465-e1ec-5ce0-855d-162d817aeca2
│ │ │                     06fb2b19-690e-5296-8d0a-6e32a6fccd84
│ │ │   Device Flags:     • Updatable
│ │ │                     • Cryptographic hash verification is available
│ │ │                     • Device stages updates
│ │ │                     • Device can recover flash failures
│ │ │ 
│ │ ├─USB3.1 Hub:
│ │ │ │   Device ID:      e487b5ff307494d6ff44d1b6dc1312bb69b0fe90
│ │ │ │   Summary:        USB 3.x Hub
│ │ │ │   Current version:51.13
[... 381 lines cut ...]
├── initramfs-5.12.5-300.fc34.x86_64.img
├── initramfs-5.12.7-300.fc34.x86_64.img
├── initramfs-5.12.8-300.fc34.x86_64.img
├── loader
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.31
├── System.map-5.12.5-300.fc34.x86_64
├── System.map-5.12.7-300.fc34.x86_64
├── System.map-5.12.8-300.fc34.x86_64
├── vmlinuz-0-rescue-d031c07a49c746449d4884d56f50eba4
├── vmlinuz-5.12.5-300.fc34.x86_64
├── vmlinuz-5.12.7-300.fc34.x86_64
└── vmlinuz-5.12.8-300.fc34.x86_64

6 directories, 77 files
```

**Additional questions**
- Operating system and version: Fedora 34
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Lenovo, yes. 'Boot Order Lock' is turned off.

neupsh, 2021-06-15:
I have same issue with Lenovo P1 Gen 3 as described here: https://github.com/fwupd/firmware-lenovo-thinkpad/issues/80#issuecomment-857063805

mrhpearson, 2021-07-29:
Can you try the update with secure boot disabled please - there's an issue related to shim and fwupd that the shim team is working on. I'm thinking this might be impacting you as you have SB enabled.

neupsh, 2021-07-29:
@mrhpearson , I just checked and secure boot was already disabled. But looks like my firmware updates got applied on my last update. I had updated my system and also applied the `fwupdmgr update` and had rebooted the computer, but did not realize until now that they had been applied. I am not really sure what fixed it as I had lots of updates (archlinux rolling upates).

fidencio, 2021-07-30:
@mrhpearson, in my case it did work, the updates were applied after turning the secure boot off.

mrhpearson, 2021-07-30:
Great - thanks for the update.
This is a known issue that has been fixed upstream but hasn't necessarily rolled out everywhere (e.g for Ubuntu https://bugs.launchpad.net/ubuntu/+source/shim/+bug/1929471)
Closing the bug as I think this is now understood.

closed 2021-07-30 by mrhpearson

fidencio, 2021-07-30:
@mrhpearson, just to confirm, is it okay to enable secure boot again?  As in, once I enable secure boot again things will still work as expect wrt future firmware updates?

mrhpearson, 2021-07-30:
Yep - my understanding is the issue is only with running the fwupdx64.efi update executable. Once the update is done you're back to normal.

neupsh, 2021-07-30:
@mrhpearson should I open a new ticket as it is not resolved for me? I have secure boot disabled.

mrhpearson, 2021-08-09:
@neupsh - yes please, with the details from your system.
Thanks!
Mark