fwupd/firmware-lenovo #133: 1.37 firmware is missing for T480

xvitaly, 2021-06-16:
1.37 firmware update is missing on LVFS for T480.

Update page: https://pcsupport.lenovo.com/it/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480-type-20l5-20l6/downloads/ds502355-bios-update-utility-bootable-cd-for-windows-10-64-bit-linux-thinkpad-t480

mrhpearson, 2021-07-29:
Looks like 1.39 is up in testing now - hopefully makes it to stable soon

xvitaly, 2021-07-30:
> Looks like 1.39 is up in testing now - hopefully makes it to stable soon

1.39? I don't see it on [Lenovo website](Looks like 1.39 is up in testing now - hopefully makes it to stable soon) even for Windows.

MrReSc, 2021-08-07:
Is it possible that something is wrong with the version number?

mrhpearson, 2021-08-09:
I don't think the so - the metainfo state 1.39 in a few places. It likely hasn't been released yet.
I'll see if I can get an update from the FW team on what is happening here.
Mark

Charlyo, 2021-08-17:
Any news on the 1.39? Is undervolt still enabled to be done?

closed 2021-11-12 by xvitaly