fwupd/firmware-lenovo #134: X1 Carbon Gen6 stuck on 0.1.8 (need 0.1.27)

vcavallo, 2021-06-28:
Hello,

This is my first foray into BIOS updates _and_ `fwupd`. As far as I can tell from investigating, I'm on BIOS 0.1.8 and cannot update using either `fwupd` or the published versions on Lenovo's support site because they all fail with:

> Not compatible with firmware version 0.1.8, requires >= 0.1.27

I'm currently running the `snap` version of `fwupd` and have refreshed and re-fetched everything a few times. 

I'm seeing suggestions that using Windows or some USB image are the only solutions. Is that the case? If so, is this path documented anywhere centrally, or just to be found in various GitHub comments?

Hopefully I'm including the right details. This is the output of `get-devices`:

```
20KG0022US
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     GUIDs:              3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│                         7090c94a-ee40-593e-91bc-8cfd5a8c0a5d ← UEFI\RES_{3BABCA5F-B2BF-4F4B-A72E-2BDC84EB4019}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Current version:    184.86.3909
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              9c9d9769-32fa-4841-b550-ea998e754e99
│                         1a4eea87-292a-5d99-82bd-fd244d4e0194 ← UEFI\RES_{9C9D9769-32FA-4841-B550-EA998E754E99}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─SAMSUNG MZVLB512HAJQ-000L7:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express Solid State Drive
│     Current version:    5L2QEXA7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:      S3TNNE0JC55530
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801&REV_00
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                         47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         79d6cfae-a5a2-5936-9248-5aebd23480f7 ← SAMSUNG MZVLB512HAJQ-000L7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Current version:    0.1.8
│ │   Minimum Version:    0.0.1
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   GUIDs:              a4b51dca-8f97-4310-8821-3330f83c9135
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       6b3879ff-921f-5bc5-a0c2-5cec28f31d2f ← UEFI\RES_{A4B51DCA-8F97-4310-8821-3330F83C9135}
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649
│                         5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
└─Thunderbolt host controller:
      Device ID:          755bd261a34d24b3d85d22f6e2408754aa64d3eb
      Summary:            Unmatched performance for high-speed I/O
      Current version:    30.00
      Vendor:             Lenovo (THUNDERBOLT:0x0109, TBT:0x0109)
      GUIDs:              e69c694c-de9f-5f9a-8d30-9b695ab4744b ← THUNDERBOLT\VEN_0109&DEV_1706&REV_00
                          74e58157-171f-57bc-a660-bf467ded54cb ← THUNDERBOLT\VEN_0109&DEV_1706
                          2da42a33-cd30-5ef5-a8fb-2c800a4b760f ← TBT-01091706-native
                          96411853-9f56-53d2-b05c-a54fcad5ee62 ← TBT-01091706-native-controller0-0
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Device stages updates
```

I'm on Ubuntu 18.04.5, `5.4.0-77-generic #86~18.04.1-Ubuntu SMP Fri Jun 18 01:23:22 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

mrhpearson, 2021-06-28:
I'm looking through the internal BIOS release notes....and I'm not seeing why that block is in place - but there's usually a good reason (bricked a few systems by not following recommendations). 
I do have access to the 1.27 iso file, which would I think let you update from a USB stick - but I don't have an X1C6 to try it on.
I'll check guidance with the FW team and get back to you on the recommended next steps
Do you happen to know if you have vPro or non vPro?

vcavallo, 2021-07-01:
Thanks, @mrhpearson !!

non-vPro, it seems.

`cpuinfo` reports `Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz`, which [intel](https://www.intel.com/content/www/us/en/products/sku/122589/intel-core-i78550u-processor-8m-cache-up-to-4-00-ghz/specifications.html) reports is not vPro eligible.

mrhpearson, 2021-07-07:
Hi - can you send me your contact details (markpearson at lenovo dot com) and I'll get you the 1.27 iso and we'll go from there and hopefully get you up to date.
Mark

vcavallo, 2021-07-07:
> Hi - can you send me your contact details (markpearson at lenovo dot com) and I'll get you the 1.27 iso and we'll go from there and hopefully get you up to date.
> Mark

Thanks!! I replied off this thread.

mrhpearson, 2021-07-29:
Hi @vcavallo - did you get this all resolved? Can we close the ticket?