fwupd/firmware-lenovo #136: Lenovo ThinkPad X1 Yoga (Gen 5) Embedded Controller Version is set to 0 and is not updatable by fwupd

linearpone, 2021-07-14:
**Describe the bug**
The problem occurred last year with fwupd version 1.5.1.
My laptop "Lenovo ThinkPad X1 Yoga (Gen 5)" was on ac with a low battery level.
Using ```fwupdmgr update```, I updated the firmware of the devices Embedded Controller, Intel Management Engine, System Firmware and some other devices. Then I restarted my laptop to complete the update. The update was cancelled during the reboot because of low battery status. After that, the Embedded Controller and Intel Management Engine was set to version 0 according to fwupd. 
I could update the Intel Management Engine to the latest version at the time.
```Intel Management Engine (0 → 224.39.1367)```
However, for the Embedded Controller there was no update available to update to the latest version at the time (1.08) according to fwupd. (I have also attempted to install a .cab manually downloaded from fwupd.org. It failed because the version of the Embedded Controller is too low)

Afterwards I tried to update the Embedded Controller via Lenovo's firmware update utility (bootable iso). That works and in the Bios the Embedded Controller is currently reported as version 1.08

I'm also afraid other devices might have been set to version 0 and made un-updateable because if two devices were affected (Embedded Controller and Intel Management Engine) there might as well be more. However, I don't know if any other devices are affected.

The problem is still present with fwupd version 1.6.1

**Steps to Reproduce**
Have low battery level while charging and update the Embedded Controller, Intel Management Engine and System Firmware on the Lenovo ThinkPad X1 Yoga (Gen 5):
Intel Management Engine update to 224.39.1367
System Firmware update to 0.1.14
Embedded Controller update to 1.08

Reboot the device to complete the update and have the update cancelled because of low battery. 

Intel Management Engine and Embedded Controller are set to version 0 in fwupd.

**Expected behavior**
The firmware of the Embedded Controller and Intel Management Engine should have the version they had before the update process and should remain updatable via fwupd afterwards. 

**fwupd version information**
```shell
$ fwupdmgr --version
Current Version:
client version:	1.6.1
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.1
```
Version where the bug happened should be 1.5.1

I installed and updated fwupd with pacman:
https://archlinux.org/packages/community/x86_64/fwupd/

**fwupd device information**

Created with fwupd version 1.6.1
```shell
$ fwupdmgr get-devices --show-all-devices
20UCS03S00
│
├─CometLake-U GT2 [UHD Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              9a3b2da6-943a-5798-9478-9b01e85c49ee
│                         29ee39cc-ad99-55d0-86c1-4f60814707bf
│                         9779d679-0f31-5dd5-b3c2-f6ff275a9ca9
│                         5589d313-88e5-54d3-8f4c-b2ee3882d4bc
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Current version:    0
│     Vendor:             DMI:LENOVO
│     GUIDs:              1ebe8ad3-3d40-4178-92ce-bbb3d0906332
│                         6dd2bd94-436c-5bb1-96a1-c8d0a87e4f09
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
[... 142 lines cut ...]
│                         f60da881-f7a8-5eb2-ab50-f578e2fb9e86
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─UEFI Device Firmware:
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Current version:    22552
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      GUIDs:              9e302497-0dc6-40f0-bf65-0108862ec500
                          00d145c9-0d90-5c63-b6c4-43cdc6169f76
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version: Arch Linux
- Have you tried rebooting? yes
- Is this a regression? Not that I would know.

mrhpearson, 2021-07-15:
Thanks for the note - I've raised a ticket to the FW team (for my reference LO-1303)
Thanks
Mark

ChiWei-Chen, 2021-08-12:
@linearpone , I'm Chi-Wei be working with @mrhpearson .
The HW team already got pass result with trial BIOS and will release BIOS N2WET30W(1.20) fix this issue recently. I believe the BIOS would be upload to LVFS recently(few weeks).
Any concern please let us know. Thank you.

linearpone, 2021-08-15:
Hi @ChiWei-Chen @mrhpearson , 
the Version 1.20 doesn't seem to be released yet. https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2WET.firmware When it is released, will I be able to update via "fwupdmgr update" or will I have to install it manually via for example the Lenovo ISO Installer?
I will note here if it has resolved my problem. Thanks for the fix.

ChiWei-Chen, 2021-08-16:
@linearpone , thanks.
The new BIOS release is under internal release process, I'll keep you update here if I got any concrete information from the FW(BIOS) team.

ChiWei-Chen, 2021-09-24:
@linearpone ,
The BIOS 0.1.20 is released now, please let me know if this does not work for you.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2WET.firmware

linearpone, 2021-09-26:
Hi @ChiWei-Chen ,
the firmware of the Embedded Controller is correctly detected after the system update 0.1.20 as version 0.1.8.

However I couldn't test if I can update my Embedded Controller via fwupd afterwards because there is no newer stable version than 0.1.08 on fwupd [1]. However on the official Lenovo website they offer Embedded Controller Program Version 1.13 [2] which is way newer than the version on fwupd [1]. Do you know why fwupd don't offer the latest "Embedded Controller Firmware"?

[1] https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2WHT.firmware
[2] https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-yoga-5th-gen-type-20ub-20uc/downloads/driver-list/component?name=BIOS

linearpone, 2021-10-06:
In the meantime, there was a new version released for the EC on fwupd and I was able to update. Thank you very much for the work. <3
I'll close the issue.

closed 2021-10-06 by linearpone