fwupd/firmware-lenovo #137: 20U70004UK checksum invalid

mccuskk, 2021-07-22:
Hi

Very excited to see my ThinkPad L15 AMD Gen1 (Fedora 34) want to upgrade to 0.1.10 to 0.1.19

Unfortunately it fails with the error:

Upgrade available for System Firmware from 0.1.10 to 0.1.19
20U70004UK must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading…             [ -                                     ]checksum invalid, expected eab914f9f9357cd5c6e173cf88d275b05ed5d1639ce819cb6f243a6b7a9a31bd got 3850dfdbf4489250268b5f0740240a9f4445e7c5c29e1d03aa0c5446808d7507

Will this eventually work? Id there anything I need to do.

Thanks

Kieran

absurd34, 2021-07-24:
Hi,

+1 from here.

I do see the very same (i.e., exactly same hashes) error for my Thinkpad L14 AMD (20U50001GE), using Debian bullseye.

Hth!

S

hughsie, 2021-07-24:
Can you try "fwupdmgr refresh --force" and then try again? If that fails, can you do "fwupdmgr update --verbose" and tell me the exact https:// link it's trying to download please.

bjoerns1983, 2021-07-27:
Hi there,
i have exactly the same Problem on my Thinkpas L14 AMD (20U50001GE) using Manjaro and fwupdmgr in v 1.6.1
Tried refresh --force but didnt help.
The download link which it is trying to use is the following:
https://fwupd.org/downloads/eab914f9f9357cd5c6e173cf88d275b05ed5d1639ce819cb6f243a6b7a9a31bd-R19ET35W.cab

absurd34, 2021-08-18:
Hi,

Thinkpad L14 AMD (20U50001GE) (0.1.15 → 0.1.19)

fwiw, I retried minutes ago, and this time everything worked fine.

Hth!

S