fwupd/firmware-lenovo #138: Intel ME Update 11.8.86.3909	missing for T470s

dkadioglu, 2021-07-28:
Intel ME Update to version 11.8.86.3909 is missing for ThinkPsd T470s:
https://pcsupport.lenovo.com/us/de/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t470s/downloads/ds121070-intel-management-engine-118-firmware-for-windows-10-64-bit-7-32-bit-64-bit-thinkpad-25-t470-t470s

dkadioglu, 2021-09-17:
The update is available now via LVFS and I successfully applied the update.
However, the device is still shown as updateable, despite the version number is the same now:

```
20HGS00V00
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Zusammenfassung:    UEFI ESRT device
  │   Aktuelle Version:   184.83.3874
  │   Minimale Version:   184.83.3874
  │   Anbieter:           DMI:LENOVO
  │   Aktualisierungsstatus:Success
  │   GUIDs:              e9124c4a-fdff-42e5-b2ad-f745db345953
  │                       fa685a94-138c-531e-8bc2-5d61edfacba8 ← UEFI\RES_{E9124C4A-FDFF-42E5-B2AD-F745DB345953}
  │   Device Flags:       • Internes Gerät
  │                       • Aktualisierbar
  │                       • System benötigt externe Stromquelle
  │                       • Unterstützt auf dem entfernten Server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T470s Corporate ME Update:
        Neue Version:     184.86.3909
        Remote ID:        lvfs
        Zusammenfassung:  Lenovo ThinkPad T470s Corporate ME Firmware
        Lizenz:           Proprietär
        Größe:            7,5 MB
        Erstellt:         2016-07-08
        Dringlichkeit:    Hoch
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N1WRM23W
        Anbieter:         Lenovo Ltd.
        Flags:            is-upgrade
        Beschreibung:     
        This stable release fixes the following issues:
        
        • Mitigated the following security vulnerabilities
```

dkadioglu, 2021-09-17:
Maybe the same as: https://github.com/fwupd/firmware-lenovo-thinkpad/issues/89 and https://github.com/fwupd/firmware-lenovo-thinkpad/issues/70#issuecomment-919238383
I already tried to do `fwupdmgr refresh` or `fwupdmgr refresh --force`, the update is still shown.

dkadioglu, 2022-06-17:
As the various issues concerning wrongly reported ME versions by FWUPD (due to not updated ESRT?) are difficult to follow, I'd like to ask, which of the issues would be the best to follow for a fix for the T470s? Do I understand it right, that the ESRT update issue might already be solved and is rolled out for the different devices, e.g. here https://github.com/fwupd/firmware-lenovo/issues/208#issuecomment-1082214504)? Is there an estimate, when the T470s will get this potential fix?

dkadioglu, 2022-06-24:
I just updated to BIOS 1.46 and afterwards updated Intel ME from 184.86.3909 to 184.92.4222. Although the BIOS shows version 184.92.4222 of Intel ME, fwupdmgr get-upgrades shows still 184.83.3874, which is the case since the update to 184.86.3909 I did last year with fwupdmgr. So BIOS 1.46 still didn't fix the ESRT issue as for example described in #89 and #70 and #156.

Version of fwupd:
```
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.6-zen1-1-zen
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10
```

ChiWei-Chen, 2022-07-21:
@dkadioglu , Thanks for reporting the issue. This seems related to #189 and #201 , I've sent you the link to trial BIOS, please let me know if this doesn't work for you.

dkadioglu, 2022-07-23:
@ChiWei-Chen Thank you for the trial BIOS. I just updated my T470s from 1.46 to 1.47, which went as usual without errors. After the reboot I checked fwupdmgr get-devices which showed a successful BIOS update (Aktualisierungsstatus:Erfolg is german and means successful):

```
System Firmware:
      Gerätekennung:      a45df35ac0e948ee180fe216a5f703f32dda163f
      Zusammenfassung:    UEFI ESRT device
      Aktuelle Version:   0.1.47
      Minimale Version:   0.1.27
      Anbieter:           Lenovo (DMI:LENOVO)
      Aktualisierungsstatus:Erfolg
      GUIDs:              7a176688-0960-47ba-931b-7829849e8347
                          230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
      Device Flags:       • Internes Gerät
                          • Aktualisierbar
                          • System benötigt externe Stromquelle
                          • Unterstützt auf dem entfernten Server
                          • Benötigt einen Neustart nach der Installation
                          • Kryptografische Hash-Verifizierung ist verfügbar
                          • Das Gerät ist während der Dauer der Aktualisierung nutzbar
```

Without even reinstalling the most recent version of ME, fwupdmgr also showed the correct version for it:
```
Intel Management Engine:
     Gerätekennung:      349bb341230b1a86e5effe7dfe4337e1590227bd
     Zusammenfassung:    UEFI ESRT device
     Aktuelle Version:   184.92.4222
     Minimale Version:   0.0.1
     Anbieter:           Lenovo (DMI:LENOVO)
     Aktualisierungsstatus:Erfolg
     GUID:               e9124c4a-fdff-42e5-b2ad-f745db345953
     Device Flags:       • Internes Gerät
                         • Aktualisierbar
                         • System benötigt externe Stromquelle
                         • Unterstützt auf dem entfernten Server
                         • Benötigt einen Neustart nach der Installation
                         • Das Gerät ist während der Dauer der Aktualisierung nutzbar
```
Again, Aktualisierungsstatus:Erfolg is german and means a successful update, which I did 29 days ago, as you can read above. So finally, the trial BIOS seems to have updated the version of ME correctly in ESRT without reinstalling the ME firmware after the update to 1.47. At least for me, the issue seems fixed. Fingers crossed!

ChiWei-Chen, 2022-07-26:
@dkadioglu , thanks! It's great to know it works! I'll keep you posted once the BIOS is released to stable with the fix.

dkadioglu, 2022-08-23:
@ChiWei-Chen I just saw that the updated BIOS has been released: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1WET.firmware
Just a question: Is the trial BIOS 1.47 I installed some weeks ago the same as the released stable version? Should I reinstall?
Thanks!

ChiWei-Chen, 2022-08-25:
@dkadioglu , this is the official version, I would suggest to reinstall this one. Thanks!

dkadioglu, 2022-08-26:
@ChiWei-Chen thanks for your reply, successfully reinstalled.

I think, I can close this now, as the ESRT issue seems to be fixed. If the issue comes up again with the next ME update, I will reopen.

Thank you very much for keeping up and looking into this!

closed 2022-08-26 by dkadioglu