fwupd/firmware-lenovo #139: X1C6 failing to update with `Device Error` message

gkelly, 2021-07-29:
**Describe the bug**
My Lenovo X1 Carbon Gen 6 is failing to apply updates to the ME, System Firmware, and UEFI Device Firmware. The `fwupdmgr update` command identifies and installs the pending updates. On reboot, the machine stays on the Lenovo splash screen for a few minutes, then flashes two lines _very quickly_ in the upper left:

```
WARNING: Could not apply capsule update: Device Error
WARNING: Could not apply capsules: Device Error
```

This is fast enough that I had to record my screen and step frame by frame through the video to read it.

**Steps to Reproduce**
Attempt a `fwupdmgr update`.

**Expected behavior**
Firmware updates.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.8
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
20KH002KUS
│
├─Embedded Controller:
│     Device ID:          9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     GUIDs:              3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│                         7090c94a-ee40-593e-91bc-8cfd5a8c0a5d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel Management Engine:
│     Device ID:          e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:    184.80.3746
│     Minimum Version:    184.80.3746
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUIDs:              42a0a96e-c9f3-438f-9687-7826be33e4ce
│                         375afb87-ea51-5a6e-96d5-c35cc342dc65
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel(R) Core™ i7-8650U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
[... 418 lines cut ...]
├── initrd.img-5.11.0-25-generic
├── initrd.img-5.8.0-63-generic
├── initrd.img.old -> initrd.img-5.8.0-63-generic
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.11.0-25-generic
├── System.map-5.8.0-63-generic
├── vmlinuz -> vmlinuz-5.11.0-25-generic
├── vmlinuz-5.11.0-25-generic
├── vmlinuz-5.8.0-63-generic
└── vmlinuz.old -> vmlinuz-5.8.0-63-generic

5 directories, 297 files
```

**Additional questions**

- Operating system and version: Ubuntu 21.04
- Have you tried rebooting? Yes.
- Is this a regression? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? *No*
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? It is a Lenovo, I disabled "Boot Order Lock' when trying to fix this issue.

superm1, 2021-07-29:
> > Can you please try to enable legacy option rom and see if it helps?
> 
> Out of interest, what's your suspicion?

I saw another bug report where the user had a problem with the me update for lenovo and this helped them. I don't think it's anything more than a workaround though.

superm1, 2021-07-29:
#140

mrhpearson, 2021-07-29:
@gkelly - Are you trying to do all multiple updates in one go? Can we narrow down which update is causing the message and I can flag this to the firmware team and get their input.

gkelly, 2021-07-29:
Okay, very strangely this resolved itself. I did _not_ enable legacy option ROMs. I simply did `fwupdmgr update <X>` then `reboot` for each of the three pending updates. They all installed and my machine now appears to be up to date:

```
20KH002KUS
│
├─Embedded Controller:
│     Device ID:          9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     GUIDs:              3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│                         7090c94a-ee40-593e-91bc-8cfd5a8c0a5d ← UEFI\RES_{3BABCA5F-B2BF-4F4B-A72E-2BDC84EB4019}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel Management Engine:
│     Device ID:          e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:    184.86.3909
│     Minimum Version:    184.86.3909
│     Vendor:             DMI:LENOVO
│     GUIDs:              42a0a96e-c9f3-438f-9687-7826be33e4ce
│                         375afb87-ea51-5a6e-96d5-c35cc342dc65 ← UEFI\RES_{42A0A96E-C9F3-438F-9687-7826BE33E4CE}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Samsung SSD 970 EVO Plus 1TB:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express Solid State Drive
│     Current version:    2B2QEXM7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:      S59ANZFN519328X
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801&REV_00
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                         47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         f05ecac9-5cc2-5f77-b23b-a08ad8879ada ← Samsung SSD 970 EVO Plus 1TB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─System Firmware:
│     Device ID:          1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:    0.1.51
│     Minimum Version:    0.0.1
│     Vendor:             LENOVO (DMI:LENOVO)
│     GUIDs:              a4b51dca-8f97-4310-8821-3330f83c9135
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│                         6b3879ff-921f-5bc5-a0c2-5cec28f31d2f ← UEFI\RES_{A4B51DCA-8F97-4310-8821-3330F83C9135}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│
└─UEFI Device Firmware:
      Device ID:          9e329270a7a68d289c82fe77d32d02208ddf0890
      Current version:    0.73.20
      Minimum Version:    0.73.20
      Vendor:             DMI:LENOVO
      GUIDs:              cea87551-1701-43fb-afbc-6e8ce9728345
                          9ce4f5c8-e787-5f82-a7ef-a22c071c9423 ← UEFI\RES_{CEA87551-1701-43FB-AFBC-6E8CE9728345}
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

mrhpearson, 2021-07-30:
@hughsie - I know we've seen this on other platforms (I've hit it myself once before on the P1G3, though yesterday I did a triple update on the X1C8 with no issues). 
Any idea if this may be fwupdx64.efi related? BIOS? Something else? Not sure where to head with next for this kind of issue :)

hughsie, 2021-07-30:
@mrhpearson it's the first time I've seen this failure I think. It sure would be useful if the BIOS updater could show some kind of message on the screen with a few seconds delay :)

mrhpearson, 2021-07-30:
100% agreed. I asked previously...will try again.
I like the change @superm1 - thanks :)
Mark

LinuxOnTheDesktop, 2021-08-14:
>  I simply did fwupdmgr update X

Note for others: x is the 'Device ID' shown by `fwupdmgr get-updates`.