fwupd/firmware-lenovo #140: Intel Management Engine fails on multiple ThinkPads

MountainX, 2021-07-29:
**Describe the bug**

I am unable to update Intel Management Engine on mutliple ThinkPad X270 devices as well as on the current T470s I'm working on.

**Steps to Reproduce**

On the ThinkPad T470s I run:

    # fwupdmgr update
    Devices with the latest available firmware version:
    • Embedded Controller
    ╔══════════════════════════════════════════════════════════════════════════════╗
    ║ Upgrade Intel Management Engine from 184.77.3664 to 184.83.3874?             ║
    ╠══════════════════════════════════════════════════════════════════════════════╣
    ║ Version 11.8.83.3874 (LVFS: 184.83.3874)                                     ║
    ║ Problem Fixes                                                                ║
    ║ • Mitigated the following security vulnerabilities                           ║
    ║ 20HGS6R400 must remain plugged into a power source for the duration of the   ║
    ║ update to avoid damage.                                                      ║
    ╚══════════════════════════════════════════════════════════════════════════════╝

    Perform operation? [Y|n]: y
    Downloading…             [***************************************]
    Decompressing…           [***************************************]
    Authenticating…          [***************************************]
    Authenticating…          [***************************************]
    Updating Intel Management Engine…********************************]
    Scheduling…              [***************************************]
    Successfully installed firmware
    • SAMSUNG MZVLB512HAJQ-000L7
    ╔══════════════════════════════════════════════════════════════════════════════╗
    ║ Upgrade System Firmware from 0.1.38 to 0.1.41?                               ║
    ╠══════════════════════════════════════════════════════════════════════════════╣
    ║ Some new functionality has also been added:                                  ║
    ║ *Updated YD1 skylake MCU to MC0406E3_000000ED_000000EA and KBL MCU to        ║
    ║ MC0806E9_000000E9_000000EA                                                   ║
    ║ *Version 04.17.000 code of FIT's InROM diagnostics.                          ║
    ║ 20HGS6R400 must remain plugged into a power source for the duration of the   ║
    ║ update to avoid damage.                                                      ║
    ╚══════════════════════════════════════════════════════════════════════════════╝

    Perform operation? [Y|n]: y
    Downloading…             [***************************************]
    Decompressing…           [***************************************]
    Authenticating…          [***************************************]
    Authenticating…          [***************************************]
    Updating System Firmware…[***************************************]
    Scheduling…              [***************************************]
    Successfully installed firmware

    An update requires a reboot to complete. Restart now? [y|N]: y

Rebooting results in the system failing to boot. After the Lenovo splash screen, the text "reset system" briefly appears in the upper left of the screen, the screen goes black and in a few minutes the system fan comes on as the system heats up.

No progress is made after several hours. (I let an X270 with the same issue run for 3 days without any success.) The power button will not turn the device off. The only option is to use the system reset hole on the T470s. Fortunately, the system is not bricked. But the firmware is not installed. I have repeated these steps twice. I don't wish to do it again because my system may not recover next time (?).

**Expected behavior**

The system should reboot and the firmware should be installed.

**fwupd version information**

I'm running a fully updated Arch Linux install. I ran `pacman -Syu` immediately before running `fwupdmgr update`

    # fwupdmgr --version
    client version: 1.6.1
    compile-time dependency versions
            gusb:   0.3.7
    daemon version: 1.6.1

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

pacman

**fwupd device information**

I can share that info privately.

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? Not sure
- Are you using an NVMe disk? Yes, but I don't boot from it.
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Boot order lock is disabled.

mrhpearson, 2021-07-29:
@superm1 - for the note above, is this not the same secure boot/shim update issue we've been seeing as the user had to disable secure boot? Wondered what I'm missing

@smrqdt  - Can I just confirm that you did the update, got a black screen (for a very long time) and had to hit the hidden reset button to recover. Had the update completed succesfully afterwards or not? 
The reset button should not be needed - we've had a few other reports of that on other platforms and it's bad. I'll flag this to the firmware team because it's horrible and not intended.

superm1, 2021-07-29:
@mrhpearson no it's not the same.  The reports above mention that secure boot was disabled, and explicitly turned on legacy boot.

mrhpearson, 2021-07-29:
@superm1 - yeah, but the user also said "I had to disable secure boot too" - so I was taking that to mean the additional questions report was incorrect and SB was actually on initially

superm1, 2021-07-29:
I guess that's true, the first post could have had a mistake.
@MountainX can you please clarify the version of shim on your system?  Or were you using self signed binaries for secure boot?

klemmster, 2021-07-29:
I'm having the same issue as well on a T480:
```Upgrade Intel Management Engine from 184.83.3874 to 184.86.3909?```
with `UEFI only` the process would get stuck at a black screen. (I didn't know about the secret hole/button, so I waited for the battery to drain completely and luckily enough the computer started again.

With `both` option enabled, the Red/White Lenovo sign is shown, after a while `Reset system` is printed in the upper left corner. The update does not get applied.

smrqdt, 2021-07-29:
> @smrqdt - Can I just confirm that you did the update, got a black screen (for a very long time) and had to hit the hidden reset button to recover. Had the update completed succesfully afterwards or [...]
No, the update is not applied. The additional Embedded Controller Update that got released since also wasn’t applied. I can try to apply the Embedded Controller Update on its own to check whether the process works, if it’s safe to do that without the ME update.

I noticed that after running `fwupd update` there are 3 capsules in `/boot/efi/EFI/Arch/fw` (`fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap`, `fwupd-9e21f98b-fe98-455c-b388-da5450ab6979.cap`, `fwupd-b87a926d-189c-49a1-b1ff-921099b1de89.cap`), after the failed update `fwupd-9e21f98b-fe98-455c-b388-da5450ab6979.cap` is gone but the other ones are still there.

Is there a way to get a log from the updater? If it helps I can put a camera in front of the display, but there isn’t much to see.
I’m used to get a big red Lenovo logo and some text in the upper left corner that enumerates the capsules beeing loaded. Nothing like that happens. Last update that got applied was ME update 184.83.3874. The model is 20L5000BGE.

tgelter, 2021-08-16:
I found this issue after running into the same problem when notified of [this update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2ERM.firmware) starting this morning. I'll be happy to help test or provide additional information. Thanks!

smrqdt, 2021-08-23:
I tried again today and was able to get a different result:

I was able to install the newly released [System Firmware N24ET65W 0.1.39](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24ETXXW.firmware) and [Embedded Controller Firmware 0.1.22](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24HT.firmware), the ME Update wasn’t applied, but it did not get stuck anymore (a message said something like `system reset`). I tried to apply the ME Update on its own afterwards, but it seems it is still broken. But at least it does not get stuck anymore.

(side note: I just noticed the LVFS web page lists a 0.1.40 firmware update as available, that was released *before* 0.1.39 according to the timestamp. fwupdmgr does not recognize this update and thinks 0.1.39 was the latest update. I tried `fwupdmgr refresh --force` and the update is 5 days old, so probably not a caching thing. Maybe this is a fwupdmgr bug @hughsie?)

btw: it would be easier to find devices on LVFS if the devices would not all have the same name _UEFI Device Firmware_ in `fwupdmgr get-devices` and not show the summary _UEFI ESRT device_.

<details>
<summary>fwupdmgr get-history</summary>
```
├─Embedded Controller:
│ │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│ │   Previous version:   0.1.21
│ │   Update State:       Success
│ │   Last modified:      2021-08-23 18:03
│ │   GUID:               b87a926d-189c-49a1-b1ff-921099b1de89
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad T480 Embedded Controller Update:
│       New version:      0.1.22
│       Remote ID:        lvfs
│       Summary:          Lenovo ThinkPad T480 Embedded Controller Firmware
│       License:          Proprietary
│       Size:             791,7 kB
│       Created:          2016-07-08
│       Urgency:          High
│       Vendor:           Lenovo Ltd.
│       Description:
│       Lenovo ThinkPad T480 Embedded Controller Firmware.
│
│       The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│
│       Update item in this release are applicable to Windows OS only.
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Previous version:   0.1.36
│ │   Update State:       Success
│ │   Last modified:      2021-08-23 18:03
│ │   GUID:               dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad T480 System Update:
│       New version:      0.1.39
│       Remote ID:        lvfs
│       Summary:          Lenovo ThinkPad T480 System Firmware
│       License:          Proprietary
│       Size:             9,5 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Vendor:           Lenovo Ltd.
│       Description:
│       Lenovo ThinkPad T480 System Firmware
│
│       The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│
│       Update item in this release are applicable to Windows OS only
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   184.83.3874
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2021-08-23 18:22
  │   GUID:               9e21f98b-fe98-455c-b388-da5450ab6979
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad T480 Consumer ME Update:
        New version:      184.86.3909
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T480 Consumer ME Firmware
        License:          Proprietary
        Size:             2,2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N24RN17W
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:
        Lenovo ThinkPad T480 ME Firmware Version 11.8.86.3909(LVFS: 184.86.3909)

        The computer will be restarted automatically after updating completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.

        This stable release fixes the following issues:

        • Intel CSME IPU 2021.1:

        Addressed several critical security vulnerabilities.
```
</details>

MountainX, 2021-09-22:
> I guess that's true, the first post could have had a mistake.
> @MountainX can you please clarify the version of shim on your system? Or were you using self signed binaries for secure boot?

I am not using self-signed binaries. (I'm not sure what "shim" is.)

michaeldleslie, 2021-11-06:
I am seeing the same behavior that @smrqdt described during an update today on my T480.

Successfully installed system firmware 0.1.40 and embedded controller firmware 0.1.22. Only the ME update 184.86.3909 fails to apply. However, it also doesn't get stuck: it sits on the red-and-white Lenovo splash screen for a bit, eventually "System reset" is printed, and then the machine reboots, as others have also described. One difference from what @klemmster saw is that my machine did **not** get stuck even though it is in `UEFI only` mode (and not `both` mode).

Secure boot & boot order lock are both disabled. Up-to-date Arch Linux. Tried rebooting/full shutdown between attempts. I am booting from NVMe drive.

<details>
  <summary>fwupdmgr --version</summary>

```
# fwupdmgr --version
client version:	1.7.1
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.1
```

</details>

<details>
  <summary>fwupdmgr get-history</summary>

```
# fwupdmgr get-history
20L5CTO1WW
│
├─Thunderbolt host controller:
│ │   Device ID:          7c79e48263ee4dad661d273138430989122fa7f8
│ │   Previous version:   14.00
│ │   GUID:               8487e4cb-fe6b-5cb6-ad88-46535fb110e1
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │
│ └─  New version:      20.00
│       License:          Unknown
│
├─Embedded Controller:
│ │   Device ID:          c19580db344e418cf2b23297bcf46d75c06e432a
│ │   Previous version:   0.1.20
│ │   Update State:       Success
│ │   Last modified:      2021-01-20 05:17
│ │   GUID:               b87a926d-189c-49a1-b1ff-921099b1de89
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.21
│       Remote ID:        lvfs
│       License:          Unknown
│
├─Intel Management Engine:
│ │   Device ID:          ea0fd60df22477668e549841f6a22a8c31cae587
│ │   Previous version:   184.70.3626
│ │   Update State:       Success
│ │   Last modified:      2021-01-20 05:17
│ │   GUID:               9e21f98b-fe98-455c-b388-da5450ab6979
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
[... 93 lines cut ...]
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad T480:
        New version:      184.86.3909
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T480 Consumer ME Firmware
        License:          Proprietary
        Size:             2.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N24RN17W
        Vendor:           Lenovo
        Flags:            is-upgrade
        Description:
        Lenovo ThinkPad T480 ME Firmware Version 11.8.86.3909(LVFS: 184.86.3909)

        The computer will be restarted automatically after updating completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.

        This stable release fixes the following issues:

        • Intel CSME IPU 2021.1:

        Addressed several critical security vulnerabilities.
```
</details>

klemmster, 2021-11-07:
@michaeldleslie Thank you for the ping. 
I've updated the system firmware from 0.1.39 to 0.1.40 and the machine doesn't get stuck anymore. The behavior is now identical to the one Michael described.

MountainX, 2021-11-08:
I am continuing to see issues with *70 series ThinkPads (like X270 or T470).

javadr, 2021-11-09:
Same issue with X390 Yoga. 
```
$ sudo fwupdmgr get-updates                                                            
Devices with no available firmware updates: 
• Prometheus IOTA Config
• UEFI Device Firmware
• UEFI Device Firmware
• UEFI Device Firmware
• UEFI dbx
Devices with the latest available firmware version:
• Prometheus
• SAMSUNG MZVLB512HBJQ-000L7
• System Firmware
Idle…                    [***************************************]
Update failure is a known issue, visit this URL for more information: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
Successfully uploaded 1 report
20NQ000RMX
│
└─Intel Management Engine:
│   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│   Summary:            UEFI ESRT device
│   Current version:    192.72.1757
│   Minimum Version:    192.72.1757
│   Vendor:             DMI:LENOVO
│   Update State:       Success
│   GUIDs:              aedb5488-946b-4938-86e7-a15109e983b2
│                       b1f2154c-bd08-5a38-817c-8691ee2111ba ← UEFI\RES_{AEDB5488-946B-4938-86E7-A15109E983B2}
│   Device Flags:       • Internal device
│                       • Updatable
│                       • System requires external power source
│                       • Supported on remote server
│                       • Needs a reboot after installation
│                       • Device is usable for the duration of the update
│
└─ThinkPad X390 Yoga:
New version:      192.81.1753
Remote ID:        lvfs
Summary:          Lenovo ThinkPad X390 Yoga Corporate ME Firmware
License:          Proprietary
Size:             12.4 MB
Created:          2016-07-08
Urgency:          High
Details:          https://pcsupport.lenovo.com/de/en/search?query=N2LRM20W
Vendor:           Lenovo
Flags:            is-upgrade
Description:
This stable release fixes the following issues:

• Several Intel known issues were fixed in this release. Refer to Intel
• Closed Issues at the bottom portion of this build letter
```

Operating system and version: Manjaro
Have you tried rebooting? Yes
Is this a regression? No
Are you using an NVMe disk? Yes
Is secure boot enabled? No
Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Boot order lock is disabled.

```
$ fwupdmgr --version                                                                            
client version: 1.6.4
compile-time dependency versions
gusb:   0.3.7

daemon version: 1.6.4
```

```
System:    Kernel: 5.14.10-1-MANJARO x86_64 bits: 64 compiler: gcc v: 11.1.0 Desktop: GNOME 40.5
           Distro: Manjaro Linux base: Arch Linux
Machine:   Type: Convertible System: LENOVO product: 20NQ000RMX v: ThinkPad X390 Yoga 
           Mobo: LENOVO UEFI: LENOVO v: N2LET88W (1.88) date: 10/20/2021
```

xvitaly, 2021-11-12:
Can't install 184.86.3909 on ThinkPad T480. Tested with Secure Boot disabled:
```
client version: 1.5.12
compile-time dependency versions
        gusb:   0.3.7

daemon version: 1.5.12
```

xvitaly, 2021-11-12:
@mrhpearson Can you ask Lenovo firmware team to check this issue?

LinuxOnTheDesktop, 2021-11-28:
@mrhpearson Perhaps you could tell the team also that having problems for months and months with multiple models of ThinkPads (as evidenced by this very bugtracker) has got rather beyond a joke.

xvitaly, 2021-11-28:
After several attempts to install the update, I finally got an error message:

[image]

mrhpearson, 2021-11-30:
My first reaction was to be angry that we had broken FW uploaded - but checking the telemetry this FW has been downloaded 145732 with 77 successes and 2 fails reported (most users don't report). So I don't think it's the FW itself that is neccesarily bad (see attached screenshot)

[image]

I've raised  LO-1462 for internal tracking and see if the FW team can figure out what is going on.
I'll let you know when I hear back from the FW team. If you can think of anything that you may have tweaked in the BIOS let me know - the previous T480 case we looked into we were unable to reproduce internally, which makes it really hard to debug.

Mark

LinuxOnTheDesktop, 2021-11-30:
Mark,

Thank you.

However, you risk being disingenuous. For, what would the telemetry look like for a _range_ of ThinkPad models and a _range_ of updates? This bug tracker suggests that there is _a lot_ of failure happening. (Also: I, for one, tell fwupd to report its failures and its successes.)

hughsie, 2021-11-30:
> However, you risk being disingenuous

That's uncalled for, please don't do this again. Mark is the most open of all the vendor representatives.

> This bug tracker suggests that there is a lot of failure happening

The problem is scale; the LVFS is pumping out 2 million updates a month now, and with just 0.01% failure rate that's an awful lot of failed firmware updates being reported.

smrqdt, 2021-11-30:
> My first reaction was to be angry that we had broken FW uploaded - but checking the telemetry this FW has been downloaded 145732 with 77 successes and 2 fails reported (most users don't report). So I don't think it's the FW itself that is neccesarily bad (see attached screenshot)

The update error fwupdmgr shows is `failed to run update on reboot`.

```
$ fwupdmgr get-results 349bb341230b1a86e5effe7dfe4337e1590227bd
Intel Management Engine:
  Device ID:            349bb341230b1a86e5effe7dfe4337e1590227bd
  Previous version:     184.83.3874
  Update State:         Failed
  Update Error:         failed to run update on reboot
  Last modified:        2021-11-19 04:57
  GUID:                 9e21f98b-fe98-455c-b388-da5450ab6979
  Device Flags:         …
```

Wouldn’t this be the same error as if the fwupd EFI entry did not run during reboot? Is this even reported or maybe filtered, because this is a common problem unrelated to firmware?

btw: I have a non-vPro device and therefor problems installing the Consumer ME firmware `184.83.3874` , maybe this is relevant.

javadr, 2021-12-01:
```
$ fwupdmgr get-results 349bb341230b1a86e5effe7dfe4337e1590227bd
Intel Management Engine:
Device ID:            349bb341230b1a86e5effe7dfe4337e1590227bd
Previous version:     192.72.1757
Update State:         Failed
Update Error:         failed to run update on reboot
Last modified:        2021-12-01 04:52
GUID:                 aedb5488-946b-4938-86e7-a15109e983b2
Device Flags:         • Internal device
                      • Updatable
                      • System requires external power source
                      • Needs a reboot after installation
                      • Reported to remote server
                      • Device is usable for the duration of the update
```

kyak, 2021-12-11:
Since I wasn't able to update BIOS and EC firmware using `fwupdmgr` (see https://github.com/fwupd/firmware-lenovo-thinkpad/issues/144#issuecomment-986187162), I did it using the ISO provided by Lenovo. It worked fine.

Now Intel ME is the only firmware that needs updating. However, it fails to run update on reboot:

```
# LANG=en_US.UTF-8 fwupdmgr get-results 349bb341230b1a86e5effe7dfe4337e1590227bd
Intel Management Engine:
  Device ID:            349bb341230b1a86e5effe7dfe4337e1590227bd
  Previous version:     224.37.1165
  Update State:         Failed
  Update Error:         failed to run update on reboot
  Last modified:        2021-12-11 00:10
  GUID:                 99dbe717-e2cb-4a4b-adcd-47edb34b18f1
  Device Flags:         • Internal device
                        • Updatable
                        • System requires external power source
                        • Needs a reboot after installation
                        • Reported to remote server
                        • Device is usable for the duration of the update
```

PVasileff, 2022-01-06:
Same here. Using Lenovo T480, Fedora 35:

```
$ fwupdmgr --version
client version: 1.7.3
compile-time dependency versions
        gusb:   0.3.9

daemon version: 1.7.3

```
```
$ fwupdmgr get-results 349bb341230b1a86e5effe7dfe4337e1590227bd
Intel Management Engine:
  Device ID:            349bb341230b1a86e5effe7dfe4337e1590227bd
  Previous version:     184.82.3838
  Update State:         Failed
  Update Error:         failed to run update on reboot
  Last modified:        2022-01-05 18:00
  GUID:                 9e21f98b-fe98-455c-b388-da5450ab6979
  Device Flags:         • Internal device
                        • Updatable
                        • System requires external power source
                        • Needs a reboot after installation
                        • Reported to remote server
                        • Device is usable for the duration of the update

```

mrhpearson, 2022-01-07:
I've updated the ticket with the extra information and escalated it as I've not had any useful feedback yet. Sorry :(

Mark

pgn674, 2022-02-08:
Here's a possibility that hasn't been mentioned: Maybe the Intel ME firmware was actually updated successfully, but fwupd is reporting the version incorrectly?

Try out this tool: [Intel® Converged Security and Management Engine Version Detection Tool (Intel® CSMEVDT)](https://www.intel.com/content/www/us/en/download/19392/intel-converged-security-and-management-engine-version-detection-tool-intel-csmevdt.html)

```
fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.65.3590 to 184.86.3909?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad T480 ME Firmware Version 11.8.86.3909(LVFS: 184.86.3909)     ║
║                                                                              ║
║ The computer will be restarted automatically after updating completely. Do   ║
║ NOT turn off your computer or remove the AC adaptor while update is in       ║
║ progress.                                                                    ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Intel CSME IPU 2021.1:                                                     ║
║                                                                              ║
║ Addressed several critical security vulnerabilities.                         ║
║                                                                              ║
║ 20L50011US must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

```
./intel_csme_version_detection_tool 
Intel(R) CSME Version Detection Tool
Application Version: 6.0.1.0
*** Host Computer Information ***
Manufacturer: LENOVO
Model: 20L50011US
Processor Name: Intel(R) Core(TM) i5-8350U CPU @ 1.70GHz
OS Version: Fedora Linux 35 (Workstation Edition) (5.16.5-200.fc35.x86_64)
*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 11.8.86.3909
*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.
```

tgelter, 2022-02-08:
> Here's a possibility that hasn't been mentioned: Maybe the Intel ME firmware was actually updated successfully, but fwupd is reporting the version incorrectly?

Great question! I show similar results with a different OS/Model:
```
❯ fwupdmgr update
Devices with no available firmware updates: 
 • USB2.0 Hub
 • USB2.0 Hub
 • USB3.1 Hub
 • USB3.1 Hub
 • MSP430
 • ThinkPad Thunderbolt 3 Dock
 • ThinkPad Thunderbolt 3 Dock USB Audio
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 192.72.1757 to 192.81.1753?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 0 Intel Platform Update 2021.1 (PV Maintenance Release)                    ║
║                                                                              ║
║ Version 12.0.81.1753 (LVFS: 192.81.1753)                                     ║
║                                                                              ║
║ Problem fixes                                                                ║
║                                                                              ║
║ • Several Intel known issues were fixed in this release. Refer to Intel      ║
║ • Closed Issues at the bottom portion of Intel's release note                ║
║                                                                              ║
║ 20MF000MUS must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: n
Request canceled

~ took 4s 
❯ ~/Downloads/lenovo/intel_csme_version_detection_tool 
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2021, Intel Corporation, All rights reserved.

Application Version: 6.0.1.0
Scan date: 2022-02-08 22:28:34 GMT

*** Host Computer Information ***
Name: <redacted>
Manufacturer: LENOVO
Model: 20MF000MUS
Processor Name: Intel(R) Core(TM) i7-8850H CPU @ 2.60GHz
OS Version: Arch Linux (5.16.5-arch1-1)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.81.1753

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide
or the related Intel Security Advisory list at:
https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html
```

smrqdt, 2022-02-08:
Same result for me, `intel_csme_version_detection_tool` tells me 11.8.86.3909 is installed, `fwupdmg` wants to install `11.8.86.3909(LVFS: 184.86.3909)`.

<details>
<summary>fwupdmgr get-updates 349bb341230b1a86e5effe7dfe4337e1590227bd</summary>

```
% fwupdmgr get-updates 349bb341230b1a86e5effe7dfe4337e1590227bd
20L5000BGE
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    184.83.3874
  │   Minimum Version:    184.83.3874
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              9e21f98b-fe98-455c-b388-da5450ab6979
  │                       b4c101ff-035f-54a0-9147-23a9954dc8e0
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480:
        New version:      184.86.3909
        Remote ID:        lvfs
        Release ID:       8333
        Summary:          Lenovo ThinkPad T480 Consumer ME Firmware
        License:          Proprietary
        Size:             2,2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N24RN17W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad T480 ME Firmware Version 11.8.86.3909(LVFS: 184.86.3909)
        
        The computer will be restarted automatically after updating completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        This stable release fixes the following issues:
        
        • Intel CSME IPU 2021.1:
        
        Addressed several critical security vulnerabilities.
        
        This firmware is provided by LVFS community members and is not provided (or supported) by the original hardware vendor.
        
        Installing this update may also void any device warranty.
```
</details>
<details>
<summary> ./intel_csme_version_detection_tool</summary>

```
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2021, Intel Corporation, All rights reserved.

Application Version: 6.0.1.0
Scan date: 2022-02-08 23:17:01 GMT

*** Host Computer Information ***
Name: kalle
Manufacturer: LENOVO
Model: 20L5000BGE
Processor Name: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
OS Version: Arch Linux (5.16.5-arch1-1)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 11.8.86.3909

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide
or the related Intel Security Advisory list at:
https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html
```
</details>

@pgn674 Great catch, I hope this realization simplifies a fix.

laudrup, 2022-02-08:
@pgn674 

Exactly the same here. The Intel tool shows I'm already updated.

Guess I don't need to post the output from the tool and fwupdmgr, but I'm running on a Lenovo Carbon X1 5th gen.

This is really great. Has been annoying me for a quite a while.

kmauleon, 2022-02-09:
This is a known issue due to BIOS ESRT Table not updating correctly, thus, fwupd not getting the correct version.
Sorry, there's not much update at the moment.  We are tying to ask anyone willing to use the trial BIOS as mentioned here #67 #70 ... Thank you very much...
Any luck @mrhpearson?

hughsie, 2022-02-09:
@kmauleon I'm a bit confused why the ME update can't update the reported ESRT version itself. Is the system bios responsible for updating the ESRT version of the ME component based on some kind of hash->version mapping table? If so, that seems, well, suboptimal...

kmauleon, 2022-02-10:
hi @hughsie as mentioned before ESRT table is managed by BIOS... Intel does not know anything about it, thus we can't answer how BIOS is updating it... thank you very much.

hughsie, 2022-02-10:
> thus we can't answer how BIOS is updating it

How does the BIOS update the ME entry in the ESRT table now? There must be some kind of side-channel (e.g. AMT) to get the ME version number.

kmauleon, 2022-02-11:
@hughsie sorry i don't have the answer for that... I've been trying to ask the BIOS dev that same question but the only reply I got was we have to replicate the issue in order for them to investigate.  this is the reason why I already asked help from @mrhpearson to push BIOS Dev.  currently Mark obtain a few systems' trial BIOS and asking if some customers can try.... thank you very much

superm1, 2022-02-11:
How is this bug not wrecking havoc on Windows Update too?  It must be right?

At least for Linux though, maybe it's possible to do some custom overrides for now?

Here's an idea if someone wants to codify it:
1. Ignore the ESRT entry corresponding to the ME for Lenovo Intel devices in the `uefi` plugin. 
2. Get the version from the ME directly using either the `amt` plugin or the `pci-mei` plugin.  
3. Create a new device entry corresponding to that plugin but use the `uefi` plugin to install it.  This is similar to how the `dell`  plugin issues TPM updates.

That assumes that the ESRT GUID used for a family of ME devices is known and a fixed value.

cocotazo245, 2022-02-28:
T580 running Fedora 35. 

fwupd detecting incorrect version of IME firmware.

[image]
[image]

sudopluto, 2022-03-12:
same here, t480 running fedora 35 and latest fwupd on that relase, i get the same results as the screenshot above.

LinuxOnTheDesktop, 2022-03-12:
@[mrhpearson](https://github.com/mrhpearson), @hughsie: might I ask whether any progress is being made upon this problem - and indeed upon similar problems - please?

hughsie, 2022-03-12:
> might I ask whether any progress is being made upon this problem

Yes, we're discussing the root cause via email due to an NDA -- sorry it's not public. We're making progress.

sudopluto, 2022-03-15:
hopefully not revealing too much here, but can anyone else not downgrade their ME firmware? I get an error: 8578: image is not supported by the system upon reboot, and the downgrade install does not proceed. this is on a t480, all firmware updated with fwupd on fedora

mrhpearson, 2022-03-15:
I'll check with the FW team - but double check that you have FW downgrades allowed in your BIOS settings

CodeKlopfer, 2022-03-30:
Same here at T480 running Ubuntu 20.04:

```
$ fwupdmgr upgrade 
Devices with no available firmware updates: 
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.86.3909 to 184.90.3987?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 8 Intel Platform Update 2021.2                                             ║
║                                                                              ║
║ Version 11.8.90.3987 Fixes the  IPU 2021.2 security vulnerabilities. please  ║
║ refer to Issues List.                                                        ║
║                                                                              ║
║ 20L6S7PC2Z must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: n
Request canceled
$ sudo ./intel_csme_version_detection_tool 
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2021, Intel Corporation, All rights reserved.

Application Version: 6.0.1.0
Scan date: 2022-03-30 09:34:01 GMT

*** Host Computer Information ***
Name: ThinkPad-T480
Manufacturer: LENOVO
Model: 20L6S7PC2Z
Processor Name: Intel(R) Core(TM) i5-8350U CPU @ 1.70GHz
OS Version: Ubuntu 20.04.4 LTS (5.4.0-105-generic)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 11.8.90.3987

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide
or the related Intel Security Advisory list at:
https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html```

mrhpearson, 2022-04-01:
I missed updating this thread - apparently downgrading of ME FW is not possible as it's a security concern. We're waiting on a potential fix for this problem from the FW team. It's supposed to be released before the next ME update is available....

@CodeKlopfer - I think your system has likely been updated but the BIOS is giving the wrong version to fwupdmgr, so it keeps presenting the old version as a needed update. If you go into the F1 setup screen at early boot you can check the version there to confirm

CodeKlopfer, 2022-04-04:
> @CodeKlopfer - I think your system has likely been updated but the BIOS is giving the wrong version to fwupdmgr, so it keeps presenting the old version as a needed update. If you go into the F1 setup screen at early boot you can check the version there to confirm

@mrhpearson you were right, BIOS confirms Version: 11.8.90.3987

mrhpearson, 2022-04-04:
Thanks for the update. So afraid we need to wait for the BIOS fix to be released from the FW team. For now you can safely ignore the proposed ME FW update.

ChiWei-Chen, 2022-04-08:
Hi,
I work with Mark and have a trail BIOS for T480. If someone is willing to test that it would be appreciated - send me or @mrhpearson an email. Thanks!
(cchen50 at lenovo dot com) 

Installation will be by fwupdmgr and it will involve updating the BIOS and then trying a ME FW update.

Thanks

git1106, 2022-04-08:
Updating is impossible with the newest ME update for `"HostSku" : "LENOVO_MT_20QD_BU_Think_FM_ThinkPad X1 Carbon 7th" ` that came a couple of days ago. This is the first problem on this machine after the Ubuntu shim problem more than a year ago. The reported error is `"UpdateError" : "failed to run update on reboot"`. (I have uploaded at least 2 reports.)

```
fwupdmgr update
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 192.81.1753 to 192.85.1869?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 0 Intel Platform Update 2021.2 Product Version Maintenance Release         ║
║                                                                              ║
║ Version 12.0.85.1869 (LVFS: 192.85.1869)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities under issues. Please      ║
║ see fixed issues for details.                                                ║
║                                                                              ║
║ Note: Please suspend or disable any disk encryption tool prior to update.    ║
║                                                                              ║
║ 20QD003JMX must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
 • System Firmware
 • Thunderbolt host controller
 • UEFI Device Firmware
 • WDC PC SN730 SDBQNTY-512G-1001

An update requires a reboot to complete. Restart now? [y|N]: 
```

After retrying with `sudo mokutil --set-verbosity true` I see that the last UEFI console messages are:

* `Verification succeeded` for `fwupdx64.efi`
* `Reset system`
* `Press any key to continue` on new screen with Thinkpad logo

Intel's tool reports the old version (although they claim it's not vulnerable, which is a bit weird if 1869 is supposed to mitigate security issues):

```
$ ./intel_csme_version_detection_tool 
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2021, Intel Corporation, All rights reserved.

Application Version: 6.0.1.0
Scan date: 2022-04-07 23:57:04 GMT

*** Host Computer Information ***
Name: [redacted]
Manufacturer: LENOVO
Model: 20QD003JMX
Processor Name: Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz
OS Version: Ubuntu 20.04.4 LTS (5.13.0-39-generic)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.81.1753

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide
or the related Intel Security Advisory list at:
https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html

$
```

Setup screen also displays the old version 1753.

Package versions:
```
$ apt-cache policy shim-signed
shim-signed:
  Installed: 1.40.7+15.4-0ubuntu9
  Candidate: 1.40.7+15.4-0ubuntu9
  Version table:
 *** 1.40.7+15.4-0ubuntu9 500
        500 http://fi.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     1.40.3+15+1533136590.3beb971-0ubuntu1 500
        500 http://fi.archive.ubuntu.com/ubuntu focal/main amd64 Packages
$ apt-cache policy fwupd-signed
fwupd-signed:
  Installed: 1.27.1ubuntu7+1.2-2~20.04.1
  Candidate: 1.27.1ubuntu7+1.2-2~20.04.1
  Version table:
 *** 1.27.1ubuntu7+1.2-2~20.04.1 500
        500 http://fi.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     1.27.1ubuntu1+1.3.9-4ubuntu0.1 500
        500 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages
     1.27+1.3.9-4 500
        500 http://fi.archive.ubuntu.com/ubuntu focal/main amd64 Packages
$ apt-cache policy fwupd
fwupd:
  Installed: 1.7.5-3~20.04.1
  Candidate: 1.7.5-3~20.04.1
  Version table:
 *** 1.7.5-3~20.04.1 500
        500 http://fi.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     1.3.9-4ubuntu0.1 500
        500 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages
     1.3.9-4 500
        500 http://fi.archive.ubuntu.com/ubuntu focal/main amd64 Packages
$ 
```

Sorry if reusing the old thread is inappropriate. If you prefer a new issue I can create one.

ChiWei-Chen, 2022-04-08:
@git1106 , thanks for sharing this information, I'll ask FW team to help this. (LO-1678 for @mrhpearson and my tracking )

git1106, 2022-04-09:
Thanks @ChiWei-Chen for the quick follow-up.

Actually I forgot to ask:

> Note: Please suspend or disable any disk encryption tool prior to update.

What does that mean? Of course my rootfs is encrypted. I have the "typical" setup:

* partition 1: ESP, no encryption
* partition 2: `/boot` no encryption
* partition 3: a LUKS container containing the rootfs

I don't think there is any way to suspend or disable LUKS. But I believe `fwupd` would really only work with the ESP, so the encrypted rootfs should not matter here. It has worked several times with earlier updates and I never did suspend or disable anything. The update (I believe it is) is still on my ESP after the failed update attempts:

```
$ sudo ls -l /boot/efi/EFI/ubuntu/fw/fwupd-c3e4be53-e714-4ea1-bb9c-7fe13a98b556.cap
-rwx------ 1 root root 2992928 Apr  8 15:23 /boot/efi/EFI/ubuntu/fw/fwupd-c3e4be53-e714-4ea1-bb9c-7fe13a98b556.cap
```

kmauleon, 2022-04-11:
hi @git1106 may i know your system bios version at the moment?  please advise... thank you very much...

kmauleon, 2022-04-11:
BTW, for the disk encryption, we are referring to the tool similar with windows bit locker. if such similar disk encryption tool is enabled, kindly disable first prior to update.  thank you very much.

superm1, 2022-04-12:
> BTW, for the disk encryption, we are referring to the tool similar with windows bit locker. if such similar disk encryption tool is enabled, kindly disable first prior to update.  thank you very much.

Presumably this is specifically if you have it tied to PCR attestation that may change. If disk encryption isn't tied to a PCR attestation it *shouldn't* matter.

hughsie, 2022-04-12:
If it helps, in fwupd >= 1.7.1 (feature added in d3706e0e0b0fc210796da839b84ac391f7a251f8) automatically shows the user a warning if it detects a BitLocker volume.

git1106, 2022-04-15:
@kmauleon wrote:

> hi @git1106 may i know your system bios version at the moment? please advise... thank you very much...

I currently have

```
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    0.1.49
  │   Minimum Version:    0.0.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              4bbc40fa-f81e-4206-bc70-a1f7b744d964
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
  │                       171d8529-50f6-550b-9db6-7f21af7414ef
```

But I see 0.1.50 has be become available since I first reported above. I'll report again after updating to 0.1.50.

git1106, 2022-04-15:
> I'll report again after updating to 0.1.50.

Looks like I am unable to do so. When I run `fwupdmgr update` it first tries 

> Upgrade Intel Management Engine from 192.81.1753 to 192.85.1869?

When that fails (in the same way as reported above) it does not try anything else. No idea whether I could force the order differently and whether that would be a safe thing to do.

git1106, 2022-04-15:
Sorry, this will be a long story, so I try to organize by chronological steps:

**Step 1**

> No idea whether I could force the order differently and whether that would be a safe thing to do.

I read the release notes of system firmware. There was no mentioning of any dependency and there were instructions to install just this firmware. So I dared to do it

```
$ fwupdmgr update 4bbc40fa-f81e-4206-bc70-a1f7b744d964
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.49 to 0.1.50?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo System Firmware Version 1.50                                          ║
║                                                                              ║
║ The computer will be restarted automatically after updating BIOS completely  ║
║ . Do NOT turn off your computer or remove the AC adaptor while update is in  ║
║ progress.                                                                    ║
║                                                                              ║
║ Important updates                                                            ║
║                                                                              ║
║ • Update includes a security fix.                                            ║
║                                                                              ║
║ New functions or enhancements                                                ║
║                                                                              ║
║ • Updated the Diagnostics module to version 04.22.000.                       ║
║                                                                              ║
║ Problem fixes                                                                ║
║                                                                              ║
║ • Fixed an issue where Power-On Password is set by WMI even if Power-On      ║
║ Password Control was disabled.                                               ║
║                                                                              ║
║ 20QD003JMX must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

That seem to succeed according the output summary (a bit weird because I had requested only an update of the system firmware, but who knows whether the ME update was still "in the queue"):

```
Devices that have been updated successfully:

 • Intel Management Engine (192.81.1753 → 192.85.1869)
 • System Firmware (0.1.49 → 0.1.50)
```

But according to the JSON the ME update had failed

```
"UpdateError" : "Cannot update as System Firmware [a45df35ac0e948ee180fe216a5f703f32dda163f] needs reboot",
"Guid" : [
  "c3e4be53-e714-4ea1-bb9c-7fe13a98b556"
],
"Plugin" : "uefi_capsule",
"VersionOld" : "192.81.1753",
"VersionNew" : "192.85.1869",
```

Ok, that doesn't sound too bad, let's just retry after reboot.

**Step 2**

However, after reboot `fwupdmgr` insisted that there are no updates, also ME would be up to date. But the version reported was still 1753 in both setup screen and the Intel tool. Where did the 1869 go?

So I tried a `fwupdmgr refresh --force`. And see there, a new system firmware 0.1.51 had appeared during the last 23 hours. And indeed the changes looked promising

```
        • Fixed an issue where Intel Management Engine firmware update by LVFS is failed.
```

So I installed 0.1.51.

That seem to succeed according the output summary (a bit weird because I when starting the update there was no mentioning that the ME would be updated, but who knows whether it was still "in the queue"):

```
Devices that have been updated successfully:

 • Intel Management Engine (192.81.1753 → 192.85.1869)
 • System Firmware (0.1.50 → 0.1.51)
```

But according to the JSON the ME update had failed once again:

```
"UpdateError" : "Cannot update as System Firmware [a45df35ac0e948ee180fe216a5f703f32dda163f] needs reboot",
"Guid" : [
  "c3e4be53-e714-4ea1-bb9c-7fe13a98b556"
],
"Plugin" : "uefi_capsule",
"VersionOld" : "192.81.1753",
"VersionNew" : "192.85.1869",
```

Also the setup screen and the Intel tool confirmed that I was still at 1753. But if it's only a reboot...

**Step 3**

Indeed the ME update from 1753 to 1869 was now offered again (and nothing else). And it seemed to succeed:

```
Devices that have been updated successfully:

 • System Firmware (0.1.50 → 0.1.51)
 • Intel Management Engine (192.81.1753 → 192.85.1869)
```

What? The System firmware had already been updated earlier. And indeed the JSON said it didn't work this time:

```
"UpdateError" : "Cannot update as Intel Management Engine [349bb341230b1a86e5effe7dfe4337e1590227bd] needs reboot",
"Guid" : [
  "4bbc40fa-f81e-4206-bc70-a1f7b744d964"
],
"Plugin" : "uefi_capsule",
"VersionOld" : "0.1.50",
"VersionNew" : "0.1.51",
```

But now the setup screen said that I would be on system firmware 1.51 and ME 1869. And the Intel tool reported the the same ME version. 

So let's hope it's all solved for my machine now.

Thanks everyone who replied here and those who worked on the code! 

P.S. Not sure whether `fwupdmgr` would need fixes to the "inconsistent" reporting between summary and JSON. Or whether they already exist in the newest version (I guess Ubuntu 20.04 might not have the newest one anymore).

kmauleon, 2022-04-21:
hi @git1106 I think the original issue has been solved by latest system firmware and now you are running with existing issue with BIOS ESRT table not being updated similar to this one >> #89 #65 which is still under investigation... thank you very much...

xvitaly, 2022-04-21:
> I think the original issue has been solved by latest system firmware

T480 is still not fixed.

kmauleon, 2022-04-21:
hi @xvitaly sorry but your T480 symptom is another issue handled by ODM firmware team and is under investigation and being tracked by @ChiWei-Chen.  Will feedback once any update from firmware team.  Thank you very much.

majamee, 2022-04-21:
Same issue applies and still applies to T14. Thank you for working on a fix.

[40 more replies]

closed 2022-09-12 by ChiWei-Chen