fwupd/firmware-lenovo #141: "Lenovo ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2" name ist ambiguous

bdaase, 2021-08-02:
The entry "Lenovo ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2" on the LVFS listed here https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34ET.firmware is ambiguous. It's unclear whether that references the Intel or the AMD model (or maybe both?). Therefore, one also can't determine whether both versions are supported via the LVFS (even though the hardware manual seems to imply that both are supported).
It would be nice to clarify that on the website.

bdaase, 2021-08-06:
Alright, this seems to be about the Intel version. The AMD version is, for whatever reason, missing on the lvfs (reported in https://github.com/fwupd/firmware-lenovo-thinkpad/issues/143).

mrhpearson, 2021-08-09:
We tend to refer to Txx G2 as the Intel version and Txx AMD G2 (I've also seen T14 G2 AMD) for the AMD version. I suspect it's somewhat historical (I'm guessing). It's even worse on our Lenovo product support pages in my opinion...

The page does say the FW contains "Intel BIOS Guard and "Intel Boot Guard" which are Intel specific :) And the Intel FW won't install on AMD and vice-versa.

I'm unlikely to get this changed...any objections if I close this?
Mark

bdaase, 2021-08-09:
Yeah, that's also what I figured out now :)

Anyway, I think it would be easiest if both versions have the Intel/AMD suffix. Especially for users new to the ThinkPad lineup, I think that would make things a bit easier to understand. Anyway, just my two cents ;)

So yeah, please feel free to close it then.

mrhpearson, 2021-08-09:
I don't disagree - just picking my battles right now with the FW team ;)
Thanks!
Mark

closed 2021-08-09 by mrhpearson