fwupd/firmware-lenovo #142: X1CG6 cannot update even after Ubuntu shim upgrade (and fwupd output confuses)

LinuxOnTheDesktop, 2021-08-03:
Previously my X1 Carbon 6th Generation ThinkPad could not install an update that fwupd offered. I gather that the problem was to do with something called a 'shim' and that Ubuntu was working upon a fix. What looked like the intended fix was offered via my distribution's repositories in the form of these packages:

- shim-signed:amd64 (1.40.4+15+1552672080.a4a1fbe-0ubuntu2, 1.40.6+15.4-0ubuntu7)
- shim:amd64 (15+1552672080.a4a1fbe-0ubuntu2, 15.4-0ubuntu7)

I installed those packages and rebooted and the fwupd update still does not work. I saw the following.

```
$ fwupdmgr update
Devices with no available firmware updates:
 • Thunderbolt host controller
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • LENSE20512GMSP34MEAT2TA
 • System Firmware
Upgrade available for UEFI Device Firmware from 0.71.12 to 0.73.20
20KH001UKP must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…***********************************]
Scheduling…              [***************************************]
Successfully installed firmware
 • Unifying Receiver

An update requires a reboot to complete. Restart now? [y|N]: y
```

That text is confusing in a twofold way. 1) What is the difference between, 'Devices with no available firmware updates' and, 'Devices with the latest available firmware version'? 2) The text seems to say that one thing ('UEFI Device Firmware') will be upgraded and also that a different thing (the 'Unifying Receiver', which is a USB device for a wireless mouse) has been upgraded.

More, and worse: nothing got upgraded. For, as before: upon reboot a Lenovo logo appears; it disappears after less than a second; I am returned to the desktop; when I proceed to run fwup I am offered the same upgrade again.

I appreciate that the continued problem with the 'shim' may not be your fault but the text that I mentioned above owes to you.

**My system**

OS: Linux Mint 20.2 x86_64 
Host: 20KH001UKP ThinkPad X1 Carbon 6th
Kernel: 5.11.0-25-generic 
Shell: bash 5.0.17 
Resolution: 2560x1440 @ 60.01Hz 
DE: Cinnamon 
```

$ fwupdmgr --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.11
```

hughsie, 2021-08-03:
I think 1.6.x handles this much better. Please upgrade to 1.6.2 and try again. Thanks.

closed 2021-08-03 by hughsie

LinuxOnTheDesktop, 2022-08-22:
May this issue be re-opened, please - in accord with [this request](https://github.com/fwupd/firmware-lenovo/issues/240#issuecomment-1221647991)?