fwupd/firmware-lenovo #143: ThinkPad P14s Gen 2 AMD system firmware is missing

bdaase, 2021-08-06:
The ThinkPad P14s Gen 2 AMD system firmware is missing on the lvfs, even though it is referenced as available in the corresponding hardware manual.

mrhpearson, 2021-08-09:
We haven't released P14s G2 AMD with Linux yet - there are still HW enablement issues we're working through.
Mark

closed 2021-08-09 by mrhpearson

bdaase, 2021-08-09:
Thanks for the fast response :)

Do you have a rough ETA when we can expect it?

mrhpearson, 2021-08-09:
We've been hit by WLAN component shortages and they switched parts on us very very late in the program. The new parts don't have Linux support upstream yet (vendors are working on it right now). I'm going to guess at least another month but I don't have an actual ETA.
Mark

bdaase, 2021-08-09:
Sure, the first thing I did after purchase was swapping out the Realtek card for an AX210 - with that everything works flawlessly under Linux so far for me :)

So we first have to wait for the hardware stuff to be fixed before we can expect system firmware on the LVFS?

mrhpearson, 2021-08-09:
I'll see if the FW team will put it up sooner. The problem is that they usually test FW updates with our preload image - which we don't have yet.
Mark

bdaase, 2021-08-10:
Thank you, Mark. This is highly appreciated!

bdaase, 2021-08-30:
Thank you, Mark, it looks like its available on the lvfs now: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1MET.firmware.

However, for some reason, even after a
```
fwupdmgr refresh --force
```
and a 
```
fwupdmgr update
```

it does not show me the update as available.