fwupd/firmware-lenovo #144: firmware is not updated after reboot on Lenovo P1 Gen 3

neupsh, 2021-08-12:
**Describe the bug**
My Lenovo Thinkpad P1 Gen 3 is failing to update the system firmware. `fwupdmgr update` finds the udpate and prepares for the update too, but when the computer is rebooted to apply the update, it quickly flashes the following text (had to take a video to see what it says):
`WARNING: No updates to process. Called in error?`

**Steps to Reproduce**
Steps to reproduce the behavior.
1. Run `fwupdmgr update`
2. Reboot the computer

**Expected behavior**
The new firmware update is applied and `fwupdmgr update` shows no new updates.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
#fwupdmgr --version

client version: 1.6.3
compile-time dependency versions
gusb:   0.3.7

daemon version: 1.6.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman -S fwupd` I have `fwupd-efi 1.1` installed as well

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

```
20TJS06X00
│
├─USB2.0 Hub:
│     Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│     Summary:            USB 3.x hub
│     Current version:    3.124
│     Vendor:             VIA Labs, Inc. (USB:0x2109)
│     Install Duration:   15 seconds
│     GUIDs:              7d751c46-559f-5a8c-bddd-7a874d716056
│                         3f0581b6-15ed-5f87-91eb-296db71e0838
│                         d390a7f5-ea48-5370-87a7-eba89bf5606e
│                         7fde3a64-a5e4-5ccf-880c-5f1c4f278c09
│                         f3db5c82-5c06-58d0-b1f6-cfa1642eaa1c
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
├─USB3.0 Hub:
│     Device ID:          da77984c82b59c6fc69516431f467fd9a8d39a7f
│     Summary:            USB 3.x hub
│     Current version:    3.124
│     Vendor:             VIA Labs, Inc. (USB:0x2109)
│     Install Duration:   15 seconds
│     GUIDs:              b187e9ff-6c12-5f27-9ae3-0ac891cb8f07
│                         26f33695-3a3e-5c08-badb-f6141390ebd9
│                         10eb3a15-c177-5810-af53-1963e9200e65
│                         b15e6e94-ddf5-579d-82db-3afc7dc964f4
│                         c3f32af1-992d-5981-8a60-b5a491933490
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
├─Core™ i7-10850H CPU @ 2.70GHz:
[... 262 lines cut ...]
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── initramfs-linux-zen-fallback.img
├── initramfs-linux-zen.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch-lts-auto.conf
│   │   ├── arch-lts.conf
│   │   ├── arch-zen-auto.conf
│   │   └── arch-zen.conf
│   ├── loader.conf
│   └── random-seed
├── vmlinuz-linux-lts
└── vmlinuz-linux-zen
```

**Additional questions**

- Operating system and version: arch linux
- Have you tried rebooting?: yes
- Is this a regression?: yes, I had installed firmware updates before
- Are you using an NVMe disk?: yes
- Is secure boot enabled?: no
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?: no, it is turned off

neupsh, 2021-08-12:
Just for reference:
I had left a detailed comment here before https://github.com/fwupd/firmware-lenovo-thinkpad/issues/80#issuecomment-857063805
and here as well: https://github.com/fwupd/firmware-lenovo-thinkpad/issues/132#issuecomment-889383318

kyak, 2021-12-05:
I have exactly the same problem on ThinkPad P15v Gen 1. Found this issue by searching the "WARNING: No updates to process. Called in error?" message.

kyak, 2022-07-23:
I made another attempt to update. This time, I had errors like this: https://github.com/fwupd/firmware-lenovo/issues/140#issuecomment-1122253353

After digging a little bit, I found out that my `esp` partition was of `Linux filesystem` type. I've changed it to `EFI System` like it should be. No idea why I partitioned it like this in the first place...

But anyway, this has helped and update went fine.

I suggest that developers should collect "EspType" in addition to "EspPath" in their update reports and inform users when ESP partition type is not set correctly.

LinuxOnTheDesktop, 2022-07-29:
@kyak 

One should see something like the following, yes?

[image]

And if one does _not_ see that 'EFI System', can one easily change that?

kyak, 2022-07-30:
@LinuxOnTheDesktop 

Yes, they should see something like that. You could also run `fdisk -l /dev/nvme0n1` and verify the correct partition type.

There are many ways to change partition type. I prefer to use `fdisk` utility (it has `t` command to change partition type).

cponder, 2022-10-18:
I've been having the same update-problem on my T580.
Here's the disk partitioning:
```
--> sudo fdisk -l /dev/nvme0n1

Disk /dev/nvme0n1: 476.94 GiB, 512110190592 bytes, 1000215216 sectors
Disk model: WDC PC SN720 SDAQNTW-512G-1001          
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 4A9E44A9-C48D-4496-9736-7C65AD8C9A4D

Device            Start        End   Sectors   Size Type
/dev/nvme0n1p1     2048   10487807  10485760     5G EFI System
/dev/nvme0n1p2 10487808   31459327  20971520    10G Linux filesystem
/dev/nvme0n1p3 31459328 1000212479 968753152 461.9G Linux filesystem
```
Is there something here that I need to change?