fwupd/firmware-lenovo #147: AM620 firmware bricking Lenovo Ideapad's SSDs

zubozrout, 2021-08-17:
Hello,
it seems that multiple Lenovo Ideapad laptops are "bricked" after installing this update:
https://fwupd.org/lvfs/devices/com.lenovo.AM620.firmware

I've opened an issue on [Ubuntu's Launchpad](https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1940140), however have been redirected here as this seems to be an issue with the FW itself.

I am not sure if the hardware in these laptops is not recognized properly so a wrong update is applied or if there is actually a different bug within the firmware but unfortunately installing this makes the SSD stop working. Neither bios nor any OS installer is able to see it unfortunately and therefore there is no way to boot the OS installed.

In my case it was the "Lenovo Ideapad 3-14ITL6". And I tried it twice unfortunately as I had no idea what caused it the first time hoping this was a specific device malfunction. I've only figured out the second time and after being given more details in my Launchpad report.

Thank you

mrhpearson, 2021-08-18:
Sent in a question is in to the FW team on this. Initial feedback is we don't think this SSD is used on any of our Linux certified platforms - just on Ideapads. 
So far we're not sure why it's on LVFS (normally I'd be happy to see more firmware on LVFS...but I don't think this went through our internal process). I'm waiting on more details.

@hughsie - suggest we pull this one until I have more details on why it's been uploaded and if the Ideapad are going to support it. I don't believe I have permissions to do that - are you able to demote it please?

@zubozrout - thanks for the note and I'm sorry for the troubles it caused.

Mark

hughsie, 2021-08-18:
I've moved it to testing for now. The lenovo user that promoted it was jmagara1_at_lenovo.com if that helps.

mrhpearson, 2021-08-18:
Some internal discussion. I was wrong and the SSD is also used on our Linux certified platforms and the update was tested there. We need to understand why it's failing on this platform.

@zubozrout - is it possible to get hold of the broken SSD drive so they can do some analysis? If so please email me (markpearson at lenovo dot com) and I'll put you in touch with the team looking at this to figure out the next steps

Mark

zubozrout, 2021-08-18:
Hi @mrhpearson, I've returned both laptops I've tried to a Czech reseller (CZC.cz) as I was quite sad over it and also needed to have this resolved quickly. They should all be now back at Lenovo.

maxicarlos08, 2021-11-04:
I used this update some time ago on my Lenovo IdeaPad 5 15ALC05, and the ssd got bricked (all information lost, and no backups :sob:).  
I sent the notebook to the Lenovo support and they gave it back with a replaced ssd and windows installed (good work Lenovo :+1:), since then I haven't touched fwupd

aladkoi, 2021-12-14:
I had the same problem. After installing the firmware, the SSD died. I just bought a new one from Samsung.

paulmenzel, 2022-01-10:
On https://fwupd.org/lvfs/devices/com.lenovo.AM620.firmware, the listed version 2.2Q0628 has a release date of 2021-10-15 02:38:04, after the issue was opened. What versions did you have problems with? If with that version, the firmware update should be removed to avoid bricking devices.

mrhpearson, 2022-04-05:
Cleaning up my Inbox - and I believe this one got fixed with the update posted above, at least as far as I can tell from the internal email thread on this topic.
Closing the issue - please let me know if you're still seeing the problem with the latest FW version.