fwupd/firmware-lenovo #148: My lenovo L15 reboots and updates are not applied

GBT, 2021-08-21:
# Describe the bug
I have firmware updates offered when I run fwupmgr :
```
Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • Micron MTFDHBA256TDV
Upgrade available for System Firmware from 0.1.8 to 0.1.10
20U4S4RB00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Té
```

When I reboot  firmware update is not applied

# Steps to Reproduce
```
sudo fwupmgr update
reboot
```
# Expected behavior
After the reboot running the update command returns that I'm up to date.

# fwupd version information
Please provide the version of the daemon and client.
```
fwupdmgr --version
client version: 1.5.12
compile-time dependency versions
        gusb:   0.3.7
daemon version: 1.5.12
```
Please note how you installed it (apt, dnf, pacman, source, etc): **dnf**

# fwupd device information
Please provide the output of the fwupd devices recognized in your system.
```
fwupdmgr get-devices --show-all-devices
20U4S4RB00
│
├─Embedded Controller:
│     Device ID:          176c2a45e48dd6f44b559f4c0ae8b2d9c501472a
│     Version actuelle:   0.1.11
│     Version minimum:    0.1.11
│     Fournisseur:        DMI:LENOVO
│     Update State:       Success
│     GUIDs:              feac2f38-19c3-4fae-bf6b-1d8020e6e1ef
│                         019a55e5-f0a0-5318-a25e-cba4c2eacb67
│     Device Flags:       • Périphérique interne
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          8968a3bef0b7bbcda914f444d4c5ebcda641032f
│     Version actuelle:   58.18
│     Fournisseur:        Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              a15467f7-ddae-5090-9105-5d99f915f936
│                         5e9afdd3-1159-5da5-9b47-113b031f5409
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
[... 265 lines cut ...]
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.31
├── System.map-5.13.10-200.fc34.x86_64
├── System.map-5.13.12-200.fc34.x86_64
├── System.map-5.13.9-200.fc34.x86_64
├── vmlinuz-0-rescue-be52e3b1212f4e019be328c7fc9b8234
├── vmlinuz-5.13.10-200.fc34.x86_64
├── vmlinuz-5.13.12-200.fc34.x86_64
└── vmlinuz-5.13.9-200.fc34.x86_64

6 directories, 77 files
```

# Additional questions
    Operating system and version:
    Fedora 34
    Have you tried rebooting?
* yes of course
    Is this a regression?
* I think so.
    Are you using an NVMe disk?
* yes
    Is secure boot enabled?
* i dont know

deimosfr, 2021-09-04:
Same for me on Arch Linux

elagost, 2021-09-19:
For what it's worth I encountered something similar today (Fedora 34, T480) - repeatedly trying to apply updates, but the boot entry doing nothing and just kicking me back to Fedora's GRUB. I had applied updates a number of months ago. The fix as others have mentioned was to delete my Linux-Firmware-Updater entry in efibootmgr (`efibootmgr -b <entry> -B`; find `<entry>` by running `efibootmgr` with no arguments) and re-run the update. Only then did it succeed.

LinuxOnTheDesktop, 2021-10-05:
Let me elaborate on elagost's procedure, via what I see on my machine.

```
me@x1 ~ $ efibootmgr
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0018,0019,001A,001B,001C,001D,001E,001F,0002
Boot0000* Windows Boot Manager
Boot0001* ubuntu
Boot0002* Linux-Firmware-Updater
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Regulatory Information
Boot0015  Startup Interrupt Menu
Boot0016  Rescue and Recovery
Boot0017  MEBx Hot Key
Boot0018* USB CD
Boot0019* USB FDD
Boot001A* NVMe0
Boot001B* ATA HDD0
Boot001C* USB HDD
Boot001D* PCI LAN
Boot001E  Other CD
Boot001F  Other HDD
Boot0020* USBR BOOT CDROM
Boot0021* USBR BOOT Floppy
Boot0022* ATA HDD
Boot0023* ATAPI CD
me@x1 ~ $ efibootmgr -b 2 -B
Could not delete variable: Permission denied
me@x1 ~ $ sudo efibootmgr -b 2 -B
[sudo] password for me: 
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0018,0019,001A,001B,001C,001D,001E,001F
Boot0000* Windows Boot Manager
Boot0001* ubuntu
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Regulatory Information
Boot0015  Startup Interrupt Menu
Boot0016  Rescue and Recovery
Boot0017  MEBx Hot Key
Boot0018* USB CD
Boot0019* USB FDD
Boot001A* NVMe0
Boot001B* ATA HDD0
Boot001C* USB HDD
Boot001D* PCI LAN
Boot001E  Other CD
Boot001F  Other HDD
Boot0020* USBR BOOT CDROM
Boot0021* USBR BOOT Floppy
Boot0022* ATA HDD
Boot0023* ATAPI CD
```
Right - let's see whether it works i.e. whether the update gets applied:
```
me@x1 ~ $ sudo fwupdmgr get-updates && sudo fwupdmgr update && sudo reboot
```

LinuxOnTheDesktop, 2021-10-05:
Postscript to my post immediately above: it did not work (and I even tried deleting the Windows boot manager, but that did not help).

elagost, 2021-10-05:
@LinuxOnTheDesktop do you have secure boot on?

mrhpearson, 2021-10-05:
Also (just because I literally suffered this myself on a P15 and scratched by head a bit before realising what I was missing)
 - Make sure your system is plugged in
 - Make sure your battery is charged more than 60% (I think it needs 40%...but not completely sure so playing safe)
 - Recommend disabling secure boot just because there have been some issues with shim that are being fixed but aren't rolled out everywhere. The only failure report against this firmware on LVFS is for a system with SB enabled.

LinuxOnTheDesktop, 2021-10-05:
@elagost: no, nor do I have boot order locked.

MartiniMoe, 2021-12-01:
I'm also having this exact same issue, but on a T14s AMD.
Secure Boot is off, Boot order not locked, Battery charged, Power plugged in. I also deleted the Firmware Updater EFI entry. Nothing seems to work :/

MartiniMoe, 2021-12-01:
> I'm also having this exact same issue, but on a T14s AMD. Secure Boot is off, Boot order not locked, Battery charged, Power plugged in. I also deleted the Firmware Updater EFI entry. Nothing seems to work :/

Nvm, it worked when I deleted the Firmware Update EFI entry again after disabling secure boot and then installing the update again :)