fwupd/firmware-lenovo #149: Thinkpad T490s Intel Management Engine Update No updates to process, called in error, ...

jknaus, 2021-08-24:
fwupd presented me with an update for Intel ME (192.72.1757 -> 192.81.1753).
The update was seemingly applied correctly (192.81.1753 is shown to be installed).
However, at every boot the message
`No updates to process, called in error, start_image: invalid parameter {}`
is shortly shown on the boot screen.
Strangely enough, I cannot find version 192.81.1753 among the Intel ME updates on fwupd.org.
I already tried to reinstall (no change) and to downgrade (aborted with error) but this did not help.
How can I get rid of the error message and make sure that the fw is correctly installed?

OS: Fedora 33, 
secure boot enabled, 
fwupd installed by default,
nvme disk

```
$ fwupdmgr --version
client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.9
```
```
$ fwupdmgr get-devices
20NXCTO1WW
│
├─Thunderbolt host controller:
│     Device ID:          776aac87cf5df2f523162c58ef290fdeb1aeff73
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              ea3c7ac8-d937-568a-aab5-e053bd6af9ce
│                         3d3f953a-3408-56fa-81f3-042d73c59c16
│                         e773c51e-a20c-5b29-9f09-6bb0e0ef7560
│                         f1793149-6c12-5a7f-95db-9fefd116748d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          09ce7908b2d90a3b53473c3f328410590a527f29
│     Current version:    0.1.19
│     Minimum Version:    0.1.19
│     Vendor:             DMI:LENOVO
│     GUIDs:              be9a5f93-930b-40bf-8864-815268ea4ea3
│                         98129d04-2653-5233-89f1-8aa461e54a25
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─INTEL SSDPEKKF010T8L:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:            NVM Express Solid State Drive
│     Current version:    L11P
│     Vendor:             Intel Corporation (NVME:0x8086)
│     GUIDs:              f91a5c60-8696-539e-9d0a-57d194f74ac4
│                         396fb4fd-7656-564f-88b5-381e83ca30c0
│                         79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                         68db11e5-b0cf-5bc9-a94e-17e28496e505
│                         691bd9e3-7825-50b0-837e-2cfe70fe7e87
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
[... 126 lines cut ...]
Boot0011* LENOVO CLOUD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri(https://download.lenovo.com/pccbbs/cdeploy/efi/boot.efi)
Boot0012* HTTPS BOOT	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri()
Boot0013  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0014  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0015  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0016  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0017  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0018  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0019  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot001A  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot001B* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot001C* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001D* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001E* NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001F* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot0020* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot0021* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0022* PXE BOOT	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0023  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0024  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0025* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0026* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0027* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0028* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

jknaus, 2021-09-02:
I updated the complete error message that appears at each boot.

The update was applied correctly, as fwupd shows the new version number for Intel ME.
However, the laptop tries to boot into the fw-update image/shim (I do not know the exact naming as I am not an expert in this field), which is empty (?) as everything was already processed. 

So, maybe it is just that the shim was not removed correctly at the end of the fw update?
Is there a way to do this manually?

Thanks in advance for your help.

mrhpearson, 2021-09-03:
You can use efibootmgr to change your bootorder. I think
efibootmgr -o 0000,001B,001C,001D,001E,001F,0020,0021,0022,0012,0024,0023,0011
would do it (but I don't mess with this stuff very often)

Don't remove shim - you need that :) I think the problem is just that it's trying to run the firmware-updater first on every boot when there is no update to run

@hughsie - is this a fwupdmgr thing or Lenovo FW? I thought fwupdmgr but let me know if I'm wrong - my understanding is that per the UEFI spec the FW is not supposed to remove any entries.

Thanks
Mark

hughsie, 2021-09-03:
> @hughsie - is this a fwupdmgr thing or Lenovo FW?

The latter I fear. fwupd just sets up BootNext which is only used *once* rather than changing the BootOrder. 

> per the UEFI spec the FW is not supposed to remove any entries

The spec part yes, but I wish that was true in reality. The Lenovo firmware actually looks for `Linux-Firmware-Update` and special cases that so it doesn't get "cleaned up". That's why we force the label to have tabs and not use the spaces like every other vendor: https://github.com/fwupd/fwupd/blob/master/plugins/uefi-capsule/uefi-capsule.quirk#L19

mrhpearson, 2021-09-03:
Got it - thanks! So somehow/somewhere the boot-order got set here to always run the FW updater...interesting.

@jknaus - I think if you clean up your boot-order that will fix it and get you up and running. We'll have to see if we can reproduce and are messing things up somehow.

Mark

jknaus, 2021-09-04:
Thank you. I did take out 
```
Boot0001* Linux-Firmware-Updater	HD(1,GPT,69cfa9af-c2fc-4d72-be36-fe5ff7fd8127,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
```
by applying
`efibootmgr -o 0000,001B,001C,001D,001E,001F,0020,0021,0022,0012,0024,0023,0011`
and this worked out well.
Thank you very much for helping me.

Johannes

(I have one additional question, however, as this is another topic I'll put it into another issue.)

LinuxOnTheDesktop, 2021-09-10:
Is a meta-bug (/issue) in order? For, we have fwupd + Lenovo's firmware updates resulting in the following.

_Failed attempts_: updates not being applied when they should be (see #109, #118, #140, #144, #148 etc. );
_Bogus attempts_: repeated attempts to install non-existent updates (this bug);
_Non-attempts_: updates that should be offered not being offered at all (#105, #138, #152). 

All the bug reports that I have cited are open. If the roots of the problems lie mainly with Lenovo then perhaps it would be good to try to convey a sense of urgency to the relevant persons.