fwupd/firmware-lenovo #151: Missing firmware 1.35 for Lenovo

csantelmann, 2021-09-03:
Hi,
I got a Lenovo X13 notebook.
According to the Lenovo site, there is a new firmware 1.35 available:
https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x13-type-20uf-20ug/downloads/DS544980
fwupd does not offer the new version.
Can I do anything to help fix this?
Regards
Cord

mrhpearson, 2021-09-03:
It will make it's way to LVFS - it just takes our BIOS teams some time to prepare the LVFS release, test it and deliver it. 
We're looking into how to streamline the process but it's complicated (and a honestly a little frustrating).
Mark

csantelmann, 2021-09-07:
Thanks a lot for the information! I'm in no hurry.
Cord

closed 2021-09-07 by csantelmann