fwupd/firmware-lenovo #152: Intel SSD firmware update available on Lenovo website but not via fwupdmgr

jknaus, 2021-09-04:
Hello,

I have a Thinkpad T490s with an Intel SSDPEKKF010T8L running the fw-version L11P.
The Lenovo-Support-Website <https://pcsupport.lenovo.com/fi/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t490s-type-20nx-20ny/downloads/ds545014-intel-ssd-firmware-updater-for-linux-os-thinkpad> shows there is an update available to version L14P (release date September 2020).
There is even a download available for my Linux distribution (Fedora):
<https://download.lenovo.com/pccbbs/mobiles/isful02.txt>
<https://download.lenovo.com/pccbbs/mobiles/isful02.zip>

However, `fwupdmgr` does not show me this release at all.
I guess Lenovo missed to put this on fwupd.org?

Johannes

mrhpearson, 2021-09-07:
Strange - we've released a whole bunch of SSD updates on LVFS.

Can you confirm what the GUID is for your SSD part please (fwupdmgr get-devices)?

jknaus, 2021-09-12:
@mrhpearson here you are:
```
─INTEL SSDPEKKF010T8L:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:            NVM Express Solid State Drive
│     Current version:    L11P
│     Vendor:             Intel Corporation (NVME:0x8086)
│     GUIDs:              f91a5c60-8696-539e-9d0a-57d194f74ac4
│                         396fb4fd-7656-564f-88b5-381e83ca30c0
│                         79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                         68db11e5-b0cf-5bc9-a94e-17e28496e505
│                         691bd9e3-7825-50b0-837e-2cfe70fe7e87
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
´´´