fwupd/firmware-lenovo #153: Lenovo T480 update error

oschwald, 2021-09-10:
When running `fwupdmgr update`, I get the error `Document must begin with an element` with `5a70ec7f22e97bfe090387fae06debcff2788e387c3d87fdc4ac404d744a26e3-Lenovo-ThinkPad-T480-ECFirmware-1.22.cab`.

Debug output:

```
Upgrade available for Embedded Controller from 0.1.21 to 0.1.22
20L5CTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:04.799: downloading https://fwupd.org/downloads/5a70ec7f22e97bfe090387fae06debcff2788e387c3d87fdc4ac404d744a26e3-Lenovo-ThinkPad-T480-ECFirmware-1.22.cab
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:04.800: Emitting ::status-changed() [downloading]
Downloading…             [ -                                     ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.072: download progress: 2%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.073: download progress: 2%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.073: download progress: 2%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.073: download progress: 2%
Downloading…             [                                       ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.074: download progress: 3%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.074: download progress: 3%
Downloading…             [*                                      ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.085: download progress: 5%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.085: download progress: 5%
Downloading…             [*                                      ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.088: download progress: 12%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.088: download progress: 12%
Downloading…             [****                                   ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.095: download progress: 12%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.095: download progress: 12%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.095: download progress: 12%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.096: download progress: 12%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.096: download progress: 13%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.096: download progress: 13%
Downloading…             [*****                                  ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.098: download progress: 13%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.099: download progress: 13%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.102: download progress: 23%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.102: download progress: 23%
Downloading…             [********                               ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.107: download progress: 23%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.107: download progress: 23%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.107: download progress: 25%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.108: download progress: 25%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.108: download progress: 25%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.108: download progress: 25%
Downloading…             [*********                              ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.110: download progress: 25%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.110: download progress: 25%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.111: download progress: 29%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.112: download progress: 29%
Downloading…             [***********                            ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.114: download progress: 29%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.114: download progress: 29%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.115: download progress: 31%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.115: download progress: 31%
Downloading…             [************                           ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.119: download progress: 38%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.119: download progress: 38%
Downloading…             [**************                         ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.121: download progress: 42%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.121: download progress: 42%
Downloading…             [****************                       ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.122: download progress: 46%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.122: download progress: 46%
Downloading…             [*****************                      ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.138: download progress: 50%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.138: download progress: 50%
Downloading…             [*******************                    ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.140: download progress: 85%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.140: download progress: 85%
Downloading…             [*********************************      ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.141: download progress: 85%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.141: download progress: 85%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.141: download progress: 85%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.141: download progress: 85%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.143: download progress: 87%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.143: download progress: 87%
Downloading…             [*********************************      ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.144: download progress: 87%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.144: download progress: 87%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.144: download progress: 87%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.144: download progress: 87%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.145: download progress: 89%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.145: download progress: 89%
Downloading…             [**********************************     ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.147: download progress: 93%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.147: download progress: 93%
Downloading…             [************************************   ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.148: download progress: 95%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.148: download progress: 95%
Downloading…             [*************************************  ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.149: download progress: 97%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.149: download progress: 97%
Downloading…             [*************************************  ](fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.150: download progress: 97%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.150: download progress: 97%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.150: download progress: 100%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.150: download progress: 100%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.150: download progress: 100%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.150: download progress: 100%
(fwupdmgr:709547): Fwupd-DEBUG: 13:59:05.150: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
failed to compile 5b41632e511b0a4f42b665ea95f8dd8acb95669a:prefix=components: Error on line 46 char 2: Document must begin with an element (e.g. <book>)
```

cwozny, 2021-09-11:
I have this same issue on my Lenovo Thinkpad T480 running Ubuntu 20.04.3. fwupdmgr client version 1.5.11.

Here is the output from running the update:

> $ fwupdmgr update
Devices with no available firmware updates: 
 • Thunderbolt host controller
 • TOSHIBA-RC100
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Upgrade available for Embedded Controller from 0.1.20 to 0.1.22
20L5CTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading…             [***************************************]
Decompressing…           [***************************************]failed to compile 5b41632e511b0a4f42b665ea95f8dd8acb95669a:prefix=components: Error on line 46 char 2: Document must begin with an element (e.g. <book>)

Here is my list of devices:

> $ fwupdmgr get-devices
Decompressing…           [-                                      ]20L5CTO1WW
│
├─Thunderbolt host controller:
│     Device ID:          5fe99260aee099bc15791d928679186337444171
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109, TBT:0x0109)
│     GUIDs:              8487e4cb-fe6b-5cb6-ad88-46535fb110e1
│                         6d4d771c-0579-504d-92a1-737180c7b360
│                         fe328c2d-5300-597b-a21d-991116db8ab6
│                         15a5ae1d-0ec9-5013-9221-cd6dd753fcb0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          c19580db344e418cf2b23297bcf46d75c06e432a
│     Current version:    0.1.20
│     Minimum Version:    0.1.20
│     Vendor:             DMI:LENOVO
│     GUIDs:              b87a926d-189c-49a1-b1ff-921099b1de89
│                         646e8b00-c980-5327-b889-b518b0a1cc1c
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          5decfff65e3391259bca4003a090e59bf50e3bea
│     Current version:    184.70.3626
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              dd31d983-cf3b-4c84-acc9-70caa94e827b
│                         49b00a7c-b9ac-5782-aaf9-76d3aa53f003
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          48477ba14bd9bd0c97323a07caa54b7766f917c2
│ │   Current version:    0.1.31
│ │   Minimum Version:    0.0.1
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   GUIDs:              dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
[... 40 lines cut ...]
├─UEFI Device Firmware:
│     Device ID:          27314a74b0935338fbf788ab05043e745a318147
│     Current version:    33619976
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     GUIDs:              2296f4c0-13a0-48f0-b5cb-83e7a1b868a2
│                         151d84eb-85e8-5e3a-9b26-127d2edbb3ca
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          a567cff02396e1038d53d7433ef3dbcd9558aebe
      Current version:    16842757
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      GUIDs:              d4583ddb-bf0a-4801-aefd-951930545cb9
                          792f52cb-e0bb-5378-b4cf-39dd99090835
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

ccoenen, 2021-10-03:
same here:

```
$ sudo fwupdmgr update
[...]
Upgrade available for Embedded Controller from 0.1.21 to 0.1.22
<model number redacted> must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************]
failed to compile 5b41632e511b0a4f42b665ea95f8dd8acb95669a:prefix=components: Error on line 46 char 2: Document must begin with an element (e.g. <book>)
```

hughsie, 2021-10-04:
What fwupd version?

oschwald, 2021-10-04:
I am on Ubuntu 21.04. It looks like it is using version 1.5.11.

```
$ fwupdmgr  --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.5

Decompressing…           [-                                      ]daemon version:	1.5.11
```

ccoenen, 2021-12-15:
I just verified that I still have this issue. I'm on a pretty default ubuntu 21.10 install with fwupdmgr 1.5.11. I still get the same message as described in https://github.com/fwupd/firmware-lenovo-thinkpad/issues/153#issuecomment-932975365.

outbacker, 2021-12-23:
I'm also affected. I tried forcing refresh as well as purging and reinstalling fwupd fwupdate fwupdate-signed but to no avail.
Seems like we aren't the only affected users either:
https://www.reddit.com/r/pop_os/comments/p7a4zb/error_while_trying_to_update_firmware_on_a_lenovo/

hughsie, 2021-12-24:
You need to use a newer fwupd version I'm afraid.

cwozny, 2021-12-24:
> You need to use a newer fwupd version I'm afraid.

Which newer version specifically resolved this issue?

hughsie, 2021-12-24:
> Which newer version specifically resolved this issue?

I think literally any newer version than you have installed.

outbacker, 2021-12-24:
@hughsie Thanks for the hint. I'm on Kubuntu 20.4 and was able to get past the issue of this ticket, but it was a bit fiddly.

First I had to apt purge fwupd. Then I tried installing the snap version via the recommended "snap install fwupd --classic" but instantly ran into another error:
"Run install hook of "fwupd" snap if present (run hook "install": install: cannot stat '/snap/fwupd/3664/share/dbus-1/interfaces/org.freedesktop.fwupd.xml': No such file or directory) "

I did manage to install the candidate version thougn:
# snap install fwupd --candidate --classic

So now I seem to be stuck at the [infamous shim issue](https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot ), as the boot order sn't locked and I tried updating one firmware at a time with:
# fwupdmgr update update_ID

I guess it's time to get my bootable windows thumbdrive out. ;)

cwozny, 2021-12-26:
Updated to fwupdmgr 1.7.3 via snap after purging 1.5.11. Now I'm hitting a different error:

WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
start_image() returned Unsupported

Which I believe is related to issue #1190 and what @outbacker ran in to

For the life of me, it feels like an impossible task to update the ThinkPad T480 firmware in Linux 😭

camilleacey, 2021-12-27:
After installing the candidate, everything went wrong something about "AppArmor Policy prevents this sender" so I ended up 
running

`sudo apt purge fwupd`
then 
`sudo apt install fwupd`
then
`sudo apt update`

Now when I check Ubuntu Software > Updates it says software is up to date.

mrhpearson, 2022-04-05:
I think we concluded this was fixed by a newer version of fwupd.

 As no updates in the last 3 months I'll close the ticket - let me know if that's incorrect

closed 2022-04-05 by mrhpearson

ccoenen, 2022-05-16:
This issue disappeared for me with ubuntu 22.04 LTS, so I guess the newer version finally made it all the way to me :-)