fwupd/firmware-lenovo #154: BIOS Update 1.22 - ThinkPad P14s Gen 1, P15s Gen 1, T14 Gen 1, T15 Gen 1

plafue, 2021-09-14:
Support page:
https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14-type-20s0-20s1/20s0/20s0cto1ww/pf2aqpz7/downloads/ds544549-bios-update-utility-bootable-cd-for-windows-10-64-bit-linux-thinkpad-p14s-gen-1-p15s-gen-1-t14-gen-1-t15-gen-1

Direct zip download:

https://download.lenovo.com/pccbbs/mobiles/n2xul14w.zip

Zip file contains UEFI BIOS update (`N2XET32W.cab`) and embedded controller update (`N2XHT23W.cab`).

Bios update worked without problems via `fwupdmgr install`. Embedded controller update failed with the following message:

```
≻  fwupdmgr install Downloads/N2XHT23W.cab
Decompressing…           [***************************************]
Specified firmware is older than installed '0.1.13 < 65541'
```

Get-devices shows the version of the controller as `65541`
```
├─Embedded Controller:
│     Device ID:          xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
│     Current version:    65541
│     Minimum Version:    65541
│     Vendor:             DMI:LENOVO
│     GUIDs:              359c1169-abd6-4a0d-8bce-e4d4713335c1
│                         7ca908cb-fb7d-5573-834c-33b8054e2b7a
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```

Thank you very much in advance!

mrhpearson, 2021-09-14:
Looks like they haven't set the version format in the cap file (interestingly they did it on the BIOS file...) I'll forward to the FW team.

FWIW - I'd wait until the FW was released on LVFS too, as then they'll have tested it with fwupdmgr and certified it. Might save you some headaches

Mark

mrhpearson, 2021-09-14:
For my records: DFT163706

plafue, 2021-09-14:
Thanks, @mrhpearson !

plafue, 2021-12-16:
A quick note: there was an update in November (BIOS 1.23, embedded controller 1.13 : https://download.lenovo.com/pccbbs/mobiles/n2xul15w.zip)

mrhpearson, 2022-04-05:
I think this all got resolved - checking the latest on LVFS looks good anyway.