fwupd/firmware-lenovo #155: ThinkPad T480s - Unable to update ME and Bios (Error 105)

saikocat, 2021-09-17:
**Describe the bug**

Devices that were not updated correctly:
 • Intel Management Engine (184.80.3746 → 184.86.3909)
 • System Firmware (0.1.42 → 0.1.47)

After running `fwupdmgr update` and said yes to all the prompt, and system restarted. While applying the update, it will report the failure in applying update with the following error message:

```
Error 105 - Command line error! Status = 105.
```

**System info**

Secure Boot: No

```
$ fwupdmgr --version
client version: 1.6.3
compile-time dependency versions
        gusb:   0.3.7

daemon version: 1.6.3
```

```
uname -a
Linux shingetsu 5.14.5-arch1-1 #1 SMP PREEMPT Thu, 16 Sep 2021 11:02:40 +0000 x86_64 GNU/Linux
```

Croissong, 2021-11-07:
I was able to at least update the Bios by selecting it via `fwupdmgr update <Device ID>`

sydneymeyer, 2022-02-21:
@saikocat Have you found a way to apply the updates, specifically the Intel ME update, in the meantime? I was able to update, at least, the BIOS/System FW directly via ` fwupdmgr update <Device ID>`, as suggested by @Croissong, but get the same error message when trying to update the Intel ME on a ThinkPad X1 (6th Gen) from 2018.

saikocat, 2022-02-22:
@sydneymeyer no, I still can't update the Intel ME. Keep a pin handy for the emergency reset button at the bottom of the laptop, as that's only way to reset the laptop when upgrading stopped half way.
And yes, the BIOS firmware can be updated via the method suggested by croissong.

sydneymeyer, 2022-02-22:
@saikocat Thanks for the update. I also have tried to install the update manually by downloading the ME update cab file for linux from the official lenovo support site, but with that i only get `firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer.`

It appears that Lenovo has botched this particular ME update for good then..

hughsie, 2022-02-23:
> but with that i only get firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf

This is because Ubuntu hasn't updated libjcat since 2020. Update libjcat and this will go away.

> It appears that Lenovo has botched this particular ME update for good then..

No, this is an Ubuntu problem. I've raised the jcat dep in the next fwupd version artificially to force them to update the version when they backport the next fwupd.

sydneymeyer, 2022-02-23:
I could have mentioned this earlier, i'm running NixOS unstable with libjcat 0.1.10. @hughsie should open a separate issue?

hughsie, 2022-02-23:
> NixOS unstable with libjcat 0.1.10

No idea then. I guess the daemon verbose log might show you what's going on, i.e. if it's failing to find the shipped public keys perhaps.

mrhpearson, 2022-04-07:
Hi @sydneymeyer 

Just because we've had a ton of problems around the ME FW update - can you go into the BIOS setup screen (F1 during early boot) and confirm which version of FW is installed please? It's possible your ME has installed succesfully and you've hit a common problem where the version is incorrectly reported to fwupd so it thinks it has failed. 

Mark

sydneymeyer, 2022-04-07:
@mrhpearson haven't you asked me this already about a month ago or am i missing something? Please see the response to this comment: https://github.com/fwupd/firmware-lenovo/issues/202#issuecomment-1059437303

mrhpearson, 2022-04-11:
Ah - apologies - I hadn't cross-referenced the two different issues.

So yeah, that confirms that you've hit the ESRT table issue and your FW is updated, just not being reported correctly. So I think that means the failure update is likely because there isn't anything to update - you are updated to 3987 at this point.

We're waiting on the fix from the FW team on the ESRT issue.

kparal, 2022-06-20:
I just received System Firmware `0.1.50` for my T480s. Among the changelog, there is:
> Fixed an issue where Intel Management Engine firmware update by LVFS is failed.

However, it doesn't seem to work. Even after updating to `0.1.50`, I still see Intel ME `184.90.3987` as a pending update (and `184.86.3909` as the current version), and trying to update it doesn't change anything. I see the same screen as above for a short while, then it reboots, and still shows Intel ME update available. It no longer marks the attempt as failed, though, I think. But otherwise no change.

closed 2022-07-01 by saikocat

rotanid, 2022-07-01:
> This has been resolved for me. After updating to `0.1.50` version of the bios, everything updated correctly.
> 
> ```
> Devices that have been updated successfully:
> 
>  • Intel Management Engine (184.80.3746 → 184.86.3909)
>  • System Firmware (0.1.49 → 0.1.50)
> ```

the update to system firmware 0.1.50 also worked for me, but 184.86.3909 is not up to date, this version i already have.
up to date would be 184.92.4222 which is offered by "get-updates" but the update doesn't succeed for me.

kparal, 2022-07-04:
I can confirm what @rotanid said. I also received update to **184.92.4222** which I applied and it shows up in BIOS. But fwupdmgr still thinks the current version is **184.86.3909**. So the new update didn't solve anything, the problem is still the same.

@saikocat I doubt your system is working as expected either. You posted:
> Devices that have been updated successfully:
> • Intel Management Engine (184.80.3746 → 184.86.3909)

That's **184.86.3909**, the old problematic version. Can you check what `fwupdmgr update` tells you and what BIOS tells you? Perhaps it also thinks you're still at the old version while you applied a newer one.

I think this should either be reopened or we should file a new bug, because gnome-software/fwupdmgr is constantly bugging me to apply an update I already applied.

@mrhpearson Can you please poke the FW team again? The `0.1.50` system firmware update fixed the update failure (it can be applied now), but it didn't fix version reporting back to fwupdmgr (or there's another bug in it as well).

saikocat, 2022-07-04:
@kparal It (184.92.4222) doesn't even get offered to me. But yeah, I do see it on Lenovo page and the lvfs firmware.

```
 $ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 4 local devices supported

$ fwupdmgr get-updates
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • System Firmware
 • Thunderbolt host controller
No updates available

$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.3
runtime   com.hughsie.libjcat           0.1.11
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.9-arch1-1
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10

$ fwupdmgr get-devices
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.50
│ │   Minimum Version:    0.1.32
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    3092254370
│     Minimum Version:    3092254370
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success

```

kparal, 2022-07-04:
This is so weird. You have:
```
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    3092254370
```
and I have:
```
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.86.3909
```
Where did you get that weird **3092254370** version? Perhaps that's the reason why updates are not offered to you? What do you see as an ME version directly in BIOS?

saikocat, 2022-07-04:
Probably I am hit by this ESRT being incorrect issue https://github.com/fwupd/firmware-lenovo/issues/89 .

kparal, 2022-07-04:
Thanks for the link. Sigh, that seems like all of us (@saikocat, @rotanid, me) are affected by that issue, just in a bit different way. I'll subscribe in that ticket.